procount/pinn #483: [OS REQUEST] Risc OS Direct

KrzysztofJeszke, 2021-02-11:
It's basically RISC OS but with a lot more software included, better for beginners than the regular RISC OS.
[(https://riscosdev.com/direct/)]

procount, 2021-02-11:
RISCOS has always been a bit awkward due to the file system it uses, therefore PINN (and NOOBS) have relied upon a binary blob that the maintainers produce in order to get it to work in a multiboot fashion.
I know that they were working on using a more common filesystem, but I don't know if it has been implemented yet, or if a binary blob to support the direct version is available.

KrzysztofJeszke, 2021-02-12:
> RISCOS has always been a bit awkward due to the file system it uses, therefore PINN (and NOOBS) have relied upon a binary blob that the maintainers produce in order to get it to work in a multiboot [...]

What is the filename of the binary blob? i might request it from the maintainers.

procount, 2021-02-12:
riscos-boot.bin. Also needed is the riscos offset (it was 1760 before I think). It may require them to produce specific NOOBS output files too.

lurch, 2021-02-12:
> riscos-boot.bin.

https://github.com/procount/pinn/blob/master/recovery/config.h#L69

> Also needed is the riscos offset (it was 1760 before I think).

Yep, http://downloads.raspberrypi.org/os_list_v3.json still says `"riscos_offset": 1760,`
Used here: https://github.com/procount/pinn/blob/master/recovery/mainwindow.cpp#L820

> It may require them to produce specific NOOBS output files too.

Also true:
```
            "tarballs": [
                "http://packages.riscosopen.org/noobs/RISC_OS_Boot.tar.xz",
                "http://packages.riscosopen.org/noobs/RISC_OS.xz"
            ]
```

Fact-checking @lurch :wink: