procount/pinn #484: (Allow to) Increase font size

redm123, 2021-02-14:
I run PINN on my TV screen (1080p), which is not that small, but I'm usually a few meters away. From that distance the texts in PINN are somewhat hard to read. It would be great if there was a knob somewhere to increase fontsize (and probably the entire window). It should be easily accessible. Maybe in that language bar at the bottom?
Or the fancy version: if a TV is detected make it bigger by default.
(I wonder what it would look like on a 4K TV?)

procount, 2021-02-15:
Try pressing +/- keys.

procount, 2021-02-15:
https://github.com/procount/pinn/blob/master/README_PINN.md#display-customisation
https://www.raspberrypi.org/forums/viewtopic.php?p=1607700#p1607700

closed 2021-02-15 by procount

redm123, 2021-02-16:
Ah ok, thanks, I missed that from the docs. Still leaves the discoverability in the UI... ;)