procount/pinn #487: Touch offset

dhxh, 2021-02-18:
hello
    Thank you for your work！I encountered a problem in the process of using，The touch position is not the same as the response position, there is an offset 。
[image]

procount, 2021-02-18:
The only touchscreen I have tested it with is the official 7" touchscreen.
Which touchscreen are you using?
What is your screen resolution?
Have you changed your config.txt file? What are its contents?

dhxh, 2021-02-18:
lcd 
capacitive screen 
1920x1080 HDMI

i changed config.txt
max_usb_current=1
hdmi_group=2
hdmi_mode=82
hdmi_cvt 1920 1080 60 6 0 0 0

procount, 2021-02-18:
You probably have to configure and calibrate your touch controller to match your screen resolution.
Does it work ok in Raspbian / Raspberry Pi OS?
Maybe it uses a different touch controller that I haven't got a driver for.
Have a look how it works in Raspbian. I might be able to add additional drivers, but remember that PINN does not use XWindows or a desktop, so any tools like that that are used in Raspbian will not be appropriate for PINN.

dhxh, 2021-02-18:
it work ok in Raspbian / Raspberry Pi OS Manjaro kali ....

dhxh, 2021-02-18:
It only works abnormally on pinn

dhxh, 2021-02-18:
It work ok in Raspbian / Raspberry Pi OS Manjaro Kali .....
It only works abnormally on pinn

procount, 2021-02-18:
In Rasbian, what is the output of 
`lsusb` and `lsmod`

dhxh, 2021-02-18:
I: Bus=0003 Vendor=222a Product=0001 Version=0110
N: Name="ILITEK Multi-Touch-V3000"
P: Phys=usb-0000:01:00.0-1.2/input0
S: Sysfs=/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:222A:0001.0001/input/input7
U: Uniq=
H: Handlers=mouse0 event5 
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
B: ABS=260800000000003
B: MSC=20

I: Bus=0003 Vendor=222a Product=0001 Version=0110
N: Name="ILITEK Multi-Touch-V3000 Mouse"
P: Phys=usb-0000:01:00.0-1.2/input1
S: Sysfs=/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:222A:0001.0002/input/input1
U: Uniq=
H: Handlers=mouse1 event0 js0 
B: PROP=0
B: EV=1b
B: KEY=1f0000 0 0 0 0
B: ABS=3
B: MSC=10

# xinput
Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ILITEK Multi-Touch-V3000                	id=6	[slave  pointer  (2)]
⎜   ↳ ILITEK Multi-Touch-V3000 Mouse          	id=7	[slave  pointer  (2)]
⎜   ↳ USB OPTICAL MOUSE                       	id=8	[slave  pointer  (2)]

# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0461:4e12 Primax Electronics, Ltd 
Bus 001 Device 004: ID 28a0:1285  
Bus 001 Device 003: ID 222a:0001  
Bus 001 Device 007: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# lsmod
Module                  Size  Used by
sg                     36864  0
aes_neon_blk           32768  1
crypto_simd            16384  1 aes_neon_blk
cryptd                 24576  1 crypto_simd
rfcomm                 57344  4
bnep                   28672  2
hci_uart               49152  1
btbcm                  20480  1 hci_uart
bluetooth             413696  29 hci_uart,btbcm,bnep,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
fuse                  122880  3
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 stp,garp
vc4                   253952  3
cec                    49152  1 vc4
brcmfmac              315392  0
brcmutil               20480  1 brcmfmac
drm_kms_helper        192512  3 vc4
joydev                 28672  0
v3d                    69632  3
evdev                  24576  14
gpu_sched              36864  1 v3d
sha256_generic         16384  0
libsha256              20480  1 sha256_generic
drm                   491520  10 gpu_sched,drm_kms_helper,v3d,vc4
hid_multitouch         28672  0
cfg80211              733184  1 brcmfmac
drm_panel_orientation_quirks    20480  1 drm
raspberrypi_hwmon      16384  0
rfkill                 32768  6 bluetooth,cfg80211
bcm2835_codec          49152  0
snd_soc_core          208896  1 vc4
bcm2835_isp            32768  0
videobuf2_dma_contig    20480  2 bcm2835_codec,bcm2835_isp
v4l2_mem2mem           32768  1 bcm2835_codec
bcm2835_v4l2           45056  0
bcm2835_mmal_vchiq     32768  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
videobuf2_vmalloc      20480  1 bcm2835_v4l2
snd_compress           20480  1 snd_soc_core
videobuf2_memops       16384  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         28672  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videobuf2_common       57344  5 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
snd_pcm_dmaengine      20480  1 snd_soc_core
snd_bcm2835            28672  3
videodev              274432  6 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
snd_pcm               122880  4 vc4,snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
vc_sm_cma              40960  2 bcm2835_mmal_vchiq,bcm2835_isp
mc                     53248  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
snd_timer              40960  1 snd_pcm
snd                    90112  11 snd_bcm2835,snd_timer,snd_compress,snd_soc_core,snd_pcm
syscopyarea            16384  1 drm_kms_helper
rpivid_mem             16384  0
sysfillrect            16384  1 drm_kms_helper
uio_pdrv_genirq        16384  0
sysimgblt              16384  1 drm_kms_helper
uio                    24576  1 uio_pdrv_genirq
fb_sys_fops            16384  1 drm_kms_helper
i2c_dev                20480  0
ip_tables              32768  0
x_tables               40960  1 ip_tables
ipv6                  491520  24
nf_defrag_ipv6         20480  1 ipv6

procount, 2021-02-18:
I edited your post to make it a bit more readable.

I've added a touchscreen driver which might cover your touchscreen.
Please download https://sourceforge.net/projects/pinn/files/testing/pinn-355a.zip/download.
It's a regular PINN installation but without the recovery.cmdline and config.txt files, so you should be able to unzip it over the top of an existing PINN installation. Alternatively, you can unzip it to a blank FAT32 formatted SD card, but in this case you will need to copy over your recovery.cmdline and config.txt files.

Try it out and let me know if it works for you.

dhxh, 2021-02-19:
It is not working

procount, 2021-02-19:
Do you mean it is not working at all? i.e. worse than before where at least you had an offset?

dhxh, 2021-02-19:
It is working but the touch offset problem is still

procount, 2021-02-19:
It might be useful if you could capture some debug logs for me. See https://github.com/procount/pinn/wiki/Troubleshooting.
But also, a copy of the dmesg output might be more useful.

dhxh, 2021-02-19:
[debug.txt](https://github.com/procount/pinn/files/6010147/debug.txt)
[dmesg.txt](https://github.com/procount/pinn/files/6010148/dmesg.txt)
[log.txt](https://github.com/procount/pinn/files/6010159/log.txt)

procount, 2021-02-19:
Well, it looks like the driver has been installed and is being used with your touchscreen, although your touchscreen seems to appear as a mouse and is being picked up as a joystick too.
I'm not sure how to calibrate the touchscreen to remove the offset though. Is the offset just vertical or is it horizontal as well?
Is it a constatnt offset from top left to bottom right of the screen, or is it scaled in some way?
Do you have a black border around the edge of the screen?

dhxh, 2021-02-19:
Touch offset both horizontally and vertically
Black frame on the edge of the screen
There is no black box after modifying config.txt

dhxh, 2021-02-19:
Touch calibration can use this xinput_calibrator？

procount, 2021-02-19:
No, because PINN does not support Xorg/Xwindows/Xanything.

dhxh, 2021-02-20:
thank you very much

procount, 2021-02-20:
If I find any other way to calibrate it I will post back if you're willing to test.
These things are always very difficult when I don't have the same hardware available to test it on.

BangerTech, 2021-11-02:
same problem here these offset is killing me ;-) using the RasPad 3 with a 1920x1080 Touchscreen as well. I´ve tried your testing pinn 355a but it´s not loading at all... help would be appreciated. Thanks for your great work!!!

procount, 2021-11-02:
Well, like I said before, I've only ever tried PINN with the RPF 7" touchscreen and the Pimoroni Hyperpixels and without access to any other touchscreen it's difficult to fix.

If you can provide the debug logs as dxhx provided above, it might help. Also if you can identify which touchscreen driver is being loaded when in Raspberry Pi OS, that might help too.  Can't promise anything though.

ZAIAC001, 2024-06-21:
Is there any solution?
I have same problem

procount, 2024-06-21:
@ZAIAC001 - Please start a new topic, unless you are using an RPI4 on a Raspad. 
Given your other post, I suspect you are using a Pi5 and PINN does not yet support the official touchscreen in Pi5

closed 2024-06-21 by procount