procount/pinn #488: PINN v3.5.5 Backup Bugs

annaclets, 2021-02-19:
I can confirm from tests I have made that PINN v3.5.5 indeed solves issue #447. There are, however, three minor backup bugs that I have observed:

1. Backup of lineage completes successfully, but fails during restore because in partitions.json the filesystem_type for "system" and "vendor" are specified as "ext4". Correcting these to "raw" solves it.

2. Restored backup of ubuntu 2010 results in no permission/unauthorized access of mounted drives. Applying the fix suggested by @lurch solves it: sudo setfacl -m user:pi:r-x /media/pi

3. Backup of recalbox (PINN maintained) succeeds and includes the "share" partition. Restore, however, succeeds for partitions "recalbox" and "overlay", but not for "share" which remains unformatted.

procount, 2021-02-19:
Thanks for your testing. I will look into those issues.
Which Lineage version did you test?

annaclets, 2021-02-19:
lineage17-rpi4

annaclets, 2021-02-19:
The solution to the abovementioned bug #3 would depend on PINN's ability to restore an **exfat** partition. Is this even possible?

procount, 2022-02-16:
I believe this is now resolved.

closed 2022-02-16 by procount