procount/pinn #489: twisterOS outdated

Ae3NerdGod, 2021-02-20:
currently takes at least six reboots to update twisteros to latest version
i made some tars of 1.9.7 which only takes 2 updates/reboots currently to get up to date they should be here https://sourceforge.net/projects/pinng/files/tars/twister/TwisterOSv1_9_7/
this was my command if it matters:
```
		sudo bsdtar --numeric-owner --format gnutar --one-file-system -cpf "$cwd/tars/$imgname/$label.tar" .
		echo zipping...
		xz -3 --threads=10 -v "$cwd/tars/$imgname/$label.tar"
```

mbernardi1961, 2021-04-16:
Note the Current release of Twister OS is ver 2.0.1 which requires considerable updating on the ver 1.8.4 that is available via PINN OS.

lurch, 2021-04-16:
> currently takes at least six reboots to update twisteros to latest version

Ouch, I wonder why twisteros hasn't developed a better update process? :confused:

Ae3NerdGod, 2021-04-16:
> > currently takes at least six reboots to update twisteros to latest version
> 
> Ouch, I wonder why twisteros hasn't developed a better update process? :confused: 

I don't know of any update process that doesn't either require either incremental updates or a fresh install of the newest version. 

Perhaps you could suggest your better solution to their dev team.

DVDIsDead, 2021-04-16:
>Or, maybe, someone could update twister in the PINN repo? Like I've had to on my own repo.

Ouch, I wonder why PINN hasn't developed a better repo updating process.

lurch, 2021-04-17:
Just to be clear I wasn't saying that a reboot seems unnecessary, it's just the fact that 6 separate reboots are necessary that seems a bit excessive. :shrug:  But I'm not gonna suggest a "better solution" as I've only ever used twisteros once, and that was only to help debug a PINN OS-backup problem.

AFAIK @procount has to manually convert the twisterOS disk-image into a PINN-compatible format, and unfortunately he's a very busy man at the moment. **IF** twisteros themselves produced PINN-compatible files, then @procount 's job would be limited to just editing a JSON file.

EctoOne, 2021-04-19:
I just want to mention that there is also a Lite version of TwisterOS available now. And I would love to see that added to PINN as well. Just because I'm not a huge fan of Distros with that many pre installed stuff. But no need to rush, my current RaspiOS lite with box86/wine runs fine and has replaced my windows laptop.

procount, 2021-11-01:
TwisterOS is now updated to v2.1.2