procount/pinn #490: Not possible to reload an OS with PINN

jam75jam, 2021-03-01:
HI,
I have a PINN SD with raspbian and LibreELEC;
No problem to boot with raspbian, but with LibreElec, the PI stops very fast, with displaying anything visible.
When I start and get PINN interface, "reload" possibility is not highlighted. And I cannot do anything concerning LibreELEC.
Thanks for your help.

procount, 2021-03-01:
Which version of PINN?
Full PINN or pinn-lite?
Which model of RPi?
What type of display and how is it connected?
What resolution?

There is no "Reload" feature in PINN. Do you mean Replace or Reinstall as per the following links to instructions:
https://github.com/procount/pinn/blob/master/README_PINN.md#replace-individual-oses
https://github.com/procount/pinn/blob/master/README_PINN.md#reinstall-individual-oses ?

If you have a Pi4 and are using HDMI, try using the other HDMI socket on the PI4.

jam75jam, 2021-03-02:
Thanks a lot for your answer.
Before answering your questions :
. the problem comes from copying SD cards with 'SD copier' : the original
SD works, but, on the copied one, only raspbian works, LibreELEC does not
. therefore I made, with PINN, a spécific installation on each PI (I have 6
of them). Of course, next time, I will copy SD cards on Ubuntu to try to
solve the difficulty

Now your questions:
. PINN lite v3.5.5
. RPI model: RPI 2, RPI 3, RPI 4, pi400 (all work correctly now, I "keep"
the current problem only on the pi400 because its solution coil be
interesting)
. the current display : on old NEC MultiSYnc LCD 1525M with DVI input. I
use an HDMI/DVI converter to connect the pi400, and this works very well
under raspbian, and PINN. The problem is the same with another RPI
connected to an HDMI screen or to the TV with HDMI
. there is a "Reload Repo" (highlighted) on PINN, but I was certainly
thinking to the 'Replace' (not highlighted. When I click on LibreELEC, only
'Edit', 'Password', 'Info' and 'Alias' are highlighted, the same with
'Raspbian')

The problem is the same with the other HDMI socket of the pi400.

Thanks for your help. If no solution, I will make a new PINN SD for the
pi400 and install like for the other ones, but a way to correct the problem
would be an intellectual satisfaction.

Best regards

procount, 2021-03-02:
You mean the SD copier in Raspbian?
When you copied the SD, did you choose to have new PARTUUIDs? That is probably the issue.

Go to the maintenance Menu.
Check the box next to your LIbreElec OS.
Select Fix from the menu
Tick the "Re-run partition setup.sh" option and click ok.

Hopefully LibreElec will then boot properly.

jam75jam, 2021-03-02:
yes, under raspbian, because the source is the raspbian SD.
As far as I remember, there is no option, I just specified booth SDs,
source and destination.

I made, successfully, what you did, but booting on LibreELEC just switched
off the pi400.

Thanks again

jam75jam, 2021-03-02:
... I made "re-install": again the pi400 was switched off...

jam75jam, 2021-03-02:
... on "show log", there is:
"CP437: Invalid argument
fsck.fat 3.0.26 (2014-03-07)
/dev/mmcblk0p8: 248 files..."

Best regards

procount, 2021-03-02:
Just to clarify:
1. Your original SD card with Raspbian and LibreElec worked OK on the PI400?
2. The SD card that was made using SD card Copier from Raspbian, does not boot LibreELec on the Pi400, but Raspbian is ok?

You said all the other RPi models you have work correctly now. Does that mean if you put your copied SD card into the Pi4, that it is able to run LibreElec?

It might be worth to run my troubleshooting script. See https://github.com/procount/pinn/wiki/Troubleshooting step 6 and post the output here.

jam75jam, 2021-03-02:
Thanks again,
original SD card (SD0) was done on a PI4.
The one on the PI400 is a copy.
On the other PIs, it is a 'local' card, not a copied one. They all work
correctly with raspbian and LibreELEC.

On the pi400 raspbian boots correctly. The problem I have with LibreELEC is
the one I had on all other PIs with copied cards, before making 'local'
cards. I keep the copied card on the pi400 to try to identify and solve the
problem (to play with it).

I will run your script.

Best regards

jam75jam, 2021-03-02:
...
1. I could not find '|' on my french keyboard. Therefore I could not
execute 'dmesg | less' (of course password became "rqsberry'...).
2. your point 5 : it's "cd /tmp', not 'cd \tmp'

3. I ran the commands of point 5, but on raspbian I do not find the files
in /mnt. May be it is the /mnt of LibreELEC (or PINN), not the one of
raspbian.

4. I am going to run the script 6, but it requires to take the SD card on
the PC. Is it not possible to perform all actions on the pi400 under
raspbian where I can see the RECOVERY partition with gparted, but I don't
see it on the 'easy' ways to mount it.

Thanks again for your help

procount, 2021-03-02:
The reason I asked whether your original Pi4 SD card works in the PI400, is because there are differences between the Pi4 and the Pi400. Just because it works on the PI4 does not mean it will work on the Pi400 - it requires a slightly later firmware.

So I was trying to establish if LibreELEC ever worked on the Pi400, or if it is just the cloned copy that does not work. Taking a clone of a working Pi4 card and putting it in a pi400 is not proof that there was a problem with the cloning, if it would never have worked in the first place.

Raspbian will work on all Pi models because the RPT ensure it will. It is not necessarily the same for other OSes like LIbreELEC and I have not tested it.

The '|' might be near the return key, maybe on '#' key. They are all different even on a UK keyboard. A French keyboard will make it yet more difficult!

3. You cannot run the commands from step 5 in Raspbian. They are designed to be run from PINN. The debug log is lost when you reboot or boot to another OS.  But these steps are less important for your case. Step 6 is the main one.
 
4. Yes, it is easiest to remove the SD card and copy the script across on a Windows PC.

If you prefer to do it from Raspbian you can, but you need to do the following from the command line, assuming your copy of savelog is in your home directory:

```
cd /tmp
mkdir pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
cp ~/savelog /tmp/pinn
sync
sudo umount /tmp/pinn
```
Then boot into PINN to capture the log as per the wiki instructions.
Afterwartds, you can retrieve the log using Raspbian as follows:
```
cd /tmp
mkdir pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
cp  /tmp/pinn/log ~
sync
sudo umount /tmp/pinn
```

lurch, 2021-03-02:
@procount Yeah perhaps it's the PARTUUIDs getting messed up (@jam75jam this means that on the "copied" PINN card, LibreELEC would be unable to "find" its partitions as their IDs have changed), or perhaps it might be the extended attributes thing? (I've not looked into Raspbian's SD-card-copier to see if it preserves them or not)

procount, 2021-03-02:
@lurch Yeah, I'm fairly sure it's that - I should know for sure if I get the logs. But it could also be that LibreElec hasn't got the latest firmware for the Pi400, but I doubt it - they are usually quite up to date. I just haven't tested it.

jam75jam, 2021-03-02:
thanks again.

The problem on the pi400 was the same on other pi4s where I was forced to
use a not-copied card.

I ran the commands of step 5 under PINN.

I tried everything on the keyboard for '|' (I am familiar with saxon
keyboards on french system or french keyboard on saxon systems). I will try
again.

I will try the new script.

Best regards

jam75jam, 2021-03-04:
Hi,
1. I put savelog in /tmp/pin
2. I reboot with Librelec
3. I follow point 6
4. "/mnt/savelog" leads to "savelog not found"
5. i type cd /mnt
             ./savelog
    and this leads to "/tmp/media : no such file or directory"

Of course I can create "media"..
But there is a log file I send you.

Best regards

procount, 2021-03-04:
These commands need to be executed in Raspbian.

```
cd /tmp
mkdir pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
cp ~/savelog /tmp/pinn
sync
sudo umount /tmp/pinn
```
Now reboot into PINN.
Press ctrl-ALt-F2 and login with root/raspberry (or rqsberry !)
It is here that you need to run savelog, so
```
cd /mnt
./savelog
sync
```
If you do `ls` at this point you should see the log file.
Now reboot into Raspbian and do the following:
```
cd /tmp
mkdir pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
cp  /tmp/pinn/log ~
sync
sudo umount /tmp/pinn
```

jam75jam, 2021-03-04:
... that's what I did, and I sent you the log (minus the "media" part
because of an error certainly in your script).
Best regards

procount, 2021-03-04:
Ah sorry, I misunderstood because there is no log attached.
I think instead of replying to the email, you need to go on github webapge and attach the log to this post.
https://github.com/procount/pinn/issues/490

jam75jam, 2021-03-04:
... there is a log !

---------- Forwarded message ---------
De : Jean Declaix <jean.declaix@gmail.com>
Date: jeu. 4 mars 2021 à 11:34
Subject: Re: [procount/pinn] Not possible to reload an OS with PINN (#490)
To: procount/pinn <
reply@reply.github.com>

Hi,
1. I put savelog in /tmp/pin
2. I reboot with Librelec
3. I follow point 6
4. "/mnt/savelog" leads to "savelog not found"
5. i type cd /mnt
             ./savelog
    and this leads to "/tmp/media : no such file or directory"

Of course I can create "media"..
But there is a log file I send you.

Best regards

lurch, 2021-03-04:
...but your log doesn't appear on the website at https://github.com/procount/pinn/issues/490 because GitHub-emails don't cope with attachments :wink:

jam75jam, 2021-03-04:
the log:

DEBUG
=====
Keyboard found: "/dev/input/event1"
3.5.5
Board revision is  12595504
Found recovery.rfs at "mmcblk0p1"
PINN drive: "/dev/mmcblk0"
mount: /dev/mmcblk0 is already mounted or /mnt busy
Simulate OK
CecListener thread started
Looking for  "/mnt/cec_keys.json"
Loading Mappings mappings from  "/mnt/cec_keys.json"
CecListener done initializing
mount: can't find /settings in /etc/fstab
New =  bootSelection ,  any
setWindow  "bootSelection"
setMenu  "any"
Starting  10  second timer before booting into partition 6
partnr =  6
stickynr =  800
Partition  0  =  6
Previous OS  6  at  0
Partition  1  =  8

INSTALLED_OS.JSON
=================
[
 {
  "bootable" : true,
  "description" : "A port of Debian with desktop and recommended
applications",
  "folder" : "/settings/os/Raspberry_Pi_OS_Full_(32-bit)",
  "group" : "General",
  "icon" : "/settings/os/Raspberry_Pi_OS_Full_(32-bit)/icon.png",
  "name" : "Raspberry Pi OS Full (32-bit)",
  "partitions" : [
"/dev/mmcblk0p6",
"/dev/mmcblk0p7"
  ],
  "password" : "raspberry",
  "release_date" : "2021-01-11",
  "supported_models" : [
"Pi Model",
"Pi 2",
"Pi Zero",
"Pi 3",
"Pi Compute Module 3",
"Pi 4"
  ],
  "supports_backup" : true,
  "url" : "https://www.raspbian.org/",
  "username" : "pi"
 },
 {
  "bootable" : true,
  "description" : "LibreELEC is a fast and user-friendly Kodi Entertainment
Center distribution.",
  "folder" : "/settings/os/LibreELEC_RPi4",
  "group" : "Media",
  "icon" : "/settings/os/LibreELEC_RPi4/icon.png",
  "name" : "LibreELEC_RPi4",
  "partitions" : [
"/dev/mmcblk0p8",
"/dev/mmcblk0p9"
  ],
  "password" : "libreelec",
  "release_date" : "2020-10-24",
  "supported_models" : [
[... 227 lines cut ...]
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram15: 32 MiB, 33554432 bytes, 65536 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/mmcblk0: 476.7 GiB, 511868665856 bytes, 999743488 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xf9cca749

Device          Boot     Start       End   Sectors   Size Id Type
/dev/mmcblk0p1            8192    137215    129024    63M  c W95 FAT32 (LBA)
/dev/mmcblk0p2          137216 999743487 999606272 476.7G  5 Extended
/dev/mmcblk0p5          139264    204797     65534    32M 83 Linux
/dev/mmcblk0p6          204800    729085    524286   256M  c W95 FAT32 (LBA)
/dev/mmcblk0p7          729088  38797309  38068222  18.2G 83 Linux
/dev/mmcblk0p8        38797312  39845885   1048574   512M  c W95 FAT32 (LBA)
/dev/mmcblk0p9        39845888 139675647  99829760  47.6G 83 Linux
/dev/mmcblk0p10      139677696 999743487 860065792 410.1G 83 Linux

MEDIA
=====

jam75jam, 2021-03-10:
Hi,
I "played" a little with the current PINN card, before creating a new one 'locally'.
I tried (again) to replace or reinstall LibreELEC : same result, booting on LibreELEC leads to a switch off of the pi.
Then I replaced LibreELEC by Pi OS : it works ! Both Pi OS can be booted. 
But... I cannot replace the second Pi OIS by LibreELEC !
Best regards

procount, 2021-03-10:
Sorry, I've had no time to look into this further as my day job is keeping me very busy :(
Are you able to post cmdline.txt from the boot partition of LibreElec?
Maybe even /etc/fstab from its root partition might be useful.
You could also try selecting your installed LibreElec in PINN's maintenance Menu, select Fix from the menu. and choose to "re-run partition_setup script". Then see if it will boot after that.

jam75jam, 2021-03-10:
NO problem, you already spent so much time.
I still will "play" a little and build a new SD card. The more I "play", the more I appreciate PINN.
Best regards
PS : I also made "Fix"... I will see if it is in trhe same context as the one you propose

jam75jam, 2021-03-10:
I tried to follow your tuto for changing an OS on the card. I specified the second Pi OS  to be replaced, but the first one (the 'good' one) was replaced !
I therefore built a new PINN card on the pi400 itself: LibreELEC still switches off the pi.
I made a search and there is effecticvely a problem with the pi400 and LibreELEC: 

https://forum.libreelec.tv/thread/22946-issue-with-raspberry-pi-400/

Best regards

procount, 2021-03-11:
So the real problem is one I suggested here -> https://github.com/procount/pinn/issues/490#issuecomment-789055514 but I didn't really get a definitive answer whether LibreElec had ever worked on the Pi400 before being cloned or otherwise.

So you either have to wait until a new version of LibreElec is made available that is suitable for the Pi400, or try to upgrade the LibreElec firmware to the latest version from Raspios yourself.

Well at least you have a cause for your problem now.

You also need to be careful when replacing OSes with the same name that you replace the right one. You can make it easier to identify them by using an alias name for one or both of them.

jam75jam, 2021-03-11:
> So the real problem is one I suggested here -> #490 (comment)

Thanks again.
LibreELEC with PINN works on my PI2, PI3, PI4 (I did not try on PI0...). I
do not really need LibreELEC on the pi400 (on which I have the latest
firmware).
Therefore I install Pi OS only on the pi400.
I tried to understand the point with LibreELEC on pi400 but it looks
tricky. Therefore I give up.
Best regards

procount, 2021-03-11:
OK. So now we know it is nothing you have done, and nothing wrong with PINN. The problem is LibreELEC does not have the latest Pi firmware for the Pi400. You may just have to wait until LIbreELEC is updated to run it on your pi400.
So I think this issue can be closed now.

jam75jam, 2021-03-11:
exactly.
Thanks a lot.

closed 2021-04-07 by procount