procount/pinn #491: "Error Parsing repolist.json" and "Please Wait while PINN initializes"

Wizard451345, 2021-03-06:
I get these errors when I log in into PINN, though I can still ignore it and boot and use all the menu. Why do I get these messages? Also the OS's from the repolist are not showing, I can only see Raspberry Pi OS and LibreElec. Using PINN v3.5.5 if that helps idk, im a noob :)

procount, 2021-03-07:
Have you connected PINN to the internet via ethernet or wifi?
"Please wait while PINN initializes" is not an error indication. It is just waiting for an internet connection.
"Error parsing repo_list.json" is an error and suggests it is not downloading form the internet correctly.

Wizard451345, 2021-03-07:
> Have you connected PINN to the internet via ethernet or wifi?
> "Please wait while PINN initializes" is not an error indication. It is just waiting for an internet connection.
> "Error parsing repo_list.json" is an error and suggests it is not downloading form the internet correctly.

Yes, PINN is connected to the the internet via Wi-Fi and I am able to connect to it via SSH and VNC remotely via my phone. :/

Wizard451345, 2021-03-12:
[image]

lurch, 2021-03-12:
Wow, world's biggest phone? :wink: :laughing:

Wizard451345, 2021-03-12:
> Wow, world's biggest phone? :wink: :laughing: 
> 
> EDIT: More seriously, if you're able to connect to PINN from your phone over WiFi, it just shows that they're both successfully connected to the same local-network, it doesn't necessarily mean that PINN is able to connect to the internet if it e.g. has DNS or routing problems? :shrug: 

Browser Works fine:
[image]

lurch, 2021-03-12:
Hmmm, how very strange! :confused: 

Are you able to view `http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json` (which is specified [here](https://github.com/procount/pinn/blob/master/recovery/cmdline.txt) ) in PINN's web-browser?

Wizard451345, 2021-03-12:
> Hmmm, how very strange! :confused: 
> 
> Are you able to view `http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json` (which is specified [here](https://github.com/procount/pinn/blob/master/recovery/cmdline.txt) ) in PINN's web-browser?

Yes, I'm in!
[image]

lurch, 2021-03-12:
That looks fine. Is PINN itself still not working?

Wizard451345, 2021-03-12:
No :/
Probably the best choice is to reload it with pinn again. Will take while as I have RPI OS on it and have to reinstall everything. Will do it from usb.

Wizard451345, 2021-03-12:
... I did nothing for over a month and it decided to work now...?

wat
[image]

closed 2021-03-13 by Wizard451345