procount/pinn #494: Install different OS to eMMC and to USB

NumberOneGit, 2021-03-17:
Certain OS's still don't function properly from USB (Lineage for example).  Is there any way to have PINN write Lineage to the eMMC, while writing Raspbian, LibreElec, etc to the USB drive all within PINN?

procount, 2021-03-17:
Lineage works fine from the USB, but you must follow the additional instructions in the lineage FAQ  from Konstakang.

NumberOneGit, 2021-03-18:
It was a more generalized question.  I had seen in the same section of the guide that you are referring to that Konstakang described USB boot as an experimental feature and unstable in some ways but I will try it out.

procount, 2021-03-18:
So the generalized answer is no, it is not possible to split OSes between different devices at the moment. NOOBS was not designed for that. It's a similar problem to the question "How can I add a new OS to an already installed PINN system"
You could hack it manually if you know your way around the config files, but the boot partition must always be on the same device that PINN booted on, because the firmware can only boot to a partition on that device as far as I am aware.

closed 2021-03-19 by NumberOneGit