procount/pinn #497: "reinstall" for PINN should forcefully reinstall instead of just checking for updates.

jharris1993, 2021-03-19:
Issue:

Attempting to "reinstall" PINN does ***NOT*** actually re-install PINN, it simply checks for a new version and, if a new version is available, it offers to install it.

If, for whatever reason, the user wishes to absolutely re-install PINN, he has to dink around with files in the settings folder.

This is not intuitive behavior.  "reinstall" should do precisely that.

Reinstall should, at a minimum, reinstall the current version.  Ideally it would check for an update and then offer the user the option of
1.  Updating (assuming an update exists)
2.  Reinstalling the same version as before
3. Cancel

procount, 2021-03-19:
I agree it is not intuitive.
The purpose is NOT to reinstall PINN, but to manually check for a self-update in the case where you have set the 'noupgrade' option or (possibly) when it was told to ignore the latest update.
As it is not possible to  "reinstall" pinn, I overloaded that feature with the self-update function, as the menus are getting pretty full and I didn't want to create another button, or go to a 4th menu. I thought the functionality was close enough and it is described in the readme.

jharris1993, 2021-03-20:
Unfortunately, I am guessing that the way PINN “ignores” updates is to set the version number in the build info file to the current rev being ignored.

 

This would provide both the effects described – that of ignoring the current version, (because it believes it’s already current), and remaining sensitive to the NEXT version.

 

Perhaps the “reinstall” process – when applied to PINN – can either correct the build number or obliterate that file so that PINN becomes eligible for an “upgrade”.  As it currently stands, Reinstall does nothing.

 

-----------------------------------------------------------------

Jim "JR" Harris 

Some see things as they are and ask "Why?" 
I dream things that never were and ask "Why Not".
Robert F. Kennedy 

 

There are three basic types, Mr. Pizer: the Wills, the Won'ts, and the Can'ts.

The Wills accomplish everything, the Won'ts oppose everything, and the Can'ts won't try anything.

V.I.N.Cent, from the Walt Disney film “The Black Hole”

 

“Impossible” is only found in the dictionary of a fool. 
Old Chinese Proverb

procount, 2021-03-20:
I was mis-remembering. Yes, it won't work when you have chosen to "ignore" the next version. But it does work if you chose the noupgrade option, or just chose not to upgrade on the last occassion.

Removing the BUILD-DATA file would also fix the case where you have ignored it, so I'll look to include that.

lurch, 2021-03-21:
> If, for whatever reason, the user wishes to absolutely re-install PINN, he has to dink around with files in the settings folder.

Or he could just erase / format his entire SD card, and start again from scratch? :wink: 
With how simple[1] PINN is, why would somebody want to "absolutely re-install" it?

[1] Well, not simple when compared to NOOBS or RPi-Imager, but very simple when compared to a full-blown OS :laughing:

jharris1993, 2021-03-21:
What if:

1.      The user has a number, (>2), of heavily customized O/S’s on his device which may or may not be an SD card.

2.      The SD card is inside a device and is difficult to get to without risk of damage.  (Mine requires partial disassembly of the device and use of a long, thin, pair of needle-nose pliers to get to the SD card.)  Even if the user has a SSD with a USB interface, removal and replacement may not be trivial.

3.      The user suspects that, (for whatever reason), the PINN installation might have been damaged somehow.

4.      Or, he chose to “ignore” an update and now wishes to install it.

 

At present the user is left with few choices:

1.      Completely back-up every operating system on the device, (in my case the device is large and I have four or five O/S’s on it), then wipe and re-install, THEN restore all the backups and hope they work.
This carries the very real risk that the backup may be corrupt and unusable, or may not have the permissions restored properly.  Result:  The user is thoroughly borked.
If the user’s device is buried inside a piece of equipment like a robot, instrumentation, or set-top box, removing the media may damage it.  (In my case, it is almost certain to damage it.)

2.      Physically remove the device, carry it to a Linux box and manually move the files to the device.
(I strongly suspect that unpacking the archive on a Windows-10 box and moving the files from there may not preserve whatever attributes they may have, or may not unpack them properly.)

3.      Destroy the BUILD-DATA file somehow.  (Getting to it is not trivial and requires some advanced knowledge which we cannot guarantee everyone has.)

 

--- OR ---

 

4.      Have a way to reinstall from the GUI on command.

 

The main point is that it is unwise to assume that the user will have a trivially complex installation.  Even if the installation isn’t heavily customized, the issues noted above still apply and a non-expert user will have considerable trouble.

 

·        There is one major “gotcha”.  If the PINN installation IS heavily customized, the customized files will need to be preserved.  AFAIK, this would be the config.txt and pinn_init.sh files, if they exist.

procount, 2021-03-21:
On point 2, noobs/PINN was designed especially to be able to unzip the files on Windows and copy them to the Fat32 partition.

I use a 4th option: ssh into PINN and scp copy all the updated files onto the PINN partition then reboot. I have a script that I run to do it automatically.

lurch, 2021-03-21:
On point 3, NOOBS was designed specifically to be "impossible" to be damaged. All of the NOOBS files live inside a `recovery.rfs` file which gets mounted as a read-only ramfs image. All of NOOBS's user-modifiable state gets written to the settings partition, and the first partition (where `recovery.elf` and `recovery.rfs` live) **never** gets written to.
(I don't know the intricate details of PINN to know if this is still true for PINN - although I guess it can't be true if PINN has an auto-update feature! :laughing: )

@procount Instead of relying on the `BUILD-DATA` file to determine which version of PINN the user is running (which, from the discussions in this thread, sounds a bit error-prone), _perhaps_ it'd be more reliable to determine which version a user is running by calculating file-checksums? (although you'd then have the overhead of maintaining a JSON file on the server which maps file-hash -> PINN version, although I guess you could also automate that with a script)

jharris1993, 2021-03-21:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAADAAAAAwCAYAAABXAvmHAAAABHNCSVQICAgIfAhkiAAAAB9JREFUaIHtwQEBAAAAgiD/r25IQAEAAAAAAAAAAC8GJDAAAY7rwGcAAAAASUVORK5CYII=###24x24:true###][Tracking image][https://github.com/notifications/beacon/AC36Q5D4TGOPB2TW6WX2B6LTEX4GXA5CNFSM4ZPVD2PKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOF7S4D7A.gif]

Even easier: allow unconditional backups.

procount, 2021-03-21:
> Instead of relying on the BUILD-DATA file to determine which version of PINN the user is running (which, from the discussions in this thread, sounds a bit error-prone).....

The use of BUILD-DATA works as intended, it's just that once a user ignores a version, they have to wait for the next version to become available (unless they want to delete the ignore file themselves). I mistakenly thought manually doing an upgrade would force it, but it doesn't. It's an easy addition to make though for the next version.

procount, 2021-04-07:
Included in next version

closed 2021-04-07 by procount