procount/pinn #501: PINN won't boot on my Pi 400

KentQuintel, 2021-03-30:
Hello,
After many attempts with numerous SD Cards and USB drives, I have yet to have any luck getting Pinn OS to boot on my Pi 400. All I'm getting is a quick nanosecond flash of the initial pi boot text and then a blank screen. The Pi then disengages from the display ( its green light staying lit).

I've formatted the SD Cards and USB drives using both Pi Imager and SD Card Formatter before each attempt.

The drives I own are:
- 16gb Sandisk Edge SD card
- (2)32gb Silicon Power 3D Nano SD card
- 64gb SandIsk Cruzer USB stick
- Samsung BAR 32gb USB(3.1)stick.

Appreciate greatly your time and work.
And a thank you to the community as well!

procount, 2021-03-30:
Does it work with a plain Raspbian / raspberry PiOS image?
How new is your pi400?
What version of PINN are you using?

KentQuintel, 2021-03-30:
Yes, plain PiOS works.
My Pi400 came 3/24/21, so about a month old.
I have tried the current (3.5.5) versions of PINN and PINN-Lite

Terminal Output;
Model    :   Raspberry Pi 400 Rev 1.0

procount, 2021-03-30:
Well, it works fine on my Pi400. The fact that Raspberry Pi OS works on yours suggests there is nothing wrong with your PI400, so probably something to do with the way you are installing it.
Is this the first time you are installing PINN to any Raspberry PI?

I suggest you download the image version of PINN from https://sourceforge.net/projects/pinn/files/Etcher/pinn-lite.img.zip/download and install it to your 16GB SD card using RPi Imager. It is the most foolproof way to install it.

lurch, 2021-03-30:
Have you tried both HDMI ports? Do you have any alternative displays you can try using? :man_shrugging:

KentQuintel, 2021-03-31:
> Do you have any alternative displays you can try using? 

@lurch  It was the display(!) Tried another display to find PINN has been booting the entire time (of course). I was able to run a few other OSes on the display so I overlooked that as a potential problem.

@procount Thanks a million for your brilliant work and timely response. 
Apologies for any unnecessary trouble! 
Please, keep up the amazing work!

Thanks again