procount/pinn #503: [Feature request] Add cmdline option for SSD over provisioning

BuongiornoTexas, 2021-04-02:
Hi @procount, 

Thanks very much for a incredibly useful multiboot environment!

I'm learning my way around the raspberry pi at the moment, and I'm in the happy position of having an SSD that I can use to try out multiple operating systems quickly (so I'm very much enjoying a return to Linux after many, many years away). 

The one issue I have run into is that it is not straight forward to allocate a chunk at the end of the SSD for over-provisioning. At the moment, I'm going through a fairly clunky process of creating 6 equal sized OS partitions (mainly project spaces), then deleting the partitions for the last OS and shrinking the PINN extended partition and editing the PINN configuration files to match - this leaves about 16% available for firmware overprovisioning. 

It'd be nice if you were able to add a cmdline option to reserve a percentage of the disk space or allocate a specific amount of the storage volume (X MiB/GiB etc). 

I realise this is probably a fairly low priority while the majority of Pi's boot off SD card only, so I'm cheerful to wait until it is worth addressing for a larger proportion of users (probably as SSDs become more mainstream for Pis).  But I thought it was worth raising in case it wasn't already on your development road map.

If this is either a low priority or a won't implement, I'm happy to pull together a wiki page describing the process I've used.

lurch, 2021-04-02:
> this leaves about 16% available for firmware overprovisioning.

It took me while to realise what you were asking for... until I realised you were talking about the SSD's firmware and not the Pi's firmware :laughing:

Is it actually necessary / recommended to "manually" overprovision an SSD though? I always's thought the SSD did that automatically? :shrug: 
(I just did a google search, and [this](https://superuser.com/questions/944913/over-provisioning-an-ssd-does-it-still-hold/944915) is one of the top results, with most of the answers from around 6 years ago!)

> At the moment, I'm going through a fairly clunky process of creating 6 equal sized OS partitions (mainly project spaces), then deleting the partitions for the last OS and shrinking the PINN extended partition and editing the PINN configuration files to match

There's been a few other requests for more advanced "partitioning control" / "custom partitioning" in PINN, but I think in the meantime your current workaround is probably the best option. (if you still believe it to be necessary)

BuongiornoTexas, 2021-04-02:
> until I realised you were talking about the SSD's firmware

My bad! I shouldn't post after a long day in the garden and bit of wine based recovery. 

> Is it actually necessary / recommended to "manually" overprovision an SSD though? I always's thought the SSD did that automatically? 
>  (if you still believe it to be necessary)

Most SSDs do include some built in manufacturer over-provisioning (e.g. 7% for Samsung SSDs). From the reading I've done, manufacturers still broadly state that additional user/manual over-provisioning will extend SSD life. And Samsung is still including an over-provisioning tool as part of its Windows utility. On top of which for smaller drives (such as the one I'm using for my pi), the manufacturer over-provision is small in absolute terms. 

So, necessary? - No. 

But, I've had the lifetime of one SSD trashed by a write amplification issue (admittedly a firmware problem, but in the same space), and I very much like my hardware to last as long as possible. So for me, giving up a modest amount of storage on my SSD in exchange for longer life and improved reliability seems like a very worthwhile trade off. 

Bottom line: I can work around it for myself, but if the dev(s) are considering advanced partitioning schemes, this would be a good one to include and should be relatively easy to implement. 

Side note - I've read an awful lot of posts where people say over-provisioning is no longer needed/automatically handled by the firmware by using any empty blocks it can find/other reasons OP is no longer needed. I haven't found manufacturer recommendations backing these statements. At most, they say manual over-provisioning is optional but will improve SSD life.

lurch, 2021-04-02:
> a long day in the garden and bit of wine based recovery.

Sounds like a great plan! :seedling: :wine_glass: 

> I can work around it for myself, but if the dev(s) are considering advanced partitioning schemes

I only mentioned there's been several requests for it, I've no idea if @procount is actually considering it or not... :wink: (my assumption would be not, since it seems like fairly niche advanced-usage, and therefore a low-priority task? :shrug: )

BuongiornoTexas, 2021-04-02:
> I only mentioned there's been several requests for it,

Fair enough - I guess we wait on @procount  weighing in.

procount, 2021-04-02:
I'll take a look. It sounds quite a simple thing to do, so it could make it into a future version, but if it has more dependencies, such as an assumption that the extended partition ends at the end of the drive, then it would require more work.

BuongiornoTexas, 2021-04-02:
The manual approach is to leave a chunk of storage unallocated at the end of the drive, and I believe the manufacturer utilities do exactly the same. So a disk layout would be:

* /dev/sdX1 - FAT boot partition.
* /dev/sdX2 - Extended partition containing:
  * /dev/sdX5 - OS logical partition
  * ...
  * /dev/sdXN - OS logical partition
* Unallocated space reserved for over-provisioning.

procount, 2021-07-01:
This is now implemented in v3.6 Add provision=nnn to reserve nnn MB at the end of the drive outside of the extended partition.

closed 2021-07-01 by procount

BuongiornoTexas, 2021-07-02:
Thanks for this guys - much appreciated!