procount/pinn #504: Can not boot into Raspberry Pi OS from USB using PINN

eeyatahn, 2021-04-02:
Aloha Guys.... 

The multi OS setup we worked on a few months ago has been working fine and I have even remade the whole thing using a 256 GB SD Card up from a 128 GB SD Card using the new directions in the Wiki. All has been golden ... UNTIL.

I decided that I would purchase an SSD and Enclosure and set up a USB boot for my RPI 4B and PINN setup. I refereed to several web sites to get the necessary instructions and help to set this up correctly. 

I updated my Bootloader EPROM to the most recent version released on 3-18-21. Copied over my 256 GB SD Card to a new 256 GB Samsung 850 EVO SSD using the SD Card Copier within Raspbian. Some time later after a successful copy I removed the SD card and booted from the USB SSD.

Upon booting, PINN loaded and displayed my installed OSes ( Raspberry Pi OS and LebreELEC Kodi). I selected LebreELEC Kodi and booted into Kodi with no issues at all. Everything was there and functional. I rebooted into PINN and selected Raspberry Pi OS then the problems started.

Upon booting from PINN into Raspbian all started good. The RPi rebooted and the rainbow screen appeared. Then I was greeted with this message:

` mmc1: Controller never released Inhibit bit(s)`

and the boot operation came to a GRINDING halt.

Thinking that I may have done something wrong I recopied the SD Card using the SD Card Copier from Raspbian again. Rebooted and got the same thing. I even tried using the SD Copier in PINN with the same results.

I read through the internet and found that this error has been reported before when booting off of USB but all of those posts were before the RPi 4B was released. There was nothing current that I could reference. So I tried my own trouble shooting attempt.

I tried using  a single SD card with the copy of Raspbian that I had used to make part of the Multi-OS PINN from.... Copied it over to the SSD and booted with no problem. Next I tried using the Multi-OS Archive I had made and reinstalled both OSes (Raspbian and Kodi) into PINN then copied it over to the SSD and it booted. Now this is the same image used to when I made the 256 GB SD Card that I have been using since I created this PINN setup about a month ago. It booted fine, no error.

The 256 GB SD Card I have been using for the last month has the exact same thing on it as the Multi-OS Archive I used to create it. 

Why am I able to boot from USB if I use a NEWLY created PINN Multi-OS SD Card or the single OS SD card but NOT if I Copy the SD Card I have been using.

@procount @lurch I do not know if PINN has anything to do with this other than it is the only common factor I can find. Any help would be appreciated.

procount, 2021-04-02:
Could you run step 6 of the Troubleshooting wiki page. https://github.com/procount/pinn/wiki/Troubleshooting on the 256GB Evo SSD that fails and post the output?

lurch, 2021-04-02:
> Why am I able to boot from USB if I use a NEWLY created PINN Multi-OS on the SSD or the copied single OS SD card but NOT if I Copy the SD Card I have been using.

When you install an OS using PINN, it writes a settings file telling it which partition(s) the OS has been installed to.

So when running PINN from an SD card, and installing an OS to the SD card, it might record that the OS is installed on `/dev/mmcblk0p6` and  `/dev/mmcblk0p7`.

When running PINN from a USB SSD, and installing an OS to the SSD, it might record that the OS is installed on `/dev/sda6` and  `/dev/sda7`.

So when you copy a PINN SD-card onto a USB SSD, and then try booting from this USB SSD, PINN will still be looking for the OS on `/dev/mmcblk0pN` and not `/dev/sdaN`.

I dunno of the top of my head the easiest way of fixing things, but hopefully @procount will pop by at some point :slightly_smiling_face:

procount, 2021-04-02:
Just pipped you to it! 😄 
Yes, hopefully the log will identify this.

@eeyatahn - After posting the log, (since you have a copy of your OSes on your 256GB SD card for a bcakup) you could try re-running the partition_setup.sh script on your SSD for any OS that fails. You will find it on the Fixup button in the Maintenance Menu.

eeyatahn, 2021-04-02:
> Just pipped you to it! 😄
> Yes, hopefully the log will identify this.
>
> @eeyatahn - After posting the log, (since you have a copy of your OSes on your 256GB SD card for a bcakup) you could try [...]

Ok will take a couple of hours to copy it over again.... So I simply substitue the SSD for every place it says SD Card in the trouble shooting instructions... ok

Thanks guys.... 😄

eeyatahn, 2021-04-02:
Oh one more thing.... Is there a way to reinstall only one OS of a Multi-OS PINN setup without wiping the entire SSD? I tried using the "Reinstall" button on the Maintenance page but when I try to use the install archive I get a error "destination too small for required partition size", which I know is impossible as the OS is on the drive in the first place.
Or can the archive only be used to do the initial install and can not be used for a reinstall.

eeyatahn, 2021-04-03:
> Try to recreate your issue. After it has occurred, press Ctrl-Alt-F2 to get to the recovery shell.

I used Clone from PINN Archival menu to copy my SD Card over to the SSD. Shutdown the RPi removed the SD card and restarted from USB.
PINN started, I selected Rasbperry Pi OS, saw the Rainbow screen and got my least favorite message to date:

`mmc1: Controller never released inhibit bit(s)`

and stopped at a flashing cursor not a prompt.

I tried "Ctrl-Alt-F2".... Nothing 😞 . I can type control characters and text on the screen but it is dead no response to anything but a reset "Ctrl-Alt-Del"

Going to Boot from the SSD into PINN and try executing partition_setup.sh and see what happens.....

(.... 🎶  Jeopardy Song Plays 🎵 .....)

Ok will I just ran partition_setup.sh script on the Raspberry Pi OS (32-bit) (INSTALLED). Here is what I got.....

(excuse any typos I typed this on my PC from the RPi screen) <but I think I got them all 😄 
```
Checking for partition_setp.sh scripts..
Executing /settings/os/Raspberry_Pi_OS_Full_(32-bit)/partition_setup.sh part1=PARTUUID=00000001-06 id1=UUID= part2-PARTUUID=00000001-07 id2=UUID=
Raspberry Pi OS Full (32-bit): Error executing partition setup scrit
+ [ -z PARTUUID=00000001-06 ]
+ [ -z PARTUUID=00000001-07 ]
+ mkdir -p /tmp/1 /tmp/2
+ mount PARTUUID=00000001-06 /tmp/1
mount:can't find PARTUUID=00000001-06

Raspberry Pi OS Full (32-bit) partitions_setup.sh script error

DONE - some setup seript errors detected
```

In the bottom "Download space" window. Destination drive: sdb: Mini (USB ThumbDrive) is shown, no other selections are available.

eeyatahn, 2021-04-03:
> When you install an OS using PINN, it writes a settings file telling it which partition(s) the OS has been installed to.
>
> So when running PINN from an SD card, and installing an OS to the SD card, [...]

@lurch....

In Raspberry Pi OS the Copy SD Card utility has a check box to write a new a UUID code for the copy. I tried BOTH selections with and without the NEW UUID CODE selected. No change.

By telling the SD Card Copier to assign NEW UUID codes to the partitions would that not alleviate this?

procount, 2021-04-03:
It looks like something is wrong with your partition IDs. Running partition setup may not fix these after a clone from SD to USB/SSD.
I may have to write a special script to fix this scenario.

The troubleshooting page is for identifying things in PINN so it is only intended to be run from PINN.
Only step 6 is needed, so copy the savelog script to your SS, boot PINN, press ctrl-alt-f2 from pinn's recovery menu and follow instructions from there.

It may not fully work with broken partuuids but post whatever it produces.

eeyatahn, 2021-04-03:
> The troubleshooting page is for identifying things in PINN so it is only intended to be run from PINN.
> Only step 6 is needed, so copy the savelog script to your SS, boot PINN, press ctrl-alt-f2 from [...]

Ok I could not get it to work following the instruction in step 6. I was able to look at the code in savelog and found the file it writes called log. Posting that, hope you should find what you are looking for in there.
[log.zip](https://github.com/procount/pinn/files/6252135/log.zip)

procount, 2021-04-03:
The log contains a copy of your installed_os.json that tells PINN where the partitions for each OS are located.
This shows that these are still left pointing at the/dev/mmcblk0pX partitions which is where they were on your SD card.

Cloning an SD card to a USB/SSD device will only work without modification IFF the partitions are referenced by something more constant, like the PARTUUID of the partitions, which are supposed to be unique.

So after cloning an SD card for use in a USB device, it is necessary to update any references to partitions.
This starts with /settings/installed_os.json which tells PINN where the boot partition is (for each OS).
Once this has been loaded, the OS boot partition takes over. It needs to locate the root partition, which is normally referenced from the cmdline.txt file, so this also need updating.
Many Linux OSes have a file on the root partition called /etc/fstab that causes the OS partitions to be mounted, so this also needs updating.

The re-run partition script in PINN's fixup menu was designed to fix up some of these and other OS-specific locations in case they get damaged during an upgrade or similar. However, it was not designed for fixing device changes, whcih is why it didn't work in your troubleshooting above. I will need to add another feature to do this.

In the meantime, if you update the installed_os.json, the rerun partition script *should* be able to fix up the other references. But please check and update anything else that it doesn't manage to fix.

eeyatahn, 2021-04-03:
> In the meantime, if you update the installed_os.json, the rerun partition script should be able to fix up the other references. But please check and update anything else that it doesn't manage to fix.

You should have mentioned where to find Installed_os.json was. But I had been doing my own trouble shooting and was looking at the partitions and files using Ubuntu and I just happen to see the "SETTINGS" partition. Where I remebered seeing "installed_os.json".

I edited the .json and changed 

`"/dev/mmcblk0p6",
"/dev/mmcblk0p7"`

to 

`/dev/sda6",
"/dev/sda7"`

Ran partition_setup.json again, this time is succeded with no errors.
Rebooted and started Raspberry Pi OS from USB SSD and it works. 

Raspberry Pi OS boots with no issues at all... LebreELEC Kodi also works even though I never had any problem getting that one to boot at all.

Thank you

procount, 2021-04-03:
> Oh one more thing.... Is there a way to reinstall only one OS of a Multi-OS PINN setup without wiping the entire SSD? I tried using the "Reinstall" button on the Maintenance page but when I try to [...]

Reinstall uses the name of the installed OS to find the original archive and to reinstall it.
Replace allows you to choose another OS to occupy that space instead.
In both cases, the existing partitions are re-used. They are not deleted and recreated. Therefore the partitions of the new OS must be the same size or smaller that what was originally installed. The number of partitions must be the same too.

eeyatahn, 2021-04-03:
> Reinstall uses the name of the installed OS to find the original archive and to reinstall it.
> Replace allows you to choose another OS to occupy that space instead.
> In both cases, the existing [...]

Then I don't understand why it did not work. I used the same USB stick with the same archive that I used when I installed the OS in the first place. Nothing changed. Reinstall should have worked but for some reason I got an error that the partition I was trying to reinstall into was too small and it refused to install. 

For right now that is not something I need to deal with now that I have everything working. At least untill I need to reinstall an OS. But I'll deal with that when needed.

procount, 2021-04-03:
But you tried to reinstall whilst your installed_os.json was broken, so the existing partitions could not be found.
It should work again now you have fixed that.

eeyatahn, 2021-04-03:
Ok I understand... It was looking for a partition on the SD card, only there was no SD card.

lurch, 2021-04-03:
> It was looking for a partition on the SD card, only there was no SD card.

Maybe something else for PINN to check, and provide a more helpful error message than "partition too small" ? I guess there's probably gonna be a whole **bunch** of edge cases when you're mixing SD and USB install / restore like this? :shrug:

procount, 2021-04-03:
@Lurch - Yeah, this was all done before USB & SSDs started to become popular, so I guess there will be a lot of people migrating SD cards to USB. A lot of edge cases to consider!

eeyatahn, 2021-04-03:
@procount....

One thing is strange though...

It was only Raspberry Pi OS that was effected by this. 

I also had LebreELEC Kodi install and the Log showed the partitions ID as:

`"partitions" : [
"/dev/mmcblk0p8",
"/dev/mmcblk0p9"
  ],`

For LebreELEC Kode.....

But I did not get the error during boot, and Kodi booted with no problem. I did change the Partition ID when I changed it for the Raspberry Pi OS Partition.

Question....
Why was Raspberry Pi OS and not LebreELEC effected by the Partition ID being Wrong?

procount, 2021-04-03:
When NOOBS or PINN does its double-boot to load in the OS, it can only load its boot partition from the same device as PINN was booted from, so to load the boot partition, only the partition number is required (e.g. 8 in the case of LibreElec) and this is the same between /dev/sda8 and /dev/mmcblk0p8. It usually gets stuck when the boot partition tries to load the root partition. I haven't looked at LibreELEC for a while, but maybe it does this another way, or doesn't need to do it at all if it is self-contained. But maybe it will have trouble finding its storage location. 🤷‍♂️

lurch, 2021-04-04:
There are several ways for a OS (booted from the "boot" partition) to locate its "root" partition - some do it by partition device name, some do it by UUID, and some do it by filesystem label. I have no idea which OSes use which method, but that explains why some "work" when others don't :slightly_smiling_face:  (and of course for those OSes that load entirely from an initrd on the boot partition (similar to the way PINN does), there may not even _be_ a "root" filesystem!)

P.S. It's LibreELEC, not LebreELEC :wink:

procount, 2021-07-01:
**Enhancement:** Allow partition references to be updated when SD card is cloned to USB or vice versa

iv09, 2022-02-23:
I seem to have solved that issue by running `sudo rpi-update`. This updates the Raspberry's firmware and adds more supported USB SSD enclosures. However, the installed firmware is a Beta version, you may have issues.