procount/pinn #505: Update Retropie 4.6 to 4.7.1

DerangedPineapple, 2021-04-04:
PINN has Retropie 4.6 available to be installed but that version cannot be used with usb ssd installation according retropie docs as they added usb boot support in version 4.7

Also version 4.6 is almost year old (ancient) so it would be nice to have newest version available and to be able to use usb boot.

My plan is/was to use libreELEC 10 (currently using nightly after manually updating it after installing 9.2.6) and newest retropie which is 4.7.1 booting them from usb ssd but currently I cannot really progress any further.

I assume I could follow your wiki guide to backup OS's and put them manually, I just need to figure out how to get bsdtar installed and how and where to wget  those proper *.json files for libreELEC 10 and Retropie 4.7.1

DerangedPineapple, 2021-04-24:
I wonder if this issue could be worked on in the near future. I don't want any time table but just an hint if something might happen regarding getting retropie updated.

I have switched to Recalbox for now but would really like to hop back on retropie.

procount, 2021-04-24:
I have converted it but not tested it.
If you would like to test it for me, I can make it available for you and if it is ok I can release it generally.

DerangedPineapple, 2021-04-24:
I would definitely like to test it out if possible, I just installed my pi 4 from the scratch with LE and Recalbox so I can test with different usb ssd.

procount, 2021-04-24:
Try adding `alt_image_source="http://raw.githubusercontent.com/procount/pinn-os/master/os/os_retropie.json"` to recovery.cmdline, or add it to the PINN config-edit options

DerangedPineapple, 2021-04-25:
Ok I have tested it and in short (I do apologise for the unnecessary meme image but I just could not resist):
[image]

I installed LibreELEC 9.2.6 and Retropie 4.7.1 to usb ssd successfully without any issues. Only the thumbnail is missing from OS listing under Games tab.

Next I try to update LE from 9.2.6 to 9.95.2 (LE 10 Beta2) even if the manual update is not supported not recommended. I just need to utilize both hdmi ports (one for picture and other one for sound) and get HDR and HBR audio support also.

Maybe this USB SSD installation fixes the RPi4 bot-back problem with SD cards too. I have to test NOOBS Companion add-on if that works

Anyways, thank you very much for this, highly appreciated!

jjlankinen, 2021-04-26:
Small update for users that wish to use latest LE and Retropie with NOOBS companion add-on. As it is known the boot {partition} does not work in PI 4 when using SD cards but when using USB (SSD or sticks) it seems to work just fine.

Maybe @matthuisman could update hes guide regarding boot-back with NOOBS Companion add-on. I take it that the Pinn system resize website (https://pinn.mjh.nz/) will be update soonish regarding this Retropie update as I still have to figure out other things like resizing the partitions as when using 120gb SSD LE ends up taking half and it is a bit too much in my case.

Also LE update to latest 9.95.2 was successful via LibreELEC add-on using correct update channel.

I think this issue can be closed now.

DerangedPineapple, 2021-04-26:
Small update for users that wish to use latest LE and Retropie with NOOBS companion add-on. As it is known the boot {partition} does not work in PI 4 when using SD cards but when using USB (SSD or sticks) it seems to work just fine.

Maybe @matthuisman could update hes guide regarding boot-back with NOOBS Companion add-on. I take it that the Pinn system resize website (https://pinn.mjh.nz/) will be update soonish regarding this Retropie update as I still have to figure out other things like resizing the partitions as when using 120gb SSD LE ends up taking half and it is a bit too much in my case.

Also LE update to latest 9.95.2 was successful via LibreELEC add-on using correct update channel.

I think this issue can be closed now.

procount, 2021-04-26:
Matt's resize website will not be aware of this new OS untill I release it fully. Remember that you had to change the cmdline to be able to see it.

DerangedPineapple, 2021-04-26:
Oh I know that you need to fully release it first. 

I merely meant that the Matt's website should or might be updated after your release. Bad wording on my part :)

Anyways, million thanks for you continued support on this matter!

DerangedPineapple, 2021-05-09:
I was wondering if you are not too busy to release the Retropie 4.7.1 so Matt's website might offer it in a future. I need to adjust the partition sizes for LE and Retropie as half for Retropie and half for LE is not really a good solution currently when using  alt_image_source.

Anyways, no hurry just asking whether it could be possible in the near future to get "official" release.

matthuisman, 2021-05-09:
my site will automatically pull in the latest OS's once a day around 3am NZ time I think.
So once it's in PINN - within 24 hours - you should see it on the resize app

procount, 2021-05-11:
Retropie 4.7.1 added to PINN.

DerangedPineapple, 2021-05-12:
Thank you for the addition. 

Highly appreciated! Will be testing it out soon.

closed 2021-05-12 by DerangedPineapple

DerangedPineapple, 2021-05-15:
Not trying to be annoying here but @matthuisman  's website has not been updated yet after Retropie 4.7.1 release for PINN was made.

It's been a few days so it should not be matter of synchronisation of mirrors. 

Sorry to pollute this repo for issues not directly related to PINN, I just have not found correct place to report that Matt's website has not been updated yet.

matthuisman, 2021-05-15:
@procount 

umm, they seem to list files that doesn't exist

"os_info":         "http://downloads.sourceforge.net/project/pinn/Retropie3/os.json"
"os_info":         "http://downloads.sourceforge.net/project/pinn/Retropie4/os.json",
"os_info":         "http://downloads.sourceforge.net/project/pinn/Retropie1/os.json",

they should be http://downloads.sourceforge.net/project/pinn/os/Retropie3/os.json
etc etc

procount, 2021-05-15:
Oops! Sorry for that. I didn't update the URLs properly when moving them from testing to os.
Should be fixed now after Matt's webpage has refreshed itself.

matthuisman, 2021-05-15:
Page has updated now :)