procount/pinn #506: Upgrade batocera os version (PINN downloads v29 but current is v30)

garret, 2021-04-05:
I do not know if it is in PINN power to update the batocera os version downloaded as PINN installs v29 but current version is v30. 

I am not sure if it is the right file but I also downloaded the latest batocera.tar.xz from https://batocera.org/upgrades/rpi4/stable/last/ and copied into the `os` folder. However, I received an error that there is no space. I also tried to copy the batocera.tar.xz file into an `os` folder of a usb stick (formatted as exfat) and attached to my raspberry pi 4. Nevertheless, I cannot understand how to tell PINN I would like to install that os. 

I guess the fastest option would be to alert that there is a new version of batocera since some weeks. That is why I came here, hoping is the right place.

procount, 2021-04-05:
ping @matthuisman

garret, 2021-04-06:
Is there a way to check which version of a specific os will be installed beforehand? 

I have now installed through PINN raspberry pi os and batocera v29. When the v30 can be downloaded through PINN, can I just reinstall batocera on the previous batocera space with PINN and being sure that the raspbery pi os partition is not touched?

procount, 2021-04-06:
Does Batocera include any form of upgrading, like a sudo apt update/upgrade procedure whcih will avoid you having to reinstall it each time?

Batocera was converted by Matt, so I'm not familiar with it.
The version information is normally available in the meta files and sometimes in the description. I might try to make it more visible.

garret, 2021-04-07:
@procount yes it includes an upgrade tool with its interface but for some reasons I get a "no updates available" message. They also provide a manual upgrade (https://wiki.batocera.org/upgrade_manually). I tried it and the process seemed to go fine but at the moment of rebooting I get a blank screen and nothing happens. 

Thus, it seems that the best way is to reinstall it. That is why I was asking if it is ok to just reinstall the new batocera v30 on top of the current batocera v29 with PINN? (I have only batocera and raspberry pi os as installed systems) 

Additionally, where can I find this `meta files` to check when batocera is updated to v30 on PINN? On the PINN interface I was not able to see any info regarding the version.

lurch, 2021-04-07:
Is https://github.com/matthuisman/pinn-os/blob/master/batocera_RPi3/os.json what you're looking for?

lurch, 2021-04-07:
Also, I've never used batocera but a bit of clicking around finds https://batocera.org/changelog which suggests the upgrade might be a bit fiddly? :shrug:

garret, 2021-04-07:
> Also, I've never used batocera but a bit of clicking around finds https://batocera.org/changelog which suggests the upgrade might be a bit fiddly? 🤷

Thanks, hadn't noticed that. It might explain why I did not succeed with the manual upgrade. 

@lurch when I look at the https://github.com/matthuisman/pinn-os/blob/master/batocera_RPi4/os_list.json 

```
{
    "tarballs": [
        "https://updates.batocera.org/rpi4/stable/last/boot.tar.xz"
     ]
}
```

I can see that it is pointing to the latest 32bit release. However, I just realized (again thanks to you) that batocera now seems to have switched to a 64 bit architecture. For this reason, the new address should be corrected to: 

```
https://updates.batocera.org/rpi464/stable/last/boot.tar.xz
```

I hope somebody can fix that soon but at least we have understood what was the reason 😊

lurch, 2021-04-07:
> I hope somebody can fix that soon but at least we have understood what was the reason

Cool, glad I was able to help :slightly_smiling_face:  The only person who'll be able to "fix" this is @matthuisman

garret, 2021-04-18:
May I ask when it can be expected that the link is going to be fixed? Is there a way I can obtain the same result without waiting for the fix?

matthuisman, 2021-04-18:
Sorry all, I for some reason had this repo ignored so didn't get any notifications of the mentions. 

I'll take a look today

closed 2021-06-24 by procount