procount/pinn #507: Backup Raspbian error

KilliK69, 2021-04-05:
Hello.
I am trying to backup Raspbian using PINN's option, but I am getting this error:
[image]

I guess it has to do with the OS's obsolete name. (I had issued here a similar problem in the past).
Is there something I can do to fix this? PINN is updated to its latest version.

procount, 2021-04-05:
Here are some instructions you can try:
1. boot into PINN as normal.
2. Enter the recovery shell using ctrl-alt-f2, login with root/raspberry
3. `cd /settings/Raspbian`
4. backup your existing script with `cp partition_setup.sh partition_setup.bak`
5. download a new script with `wget http://raw.githubusercontent.com/procount/pinn-os/master/partitions/Raspbian/partition_setup.sh`
6. `sync`
7. `rebootp`

KilliK69, 2021-04-06:
hello, procount, thank you for your reply.
I tried your guide, but I am stuck at step 3. When I press `cd /settings/Raspbian`, I get the error: `cd: can't cd to /settings/Raspbian`

procount, 2021-04-06:
Probably a spelling error.
`cd /settings`
`ls`
choose the folder that is most appropriate for your OS.

KilliK69, 2021-04-06:
ok, got it, it is actually `cd /settings/os/Raspbian`

procount, 2021-04-06:
Oh, yes. Sorry, I was going off my memory, which isn't always reliable :wink:

KilliK69, 2021-04-06:
now I am getting another error, when I apply the cp command.
the error is:
`partition_setup.bak: Read-only file system`

procount, 2021-04-06:
Just check you are in /settings/os/Raspbian. That should be writeable.
If it still doesn't work, post the output of `mount`

KilliK69, 2021-04-06:
ok, I think the problem was with me connecting through Putty/SSH. I tried again using directly the k/m, and it didnt give me any errors this time. 

The .bak file was created but the .sh file is still in the folder, the wget command worked but it saved the .sh from the repository, as partition_setup.sh.1. 

I assume this needs to overwrite the old partition_setup.sh file before I continue with the rest of the commands?

procount, 2021-04-06:
yes

KilliK69, 2021-04-06:
it worked perfectly. I finally completed the backup. Now I am going try what we discussed here:
https://www.raspberrypi.org/forums/viewtopic.php?p=1846767&sid=46f4217aebe5aaabdbd37e36d46206ac#p1846767

thank you very much for this great tool and all your help, procount.

closed 2021-04-07 by procount