procount/pinn #508: Allow for custom resolutions

Pechu317, 2021-04-07:
My display has a strange quirk that if it does not turn on with a 16:9 input, it will not detect any input at all. Obviously, this could be solved by using a different monitor, but I think it would be best if PiNN just had customizable resolution, instead of the 0-4 display mode thing.

procount, 2021-04-07:
You can set your own custom resolution in config.txt, just like you can in Raspberry PiOS.

Pechu317, 2021-04-07:
> You can set your own custom resolution in config.txt, just like you can in Raspberry PiOS.

so the config.txt file in the "RECOVERY" partition, right?

procount, 2021-04-07:
yes.
But be aware of the conditional filters in that file.
See https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md for details to make sure you put your config statements in the right place.