procount/pinn #509: How to install os-se from usb drive

Philip2909, 2021-04-08:


lurch, 2021-04-08:
I think you should just use the online-OSes supplied when you boot up PINN with an internet connection, rather then try to mess around with "converting" your own images (which is quite a complex process).

lurch, 2021-04-08:
@procount Is the similarity here to #482 just coincidence? Or are people somehow misunderstanding PINN's documentation?

Philip2909, 2021-04-08:
Hi, I just relized that the Os-es I needed was already there now that i did the internet connection. I tought it was gonna be the same offering as in NOOBS, I dont know if I missed that port of the docs where it says that you already uploaded a lot of oses, maybe give a list of which ones is already provieded when you add a network connection? I solved my problems but thx for the fast response and explanation!

closed 2021-04-08 by Philip2909

lurch, 2021-04-08:
https://github.com/procount/pinn/wiki/OSes-supported-by-PINN :wink:

Philip2909, 2021-04-08:
Oh, didn't see that, thx!