procount/pinn #510: Downgrade LibreELEC to 18.9

IronGibbet, 2021-04-17:
Hi,

I installed the Matrix Beta (LE 10.0) with the LibreELEC Update function to my Pi4.
Unfortunately it has flaws, that can be reproduced by the devs, in which some videos cannot playback the video file but just audio.
So I want to do a downgrade back to Leia 18.9 (LE 9.2.6).
I didn't do a backup via PINN before updating.
I wanted to use the reinstall function of PINN, but, if interpreted correctly, it only offers the Matrix version to download.

After reading the REDME_PINN.md I'm confused and would like to know what method would be the easeiest to do the downgrade without alterring the selection for LE 10.0

Thanks in advance and best regards
IronGibbet

procount, 2021-04-17:
LibreELEC is managed through the maintainers themselves via the NOOBs repositories.
It looks like the latest version they offer is 9.2.6.
If you use PINN's Reinstall feature, it will simply replace the version you currently have installed with the latest version that PINN would install normally. In the case of LibreELEC, this would be 9.2.6,

lurch, 2021-04-18:
(@procount It sounds like this is another "vote" for PINN being able to display OS-version information? :wink: )

IronGibbet, 2021-04-25:
Sorry that it took me so long to answer.
The reinstall to Kodi 18.9 or LibreELEC 9.2.6 worked like a charm. Thank you for your quick answer.
I'd also like to second @lurch with showing the version number that is about to be installed.

closed 2021-07-01 by procount