procount/pinn #512: NOt seeing any .tar.gz OSes in the /os directory of PINN.

bert269, 2021-04-23:
I am trying to add additional oses to PINN and followed the instructions and copied it to /os directory.
From the README.md file:
"PINN can install local images **that are found in the /os folder on the SD card**. These can include any of the original NOOBS images or any other OS that is formatted with a .tar.xz file for each partition."
further down: "The OSes should be stored on the **USB stick** in the same format as they would be on a PINN SD card (please see the download section):  Each OS should be stored **in its own folder beneath the /os folder.**"

Can we get some consistency here?  **Should the .tar.gz OS file be in the /os or the /os/subir located on the SD card?** I have tried BOTH options (/os and /os/subdir) and still do not see the OSes I am trying to install.
Thanks

procount, 2021-04-23:
Are you just putting .tar.gz files in those folders, or have you added all the additional metafiles you need as well?
Perhaps have a read of https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN and you may find https://github.com/procount/pinn/wiki/JSON-fields useful too.

Maybe it is a bit ambiguous if you don't realise the structure the information needed.
For an OS to be installed by PINN, you should create a folder underneath /os of the same name as the OS and include within it all the json meta-files, slides, icon files and the .tar.gz files for each partition. PINN looks for the os.json file to find the information about the OS to install. If that does not exist, it won't find your .tar.gz files no matter where they are.

If you download the Full NOOBS version you will have a template to start with as this includes Raspberry PiOS and LibreELEC installations that you can look at.

bert269, 2021-04-23:
thanks for the response. I will try to add the json meta-files for each OS. Thank you.

procount, 2021-04-23:
There is also this brief overview in the README_PINN.md -> https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version

Depending on which OS you are converting, this might be an easy process, or it might be very difficult. Some OSes just do not like being moved to different partitions, and some (like those that include UBOOT) just can't use other partitions without stripping out UBOOT since they assume they are being booted from partition 1, which is not the case after PINN installation.

procount, 2021-07-01:
PLease reopen if you need more help

closed 2021-07-01 by procount