procount/pinn #513: Pinn os with SD card or hard drive and recovery.cmdline

rserroyen, 2021-04-24:
Hello

I would like to have done under pinn os 3 os, Twister, Manjaro and Kali.
If I use pinn os with an SD card with a raspberry 4 I can see it.
If when I use a hard drive or a usb key the pinn os menu is different, I can't choose Marjaro or Kali.

when i and use edit config - edit option - install select i check Marjaro + Kali. The recovery.cmdline file contains

[image]

but when I reboot I have to I don't see Manjaro nor Kali why
thank you so much

procount, 2021-04-24:
Not all OSes can run from USB or HDD/SSD.
Not all OSes run on each RPi model.
PINN only lets you install OSes that suit your hardware, so the list will change accordingly.

procount, 2021-04-24:
By the way, I just looked more closely at what you had written, and selecting the OSes in config-edit option is not normally the way to do it. That is for a very special use case that will automatically install and overwrite those OSes whenever a new version is made available. I don't think that is what you want. I suggest you deselect those options and only select the OSes from the main menu. If they are not there, it is because they are not installable on your hardware combination.

rserroyen, 2021-04-25:
thank you so much

closed 2021-04-25 by rserroyen