procount/pinn #514: Could you add lineage18.1 (Android 11)

matthuo333, 2021-04-28:
Could you add lineage18.1 (Android 11)? Because when running lineage17.1, when the wired network is not plugged in, the system always restarts. Not in 18.1. For this issue, the related lineage thread has been confirmed.
Or can you tell me how to customize and add a new system?

matthuo333, 2021-05-02:
If the json file corresponding to 18.1 is made from the metadata below, is it possible?

https://sourceforge.net/projects/pinn/files/os/lineage17-rpi4-atv/

matthuo333, 2021-05-02:
Lineage generates four partitions, all of which need to be converted to tar.xz except boot? Through understanding the relevant JSON files of Lineage17.1 ATV, it is found that img.xz needs to be generated for system and vendor, is that right? How to make img.xz specifically?

matthuo333, 2021-05-02:
I did it and successfully started android 11 and LE10, this issue can be closed

lurch, 2021-05-02:
Perhaps you'd be willing to share some instructions, for the benefit of anybody else who might want to try the same thing?

procount, 2021-05-03:
well done!
Sorry I've not had the time to address this

matthuo333, 2021-05-06:
I tried to log in to the LE system, but the strange thing was that the ssh password was incorrect. To reset the new password, the same prompt is wrong.
Could this have something to do with the installation of dual systems in PINN?
By the way, I used a separate burned LE sd card, and then after booting the system, I had no problem logging in to SSH.

shivasiddharth, 2021-06-12:
@lurch I have generated the files for PINN. You can download them from here https://mega.nz/file/uwpmQZwZ#5c76OkdZcI5P-Ac9T3rvxdYsbRFWDtUCKN_pNlV5Eis

procount, 2021-06-24:
Lineage 18 & atv version are now added.

closed 2021-06-24 by procount