procount/pinn #515: Pinn OS + Wifi + Lali

rserroyen, 2021-05-04:
Hello 

I have a Raspberry Pi 4
For the boot I use Pinn OS
I use ssh the eth0 + wlan0 are UP and the wifi works with static + WPA2 address
You're okay
I put Twister, Manjaro and Kali as OS

When I boot with kali address changes and I don't know how to use Wifi

Why

thank you so much
```

kali @ kali: ~ $ sudo nano / etc / network / interfaces
auto lo
iface lo inet loopback

auto eth0
allow-hotplug eth0
iface eth0 inet dhcp

auto wlan0
iface wlan0 inet static
address 192.168.1.207
netmask 255.255.255.0
gateway 192.168.225.1

kali @ kali: ~ $ cat /etc/wpa_supplicant.conf
# reading passphrase from stdin
network = {
  ssid = "blackMOREOps"
  # psk = "abcd1234"
  psk = 42e1cbd0f7fbf3824393920ea41ad6cc8528957a80a404b24b5e4461a31c820c
}

  kali @ kali: / etc $ ip link show wlan0
3: wlan0: <NO-CARRIER, BROADCAST, MULTICAST, UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
     link / ether dc: a6: 32: 5d: 4b: 57 brd ff: ff: ff: ff: ff: ff

kali@kali:~$ iw wlan0 link
Not connected.
```

lurch, 2021-05-05:
> When I boot with kali address changes and I don't know how to use Wifi

Perhaps that's a question for the Kali developers rather than the PINN developer? :shrug:

procount, 2021-05-05:
Sorry, I don't know much about Kali. You could try on their website/forum for more info.
But probably it is creating its own MAC address, making DHCP on your router give it a different IP address.
Oh, and it's probably not a good idea to post your wifi password on the internet, even if it is just a temporary one, so I edited it out.

rserroyen, 2021-05-05:
Helo 

`kali@kali:~$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:5d:4b:56 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.5/24 brd 192.168.1.255 scope global dynamic eth0
       valid_lft 3436sec preferred_lft 3436sec
    inet6 fd93:7eca:83ed:0:dea6:32ff:fe5d:4b56/64 scope global dynamic mngtmpaddr
       valid_lft forever preferred_lft forever
    inet6 2a02:a03f:6c1f:e400:dea6:32ff:fe5d:4b56/64 scope global dynamic mngtmpaddr
       valid_lft 52521sec preferred_lft 38121sec
    inet6 fe80::dea6:32ff:fe5d:4b56/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether d6:30:cb:f2:31:06 brd ff:ff:ff:ff:ff:ff
kali@kali:~$ iw dev
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr e2:4c:22:e1:ce:e9
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 52:ef:02:0b:5a:60
                type managed
                channel 112 (5560 MHz), width: 40 MHz, center1: 5550 MHz
                txpower 31.00 dBm

`
I tried under vcn but the wifi is grayed out (no Interface) !!!! ???

I will try on the Kali forums

> Oh, and it's probably not a good idea to post your wifi password on the internet, even if it is just a temporary one, so I edited it out.

I put I put a do not ssid / pass fake lol

thank you so much

closed 2021-07-01 by procount