procount/pinn #516: Stop countdown with ssh

so-much-meta, 2021-05-12:
I'd like to be able to stop the countdown timer (remotecountdown) from an SSH session, but can't seem to figure out a way to do this. Any help here?

procount, 2021-05-12:
It can't be done at present from SSH.
You could do it from VNC, which was the main purpose, as VNC interacts with the GUI of PINN. But SSH does not really interact with PINN. This would need some enhancement. Not sure how at the moment.

so-much-meta, 2021-05-12:
Got it... One thought I had was to create a named pipe (mkfifo) and use that as a signaling mechanism. Looking at the code for e.g., joystick, it looks like it should be fairly straightforward as that does exactly what we need on the PINN/recovery side -- opens a file, reads from it, and then emits a signal. On the SSH side, it would just be "echo >> /tmp/stop_countdown" or similar. If I find some time, maybe I can see if I can get this working, but haven't yet actually built PINN.