procount/pinn #517: Booting PINN from PCIe on CM4

NumberOneGit, 2021-05-12:
Using the same CM4 carrier board with a dedicated PCIe slot, I am able to boot normally via a USB enclosure with an NVME drive into PINN but when booting with the same NVME drive in the PCIe slot I receive the error "Cannot find the drive with PINN files".  Is this because PINN is looking for mmcblk, sda1, etc but not nvme0n1p1?

procount, 2021-05-12:
Not really.
PINN searches all devices in /sys/class/block that are not virtual devices by mounting them and trying to locate recovery.rfs.
Does your drive show up in there?

NumberOneGit, 2021-05-12:
At the moment, after clicking past the above message I get to:
"Recovery application crashed
Starting shell" etc
When I have PINN installed on the eMMC it does not list the NVME over PCIe, but does over usb as sda

procount, 2021-05-12:
Maybe it is missing a driver?

procount, 2021-07-01:
@XECDesign - Do I need any additional drivers for NVME drives in the CM4 PCIe slot?

XECDesign, 2021-07-01:
I don't know, but @pelwell may.

lurch, 2021-07-01:
@procount Might be worth trying to compare the Linux kernel config used by PINN with that used by Raspberry Pi OS, to see if anything obvious jumps out?

pelwell, 2021-07-01:
We have CONFIG_BLK_DEV_NVME=y and CONFIG_NVMEM_RMEM=m.

procount, 2021-07-02:
Thanks @pelwell. I enabled CONFIG_NVMEM_RMEM after adding required CONFIG_PCI, but CONFIG_NVMEM_RMEM was not available. Guess I need a newer kernel for that.

NumberOneGit, 2021-11-04:
I'll try to test this out tonight, but are there any potential updates with PINN 3.7?

procount, 2021-11-04:
3.7.1 is the latest.
It has a later 5.10 kernel but I haven't enabled the additional CONFIG_NVMEM_RMEM config yet so it may still not work. I'll check if it is now available.

NumberOneGit, 2022-02-07:
Thanks again for all your great work here.  I am wondering whether you ever had a chance to see whether the CONFIG_NVMEM_RMEM config was available in the current kernel.  Thanks!

pelwell, 2022-02-07:
```
$ sudo modprobe configs
$ zcat /proc/config.gz | grep NVMEM_RMEM
```

procount, 2022-02-08:
Yes it seems to be available now, so I'll push up a version for you to test if you wouldn't mind.

NumberOneGit, 2022-02-08:
Absolutely.  Reply here once it's up and I'll give it a test.

procount, 2022-02-08:
https://sourceforge.net/projects/pinn/files/testing/pinn-3753.zip/download

procount, 2022-02-08:
I have no idea if it will work.....

NumberOneGit, 2022-02-08:
From what I can tell so far it works!  I wiped a CM4 so that only the NVME drive had PINN on it and am currently installing my OS's to PINN.  I will update again once the installation is complete and I am able to test boot some OS's, but all looks good on my end so far.  I'm hoping that the pinn_init.sh file won't have the same latency issues experienced on USB NVME's.   If all goes well with this, I will update on that front as well.

NumberOneGit, 2022-02-10:
No issues with PINN itself, but once the images are installed I will need to do quite a bit of research to figure out how to get them set up properly.  The firmware that allows this to work is pretty new, so manually updating the firmware of the various installations is necessary to even get access to PCIE boot.  I will keep experimenting with this.  Do you expect it to be enabled in updates (I notice there was an update earlier today) or not until it is more proven?

lurch, 2022-02-10:
AFAIK PCIe booting should "just work" with Raspberry Pi OS, but of course the situation with other OSes may vary... :wink: 

@timg236 or @peterharperuk Does the partition-selection used by NOOBS (something to do with PM_RSTS IIRC) also work with booting from NVMe SSDs?

NumberOneGit, 2022-02-22:
@procount   I would definitely say that it can be officially announced.  I have compiled a working kernel for OSMC using the config changes mentioned above and LibreElec has said they can add those changes to future builds.  RPOS has it enabled by default and RetroPie can boot from NVME over PCIe if it has been installed on another drive (SD or eMMC) and fully updated then cloned over.  

I don't expect Lineage to work with it any time soon.

lurch, 2022-02-22:
@NumberOneGit Does that mean that PINN's json files ought to report which OSes are bootable over NVME, and only offer to install those ones? (Like it already does for OSes that are bootable over USB, IIRC?)

NumberOneGit, 2022-02-22:
@lurch  I'm not sure how that's decided, as many OS's that I would put on the CM4 functional list seem not to appear there.  My only concern that if hiding functionality isn't working as expected, 'showall' may not work 100%.  I tried to install Kali 64-bit to USB a few weeks ago after setting up my cmdline.txt on pinn.mjh.nz and even with 'showall' enabled, it would disappear when I selected my USB drive and re-appear when I selected my SD card (on RP4B).  I use 'showall' by default when I'm using PINN with my CM4 because it's difficult to draw the line between "working" and "not working" in general at this stage.   For example, to get USB to work on OSMC I put "i2c-dev" in /etc/modules which allows my "otg_mode=1" command to work.  The dev may never make that change in the OS due to the way they choose to release it, but it takes very little effort to fix once one is aware that it's needed.

procount, 2022-02-23:
Normally, PINN hides OSes that do not include the current RPi model in the supported_models field. The `showall` option shows all OSes regardless of RPi model. But this is independent of whether the OS can be installed to USB.
The first is normally due to firmware and the second is generally due to kernel facilities.

I don't have a CM4 to test compatibility, so if you are able to do this I can add more OSes to the CM 4 family.

procount, 2022-02-23:
@lurch- how would I know it was using nvme vs usb?

lurch, 2022-02-23:
@procount I don't have a CM4 either, but I believe if you're booting off NVMe the boot device would be something like `/dev/nvme0n1p1` (at least, that's how the NVMe SSD in my Linux laptop appears).

paulwratt, 2022-02-23:
`/dev/nvme?n?p?` is how NVME on RPi is seen.

NumberOneGit, 2022-02-23:
Of course!  In general, I would say that RPOS and anything that is actually RPOS at the core (RetroPie) is at the moment the only successful OOTB NVME that can be added to the list.  As for LibreElec, I will have to check the nightly builds to see whether functionality has been added or if it was meant for a later release (the test version I verified worked OOTB).  OSMC I compiled myself and was asked to make a pull request, so not sure how long that will take.  A 'noobsconfig' that adds the line I mentioned above to /etc/modules in OSMC would be helpful if that is a system you already have in place.  I'm glad to update any other finding I make as they come along.  I'm interested in Kali, but am new to compiling and couldn't figure out how to update the kernel past 5.4.  I'll try out Ubuntu soon, but in general had some issues with Ubuntu and PINN already.

procount, 2022-02-24:
The test version for NVMe is available at: https://sourceforge.net/projects/pinn/files/testing/pinn-381.zip/download
I shall write up some tests to do in due course.

lurch, 2022-02-24:
I don't have a CM4, **BUT** I don't see how

> For example, to get USB to work on OSMC I put "i2c-dev" in /etc/modules which allows my "otg_mode=1" command to work.

makes any sense? Aren't USB and I2C entirely separate subsystems?

NumberOneGit, 2022-02-24:
I understand your confusion.  I felt the same way.  My first step was discovering that the way to enable DWC2 drivers in an OS was to add that under /etc/modules.  Because I knew that RPi Foundation now preferred to use XHCI over DWC2, I went to /etc/modules to see what was currently being used to enable the "otg_mode" command.  I wasn't expecting "i2c-dev" to enable the command either, but when I entered it it worked.  An alternative would be to add dwc2, but that seems to be deprecated.

NumberOneGit, 2022-02-24:
@procount   Thanks, I've downloaded it and will be glad to test it out.

procount, 2022-02-24:
@NumberOneGit  - Here are some tests for you:
Boot PINN from an SD card on a CM4
1. Go to Testing tab and check you can see Test1, Test2, Test3 and Test4
2. Change your Destination Drive to NVME and note which of the 4 above entries are now visible.
Boot PINN from the NVME on a CM4
3. Go to Testing tab and note which of Test1, Test2, Test3 and Test4 are visible
4. (Optionally, if you have a spare blank USB device) Change your Destination Drive to your USB device and note which of the 4 above entries are now visible.

NumberOneGit, 2022-02-24:
I only have eMMC version of the CM4 but I can try the same.  I believe PINN can't tell the difference as they both appear as mmcblk.  I'll try that out now.

NumberOneGit, 2022-02-24:
Test 1 and Test 2 are visible

procount, 2022-02-24:
Probably I made a mistake calling the OS entries TestX cos this could get confusing :/
So, there are 4 numbered STEPS in my previous post. Please could you identify the STEP you are testing and which of the TestX OS entries are visible for each one.

procount, 2022-02-24:
Not having a CM4 I've not really followed what it is or has.
So the eMMC is in place of the SD card, and is still identified as /dev/mmcblk0 ?
Your NVMe is in an external USB enclosure. What device does it show up as?

NumberOneGit, 2022-02-24:
I did the following tests on the official CM4 development board for which I have a pcie adapter.  The CM4 can be purchased with either SD slot or eMMC, I have two of the eMMC variety.  The NVME over pcie shows up as "nvme0n1", the eMMC shows up as "mmcblk0", when in the USB enclosure the NVME shows up as "sda1".  I installed PINN on only the eMMC for the first two steps of the pcie test with a blank NVME over pcie.  For the 3rd and 4th steps of the NVME pcie test I formatted the eMMC and installed the test version on the NVME over pcie.   I performed the first two USB tests with an SD card in an enclosure with PINN on eMMC and the 3rd and 4th USB tests with PINN installed on NVME over pcie.
Step 1(PCIE): all four are visible
Step 2(PCIE): 1 and 2 are visible
Step 3(PCIE): all four are visible
Step 4(PCIE):  1 and 3 are visible. 
Step 1(USB): all four are visible
Step 2(USB): all four are visible
Step 3(USB):  1 and 3 are visible (with NVME selected)
Step 4(USB):  all four are visible

NumberOneGit, 2022-03-03:
LineageOS update:  I was able to update the 5.4 kernel for Lineage 18.1 with the CONFIG_BLK_DEV_NVME=y config and can report that with this change I have found full functionality including TWRP.  I have also updated the 5.10 kernel for Lineage 19 to include CONFIG_NVMEM_RMEM=m.  I'm not sure whether you have any interest in the compiled kernels, as the change should have no negative effect on operation of the devices.  Lineage 19 actually worked already without the additional config.  I hadn't realized I had still been trying on 18.1.  That said, 18.1 is perfectly stable on the CM4 and has OpenGAPPS availability while Lineage 19 currently has functionality issues and I am only using it to try to debug and report back to the developer.   I'm glad to share any kernels that I compile for use in the /boot folder (with only one or both of the changes mentioned above depending on whether it's a later kernel that allows the RMEM change).
Other than the kernel change, I also added the most reason start4* and fixup4* firmware files to achieve functionality.

(It would still be a great help to add the dtoverlay line in /boot of PINN to future releases to avoid having to attach the CM4 to a laptop every time PINN updates to get USB back)

lurch, 2022-03-04:
> (It would still be a great help to add the dtoverlay line in /boot of PINN to future releases to avoid having to attach the CM4 to a laptop every time PINN updates to get USB back)

From what [I've read](https://www.raspberrypi.com/documentation/computers/config_txt.html#otg_mode-raspberry-pi-4-only),
```
[cm4]
otg_mode=1
```
(which is what RPiOS does) is a better option than
```
[cm4]
dtoverlay=dwc2,dr_mode=host
```

?

NumberOneGit, 2022-03-04:
Depends whether you're using DWC2 or XHCI.  I know that PINN already has DWC2 enabled but it would be fine to switch over

NumberOneGit, 2022-03-17:
I can also confirm that the latest version of Kali 64 bit works with a recompiled kernel.