procount/pinn #519: Volumio is not booting from PINN on Raspberry Pi 4

radcko, 2021-05-16:
Hi!
I've just flashed PINN v3.5.5 on my new Raspberry Pi 4 8GB and I installed LibreELEC and Volumio. LibreELEC is starting but Volumio is not booting (see picture below). The latest Volumio version (v2.882) is starting when flashed alone on SD card. The Volumio version which can be installed from PINN is the most recent one (v2.882) or is an older Volumio version?
Thank you!

[image]
[image]

lurch, 2021-05-16:
I think the `start4.elf: is not compatible` means that the `start4.elf` in your Volumio boot partition is too old to boot on your Pi4?
Hopefully this is something that @procount will be able to fix easily? :crossed_fingers:

radcko, 2021-05-16:
Is it something I can fix myself or requires more than very basic Linux konwledge? :)

lurch, 2021-05-16:
You could try mounting the Volumio boot partition from your PINN card, and copying across the start4.elf from the Volumio standalone version?

radcko, 2021-05-16:
I will try try this, thank you!

radcko, 2021-05-16:
I ran Ubuntu from an USB stick and I replaced `start4.elf` in Volumio PINN with the the `start4.elf` from Volumio standalone version. When I boot Volumio from PINN I can pass the boot screen but Volumio doesn't start, I have a black screen.

lurch, 2021-05-16:
> The good news is that Volumio is starting

Hooray! Well done for fixing your non-booting problem.

> the bad news is Volumio is not detecting my USB DAC (which is visible when using the standalone version)

Perhaps that's simply down to the version of Volumio in PINN being older than the standalone version? :shrug:

radcko, 2021-05-16:
> Hooray! Well done for fixing your non-booting problem.
>
>
> Perhaps that's simply down to the version of Volumio in PINN being older than the standalone version? 🤷

I don't think so, the older version of Volumio in PINN

procount, 2021-05-16:
Ping @matthuisman - I think Volumio is one of yours. Any chance you could update it?

matthuisman, 2021-05-16:
Current pinn volumio version is 2.779.
I'll try update to 2.882.
2.779 should boot on pi4 tho...

It is one of the tricked OS's. Requires editing the initrd

radcko, 2021-05-16:
> I'll try update to 2.882.
> I thought 2.779 should boot on pi4 tho...

Thank you!
Unfortunately it doesn't boot. If I can help with something...

matthuisman, 2021-05-16:
2.882 is up now :)
and tested on pi4 as well.

i had tested 2.779  on a Pi4 back when created...
so maybe the Pi firmware has changed since then which required the OS to have newer elfs

Please close this issue :)

procount, 2021-05-16:
Thanks Matt. Could it be because he had a Pi4 8GB?

matthuisman, 2021-05-16:
oh yeh maybe...

I swear my Pi4 acts differently recently.
It never used to show that bootloader screen with QR code...
but i did to a full upgrade in raspbian - so maybe the firmware was updated?

radcko, 2021-05-16:
> 2.882 is up now :)
> and tested on pi4 as  @well.

Oh yeah, it's working like a charm. 🤩 No problems at all with both USB audio output and HifiBerry Digi+ Pro digital outputs.
Thank you very much! 👍

Case closed! 😊

closed 2021-05-16 by radcko

lurch, 2021-05-17:
@matthuisman Yes, doing an `apt upgrade` in Raspbian will update your Pi4's EEPROM https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md (to [2020-09-03](https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L38) ).