procount/pinn #521: New OS or update - How to ?

MopheusDG, 2021-05-22:
Hi, Lakka just launched version 3.0 and it's based on LE 9.2 and 64 bits now, so my questions/doubts are:

- Is there any way to replace/update the old Lakka (32 bits) with this one without reinstalling all other OSes on my disk ? Not sure if copying the .gz file in update will work considering the kernel changes, 64 bit os, etc.

- How can we request an OS to be updated for the main OSes list is shown in PINN ?

Thanks, love PINN really, so flexible for installs and partitions and everything.

procount, 2021-05-22:
Lakka is one of the OSes that NOOBS includes, so you'll have to ask the Lakka and/or Noobs maintainers. PINN just includes the NOOBS installation list.
I tried following the links from there to Lakka and it doesn't seem to be available at the moment unless my browsing skills are failing me  🤷.

I imagine you wouldn't be able to upgrade from 32 to 64 bits, so when it does become available, you would have to use PINN's OS Replace feature to replace it.

MopheusDG, 2021-05-22:
Im currently downloading 3.0 from here:
https://le.builds.lakka.tv/RPi4.aarch64/Lakka-RPi4.aarch64-3.0.img.gz

procount, 2021-05-22:
What you are downloading is an image file for a standalone installation on an SD card.
It will need converting if you want to install it with NOOBS or PINN.

MopheusDG, 2021-05-22:
Yes, downloading in case takes too long to be added to the OS list in noobs... will give it a try on a separate mem. Can't wait to check Vulkan and other new stuff... lol

procount, 2021-05-27:
Lakka has been updated by its maintainers