procount/pinn #523: Black screen | PINN haven't started

hinzpak, 2021-06-06:
Hi PINNers! I am following all the steps, downloading and extracting PINN to the SDcard. My problem comes right after putting the SDcard into the pi4. There was a colorful screen with gradient but then it turns all black. The name of the SDcard changed to "RECOVERY" after I put it back on my mac, which I believe something was running, just that there was nothing (all black) when I put the card into pi4. any idea? Thanks!

hinzpak, 2021-06-06:
ok solved by switching to another HDMI port, like NOOBS
https://raspberrypi.stackexchange.com/questions/101842/noobs-stuck-at-rainbow-screen-raspberry-pi-4

closed 2021-06-06 by hinzpak

procount, 2021-06-06:
That's odd. If you are using an up to date version of PINN, it should now auto-detect which HDMI port is connected, provided you have the HDMI cable connected before you power on.

hinzpak, 2021-06-06:
Yea weird. But thats wht i have experienced with the latest download hours ago. So just switch the hdmi port whenever got stuck in the rainbow screen on PINN