procount/pinn #524: Wondering if this is compatible with 3.5 inch display

Jakeanator4522, 2021-06-08:


lurch, 2021-06-08:
Depends which display you mean, there's many different varieties...

Jakeanator4522, 2021-06-08:
My bad. I was wondering if it's compatible with the tft 3.5 inch LCD
touchscreen display. Attached file is the one I bought. Don't really know
how else to ask about compatability. Looking to run kali mainly. But wanted
to know if PINN is optional on this display for switching installed
operating systems. This is my first attempt at making it 100% portable.

lurch, 2021-06-08:
> Attached file is the one I bought.

Attachments don't work when you reply to GitHub emails...

Jakeanator4522, 2021-06-08:
Sorry new to all of this.

Hopefully this can help. Copied text is from Amazon.

for Raspberry Pi 4 Touch Screen with Case, 3.5 inch Touchscreen with Fan,
320x480 Monitor TFT LCD Game Display

Visit the Jun-Electron Store

4.0 out of 5 stars    687 ratings | 83 answered questions
List Price: $32.99 Details
Price: $29.99  & FREE Returns
You Save: $3.00 (9%)
Save up to 13% with business pricing. Sign up for free Amazon Business
account
Brand Jun-Electron
Display Technology LCD
Connectivity Technology Hdmi
Screen Size 3.5 Inches
Hardware Interface HDMI
About this item
New Design Screen built-in fan which can drop temperature quickly, and keep
your Pi Cool when working. The case is compatible with raspberry pi 4
perfectly.
The screen connect with SPI directly, Physical resolution is 320*480 Pixel
with resistive touch control. No additional Power.
The Screen Support 125MHz SPI signal input, stable display without flicker,
the refresh rate is about 50fps, enough to play the video and game.
The screen need install driver, otherwise screen ONLY show white. If you
don’t how to install driver, Please try the Pre-install Driver System to
test screen working statues