procount/pinn #526: Unable to mount 32GB+ USB in PINN for downloading OSs

Dimensional, 2021-06-11:
Running PIN V3.5.5, Raspberry Pi v4 4GB Ram.

Any attempt to use a USB drive larger than 16GBs will fail to be detected by the PINN interface, for use to download an OS for offline installation. If I plug in a 16GB drive, PINN will see it and allow it to be used for downloads, but anything bigger than that will be undetected. Even going into the command line and trying to mount the drive doesn't update the interface.

The drive can be a USB thumb drive or an SD connected via a USB adapter, and despite them being FAT32 formatted, the results will still be the same. This can be a problem if we are trying to test combinations of different systems and different configurations that would take up more than 16GBs of space from an offline installation source.

Steps to reproduce:
Plug in a 32GB USB drive into any of the USB ports, and try to download an OS to the attached USB.

Expected result:
PINN detects USB and allows for downloading to offline storage

Actual result:
PINN doesn't detect USB, downloading unavailable.

procount, 2021-06-11:
I'm not sure this has anything to do with the size of the drive.
Did you remember to create an `os` folder on the USB drive before inserting it? The drive won't be recognised as a target without it.

Dimensional, 2021-06-11:
It recognizes it now, though this is the first time that's happened. I never had to make the os folder on the 16GB drive before.

closed 2021-06-11 by Dimensional