procount/pinn #527: Installation issue with lineage18-rpi4-atv

shivasiddharth, 2021-06-12:
When installing the lineage18-rpi4-atv, it says that system.xz and vendor.xz are missing. If I rename the system.tar.xz and vendor.tar.xz to system.xz and vendor.xz, the installation goes through, but the OS never loads. It sets the Pi on a boot loop. Everytime I choose to boot lineage18-rpi4-atv, the Pi reboots.

shivasiddharth, 2021-06-12:
Ok. I have managed to resolve the issue. Apparently there is an issue with the system and vendor files that have been uploaded to source forge. You may update the repo with these files https://mega.nz/file/uwpmQZwZ#5c76OkdZcI5P-Ac9T3rvxdYsbRFWDtUCKN_pNlV5Eis I have tested it and its working.

procount, 2021-06-12:
Those files are not ready for release yet.
System Adn Vendor images must be img.xz files, which they currently are not.

shivasiddharth, 2021-06-13:
I have fixed those files. The files in the https://github.com/procount/pinn/issues/527#issuecomment-860048312 link have been tested and its working. I have modified the partitions JSON file as well.

closed 2021-06-21 by shivasiddharth