procount/pinn #528: Resizing Partitions

deep8787, 2021-06-16:
Hi all,

I just found about multibooting on the Pi and I successfully manage to get RetroPie, Lineage OS and Raspberry Pi OS installed and all were working fine. The only issue is that each OS has 80gb each (250gb SSD) and I really dont want it set up like that.
I tried manually editing each partition through gparted but that didnt seem to work, I ended up with a blinking cursor when I tried to load up RetroPie.
I found this website, https://pinn.mjh.nz/ , which is meant to help, it seems to do the trick but Lineage OS seems to be missing there. I chose RetroPie and two different versions of Raspberry Pi OS in the hopes I could just replace one of them with Lineage OS later on...also didnt work out when I tried to restore the original recovery.cmdline afterwards to show all OS's again. I also think with Lineage OS needing more then 2 partitions, this wouldnt have worked anyways.

Basically Im wanting 50gb for Lineage, 30gb for Raspberry Pi OS and and the rest with RetroPie. How would I go about this??
Im running a raspberry pi 4b 2gb.

Regards,

Nav

procount, 2021-06-16:
Lineage is a bit of an odd one and required additional support to boot off a USB/SSD. The https://pinn.mjh.nz website has not taken this into account yet, but I have asked Matt to make the necessary changes so hopefully it will re-appear shortly. Best advice is to wait until that happens. Gparted should have worked, but I suspect it messed with your PARTUUIDs.

deep8787, 2021-06-16:
Ok brilliant, I might give gparted another go then if it should of worked. I was only resizing all the partitions, moving the free space along,  so I dont think the UUIDs should of been affected. It is possible I did something incorrect.
Great work you have done with PINN, really great piece of software :)

deep8787, 2021-06-16:
Oh, I just checked https://pinn.mjh.nz/ again and Lineage is on there now! Happy days :D

deep8787, 2021-06-16:
I noticed that Lineage is available to install if you select SD Card instead of USB when on the opening page of https://pinn.mjh.nz/
I compared the recovery.cmdline with one made by selecting USB, they seem to identical bar the OS selected. I am currently installing the OS's now onto my SSD, fingers crossed it works ;D

procount, 2021-06-17:
Try again with Matt's website. He has updated the scripts.

deep8787, 2021-06-17:
Using the SD card method worked fine on my SSD last night, Ive been able to update both RetroPie and Raspberry Pi OS and theyre booting fine. 
I just had a looked at Matt's website, Lineage is still missing when you select USB. Hopefully the Android TV version will be put up soon, since its easier to move around the UI using a controller.

procount, 2021-06-17:
The USB/SD card selection on the website just controls the visibility of the OSes based on their capabilities. In this case, choosing SD card will work since  Lineage should work on USB. However, it would also allow you to install OSes to USB that definitely WILL NOT work.
Sorry, I forgot that although Matt has updated his scripts to correct this, it only refreshes the OSes every 24 hours, so it will take up to a day to show Lineage correctly.

deep8787, 2021-06-17:
Ahh ok ok I guesss I got lucky hehe.

Oki doki, I'll keep an eye out on Matt's website. Thanks for quick responses!

deep8787, 2021-06-18:
I seen its been updated @procount, is there any possibility you could speak with Matt so that he includes the Android TV version of Lineage as well?? 
Much appreciated.

procount, 2021-06-18:
That's not Matt's responsibility, but if you look under the Media tab, you will see it is already there.

deep8787, 2021-06-18:
Ahh I did not look there...brilliant! Thanks!

closed 2021-07-01 by procount