procount/pinn #529: Installing the RPI4 version of LibreELEC 10.0.

conmasterx, 2021-06-17:
LibreELEC have released a beta for version 10, but the only download available is a image.gz file. I'm completely new to this whole thing (installed the older version of LibreELEC and Retropie thats on the current PINN list with no problem), and the process of thaking that specifc file and someone adding it to the LibreELEC list is going right over my head. I've tried re-compressing the image within the Image.GZ into a tar.gz using 7zip, but I have no idea what to do with it. (tried copying the LibreELEC from the OS section of the NOOBS installer, but that contains 2 tar.xz files and I have no idea how to go about replacing these with that image). A step-by-step would be super helpful, as I cant seem to figure it out using the readme or the "backing up 2 OS" example

lurch, 2021-06-17:
https://github.com/procount/pinn/issues/510#issuecomment-821825348

conmasterx, 2021-06-17:
Aye I already caught that post too. I'm doing a fresh install, so I'm not really trying to update anything. I'm tryna extend past what the current stable release is and install the beta instead, and I'm aware that there is some method of adding that to the PINN list, i'm just not certain of the method and I want to be sure of which direction to go before I commit a few hours to getting it to work. :)

lurch, 2021-06-17:
If you're not sure what you're doing, your best option is probably to just wait and let somebody else do the hard work for you :wink:

conmasterx, 2021-06-17:
More than happy to wait if that's my only option, I'd just like to have it done sooner than later because that's currently the only way to get Kodi 19 on my RPi, and many of the addons I use have moved to Python 3 for Kodi 19. That's not me trying to rush any of PINN or LibreELEC's devs, I just wanna knuckle down and get my specific niche problem outa the way 😎

procount, 2021-07-01:
Best to wait until the maintainers release the next NOOBS version, which PINN will automatically pick up.

closed 2021-07-01 by procount