procount/pinn #530: config.txt query

deep8787, 2021-06-18:
Hello,

When Pinn boots up I have borders around the screen which I normally fix by editing the /boot/config.txt file. I went into the command line using CTRL+ALT+F2 and tried to find the config.txt so that I can make my various changes (including overclocking) but I couldnt find it, /boot folder was empty.

Also, would this config.txt file also be enforced for all the installed OS's or do I need to change them individually for each one? 

Thanks

procount, 2021-06-18:
As usual in Linux, there are at least 3 ways to do this.

1. Put your PINN SD card /SSD in a windows PC and edit the config.txt file on the RECOVERY partition (P1). It is not called BOOT.
2. When PINN boots, hold shift to get to the Main Menu. Press M twice to get to the Maintenance Menu. Highlight the PINN entry, press 'e' to Edit the config file. From here you can edit config.txt and recovery.cmdline. There is a virtual keyboard available if you don't have a real keyboard so you can do it with your mouse.
3. If you want to do it on the cmdline from ctrl-alt-f2, then you need to do the following, as the RECOVERY partition is mounted read-only on /mnt:
```
mount -o remount,rw /mnt
cd /mnt
nano config.txt
mount -o remount,ro /mnt
sync
cd /tmp
```

procount, 2021-07-01:
Sorry, I forgot to answer your 2nd question - Yes you need to edit config.txt on each OS separately. Each one is independent.

closed 2021-07-01 by procount