procount/pinn #531: PINN is not updating from 3.5.4 to 3.5.5

vrtlspd, 2021-06-18:
It says every time that there is an update, I hit update, it reboots but when it boots again is still in 3.5.4

Which logs do you need to check this?

procount, 2021-06-18:
Unfortunately, the logs are only temporary so they are cleared on reboot.
You could always try a manual upgrade by just unzipping the latest version of PINN-lite over the top of the old one. HOWEVER, please make sure you DO NOT replace recovery.cmdline, or at the very least make sure recovery.cmdline does not include `runinstaller`, otherwise it will erase all your existing OSes!!!!! Maybe take a backup of your SD card or OSes first!

vrtlspd, 2021-06-19:
I found the issue. The SD card has bad sectors and it's marked as write protected. For that reason it doesn't allow to update or change any file.

closed 2021-06-19 by vrtlspd