procount/pinn #533: Update an OS

xavier7179, 2021-06-22:
I apologize in advance if there is an answer out there but I could not find it. SInce updating an OS seems to delete all the others, is there a flow to update each OS to their latest version?

procount, 2021-06-22:
Please read https://github.com/procount/pinn/blob/master/README_PINN.md

PINN's job is more of an OS "installer". This is normally a destructive operation as it repartitions the drive and reformats each partition before it installs each OS.

After "Installing" several OSes, choosing to "install" again from PINN will wipe your drive and start again as it assumes you want a different set of OSes.  This is also destructive and not normally what you want to do unless you want to start from scratch.

If you want to overwrite one or a selection of your OSes with a more up to date version, then you can do an OS "Reinstall" from PINN's menu. This will destructively overwrite the OS with a new version, destroying any data you had in that OS, a bit like a factory reset. It's not normally what you want, but it does keep your other OSes intact. 

If you want to replace an OS with another one, then you can use PINN's "replace" option to replace one OS with another. There are some restrictions with this as the new OS must have the same partition layout as the old one, because it re-uses the same partitions. Again, any other OSes are left intact.

But normally, each OS has it's own upgrade procedure. Those based on Debian, like Raspberry Pi OS, use the "apt update; apt full-upgrade" procedure, but other OSes may have different package managers and different procedures. Please consult your OS on how to do this. The advantage of this approach is that it should maintain any data or customisations that you have made. This is the normal way to keep your OS up to date. You do not normally need to keep re-installing from scratch,

I hope this helps.

lurch, 2021-06-22:
Nice overview @procount :+1: 

Although a quick note about the last "each OS has it's own upgrade procedure" section - _some_ OSes _might_ assume that they're the only OS installed onto an SD-card, and their "upgrade procedure" may not be compatible with the multiple-OSes-on-a-single-SD-card methodology used by NOOBS and PINN? :shrug:  (e.g. an OS might attempt to "upgrade" by extracting the contents of an "update zipfile" onto the first partition of the SD card).
Hopefully this is a rare occurrence, but something to be aware of?

procount, 2021-06-22:
Yeah, thanks for reminding me @lurch.
I think Sakaki's Gentoo might be one of those OSes as I think it overwrites PINN's modifications. So after an upgrade it is necessary to re-run the partition_setup script from the fixup menu to get it to boot properly again.

xavier7179, 2021-06-23:
Actually, while I read the Readme section, on practical matters, I just installed the Recallbox (7.1.1) then the OS asked my about an update (7.2.2) which, after going for it, seems to broke the system because fix tools in PINN reported errors with the partitions (without recovering it) so I re-installed everything. Is it an isolated issue?

procount, 2021-06-23:
I just installed Recalbox 7.1.1 with PINN. (Be careful, there are 2 versions - Please choose the one that says "Use this one" in the description).
I set the wifi and chose to upgrade to 7.2.2.
I had to reboot/restart several times before it totally finished installing, but eventually it came up with 7.2.2-reloaded.
Seems to work for me.

lurch, 2021-06-23:
> Please choose the one that says "Use this one" in the description

:rofl: :laughing: :rofl:

procount, 2021-06-23:
(I have no control over the version of Recalbox produced by its maintainers which is sourced via the NOOBS repos. They went their own way and I've not really tested it)

lurch, 2021-06-23:
(I'm not implying you've done anything wrong, it just tickled me :slightly_smiling_face: )

xavier7179, 2021-06-23:
Yep I did it right… chose the one I had to 😊 My first impression was that the first partition was some how not containing the other OS anymore… is it possibile that it works depending on which partition it came to be stored into?

procount, 2021-06-23:
> is it possible that it works depending on which partition it came to be stored into?

No I don't think so. By installing with PINN it's already on partition 6+.
I've tried this twice now - first on a USB drive and second on an SD card. It behaved the same on both when I upgraded it. You do not need to run the fixup tools. Recalbox finds its partitions using their Label names, so it shouldn't matter which partition it is on. However, as a result you will only be able to install one copy or Recalbox on a drive.

I have my PINN configured with "forcetrigger" so it always stops in PINN when it boots.
When I choose to update Recalbox, it downloads the update and reboots.
On next boot it installs the update and reboots.
On 2nd boot it doesn't seem to do much, but it reboots again.
On 3rd boot it now loads Recalbox 7.2.2
Thereafter it works normally.

masscream, 2021-12-28:
Speaking about each OS having its own way to update, how about LineageOS? (Official builds use to have self-update, checking updates regularly.) A few days ago I started testing version 18.1 for RPi4 and found a few things which could be worth a little diggin', however after I reported some to its developer, he pointed me more on PINN support and also mentioned that the build I use (20210411) is "archaic", so any problems I can have could have been fixed already. Is there (even a manual) way please to keep the packages (almost) up to date? Or for example to upgrade manually somehow without losing any data/config changes I did already. Thank you.

procount, 2021-12-28:
I don't think anything that PINN has done to LineageOS's partition layout would have altered its update features. Please read the LineageOS FAQ on Konsta's page on "How to update from previous LineageOS 18.1 build without losing data?"

I have just updated the v18 and v19 versions of LineageOS to the latest version that PINN installs.
You should also upgrade PINN to the latest v3.7.5 version for better LineageOS support.