procount/pinn #534: Does PINN support Xbox One  wireless gamepad

ModernShrub, 2021-06-25:
I have a Xbox One controller which I want to use with PINN. Can anybody tell me how to do that? Also, I want to simulate a Shift keypress using the controller, as I do not have a keyboard. Plus, the controller supports USB and wireless. Model no : 1697. Pls help.
[image]

procount, 2021-06-25:
This is documented in the README. See https://github.com/procount/pinn/blob/master/README_PINN.md#joystick-control with further details in https://github.com/procount/pinn/blob/master/joy_keys.md.
Not sure that the wireless version will work, but USB should.

procount, 2021-06-26:
See also #376

ModernShrub, 2021-06-26:
thanks. I'll check and get back to you. Also, can you tell me where the PINN logo is on the screen?

ModernShrub, 2021-06-26:
Can you tell me how I can simulate a shift click using my controller?

procount, 2021-06-26:
With a game controller, you just need to click any button (once your game controller is mapped correctly).
The PINN logo covers the whole screen (but this is only relevant if you are using the mouse)
Why do you want to simulate a "shift-click"? The shift key is to interrupt PINN if you have a Keyboard. For a game controller you just have to click a button.
Have you looked in the logs to see if PINN has detected your controller and what it thinks it is called?

ModernShrub, 2021-06-27:
I have no idea where to check the logs

procount, 2021-06-27:
Have you got a USB keyboard that you can plug in to your Pi temporarily?
Is your Pi connected to your network?
Do you have a PC connected to your network that you can access your Pi with?
Do you know how to connect to your Pi with VNC?
Have you tried booting PINN with your XBOX One controller connected via USB? What heppened?

ModernShrub, 2021-06-28:
I have all of them. I connected the controller with a usb cable, and the controller just vibrated and didnt turn on. Only problem is that the A and Z keys on the keyboard dont work.

ModernShrub, 2021-06-28:
I have a feeling that PINN created a new controller file when I plugged in my controller

ModernShrub, 2021-06-28:
Wait no it didnt

ModernShrub, 2021-06-28:
My Pi is a rpi 3b+ 1gb, if you wanted to know.

procount, 2021-06-28:
Whilst testing, you need to stop PINN booting an OS automatically. I suggest to set the `forcetrigger` option to do this.

[ To set `forcetrigger`: 
- boot PINN and stop it booting an OS by pressing the shift key or clicking the mouse.
- Click the more button to go to the Maintenance Menu.
- Highlight the top PINN option and select Edit Config.
- Click the Edit Options button.
- Select the `Booting` tab and check the `forcetrigger` option.
- Click OK and Click OK again. 

PINN will now stop in the recovery menu each time you boot it. After testing, to continue to boot an OS, press Esc or click Exit and choose your OS to boot.
]

Please now look at the logs to see if your joystick is being detected.
To do this on the screen with a mouse:
- Go to the Maintenance Menu as above.
- Tick the checkbox to the left of the PINN entry in the list of OSes.
- Click the Fix button.
- Check the Show Debug Log and click OK.
- Unplug your joystick and click refresh (Do you get an "Error Reading Joystick" message?)
- Plug your joystick back in and click refresh

You should get something like this:
```
Looking for  "/mnt/Sony_PLAYSTATION_R_3_Controller_7A_13B.json"
Loading Mappings mappings from  "/mnt/Sony_PLAYSTATION_R_3_Controller_7A_13B.json"
```
This maybe followed by some joystick messages.
Please report which json file it is looking for and loading.

An alternative method to view the logs is using a keyboard.

- Press ctrl-alt-F2 (together) to get to the command prompt.
- Login with username=root, password=raspberry (You might have a problem if your 'A' key does not work on your keyboard)
- type `tail -f /tmp/debug`

The log messages will scroll up the screen with no need to press any Refresh key each time.

A third method is to connect over SSH from your PC using Putty. See https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-access-the-shell-or-ssh-into-pinn

procount, 2021-06-28:
From what I have read, the XBox One controller is just a redesigned XBox 360 controller. If that is true, then you shoudl be able to use one of the existing 360 json files supplied with PINN. You may just have to rename it to match what PINN expects. You'll find the name in the logs obtained in my previous comment above.

ModernShrub, 2021-06-29:
Thanks, I'll try and get back to you.

ModernShrub, 2021-06-30:
Hey @procount ,
I tried it, and it said "Looking for Microsoft_X-Box_One_pad_Firmware_2015_8A_11B.json"
Loading Mappings mappings from  "/mnt/joy.keys.json"

ModernShrub, 2021-06-30:
I also renamed the XBox 360 gamepad file as you said, but the pointer was going bonkers when I used the gamepad's joystick to move the mouse. The pointer would either go downwards contiously or it would go sideways continuosly

ModernShrub, 2021-06-30:
Is the Insider Preview to be put onto the sd card or is it downloadable online?

procount, 2021-06-30:
It looks like the X-Box One has different mappings to the 360, so you will have to map them out. 
See my above post again about viewing the logs, then just press buttons/joysticks on your gamepad and identify the button or axis number and the value when it is pressed/released or up/down/middle. (See #376 for examples). From there you will be able to create your own "Microsoft_X-Box_One_pad_Firmware_2015_8A_11B.json" file.

> Is the Insider Preview to be put onto the sd card or is it downloadable online?

Are you talking about WindowsIoT? Microsoft gave up supporting it on the RPi, so you'll have to rely on other forum posts etc to get it working. It was always an online download only.

ModernShrub, 2021-07-02:
@procount ,
Will Retropie be any help in mapping the joystick?

ModernShrub, 2021-07-02:
Because i have retropie installed on pie using pinn

procount, 2021-07-02:
I don't think so. It's quite easy to do in PINN. Just watch the logs, press the buttons and note down the messages.

ModernShrub, 2021-07-02:
ok, thanks. I'll try

ModernShrub, 2021-07-08:
Ok, it worked. I'll post the file later on this issue.

closed 2021-07-08 by ModernShrub

procount, 2021-07-08:
Excellent. I can add it in the next release to benefit other Xbos One users.