procount/pinn #536: Roll Back to earlier version. SD Card Checks before install.

Echogolf123, 2021-07-03:
Existing SD card with 3 OS (PIOS Full, Librelec and PIOS Lite) and 2 empty project spaces created using PINN downloaded on 9th May 2021. Wanted to backup one of the OSs on the SD card via PINN to my usual 64GB USB flash drive used for holding downloaded and backed up OS within the os folder. This had previously been done succesfully and is used regularly.

On boot into PINN an option to update PINN was offered. This was accepted and the update completed. Then the installed OS desired was selected to back up to the 64GB USB flash drive. Shortly after starting the backup fails with a message that there may not be enough disk space. This is not correct. Various other attemps were made including reformatting FAT32 the USB drive first and using an alternative freshly formatted drive. Same failed outcome. On inspecting the 64GB USB flash drive, PINN had created a folder for the OS to be backed up into and had placed some files before failure.

Then wishing to revert to the previous PINNs version the previously downloaded PINN was (clearly wrongly) copied back to the SD card and booted. This wiped out the installed OSs already on the SD card and resized the partitions.

Please:
1) Is there a bug in the current PINN automatic update that is failing with backups?
2) How to roll back to an earlier PINN where existing OSs are already installed onto the SD card and one wants to keep them?
3) Can a check be added to PINN such that if existing OS partitions are detected on the SD card they are either picked up and included or PINNs fails non destructively with an appropriate message.

procount, 2021-07-03:
Which of those OSes were you trying to backup? Was your 64GB USB drive formatted as FAT32 initially (i.e. before you reformattted it as FAT32?) There is a 4GB fiel size limit on FAT32 which might be a limiting factor depending on how big your OS was.

1. Not that I am aware of. If it happens again, please send me some logs and more information (which OS, partition sizes and usage of partitions, size of USB drive etc.)
2. I have made many comments about this, on the forum, github issues and in the README instructions. Whenever you install a new version of PINN, it includes the `runinstaller` option which wipes the disk on first boot. It is for initialisation only. I even added warning dialog messages to PINN if you choose to set this option in the config dialogs. However, it seems there are still ways that users miss this. If you want to install a different version of PINN to an existing installation, you MUST always delete the `runinstaller` option before booting, otherwise it WILL wipe your disk.
3. This sounds like a good idea and should make it safer. in the future. I will look into this with priority.

Echogolf123, 2021-07-03:
Cutting to the chase: The last line in the log at the point of failure reads:
_cp: can't stat '/settings/os/Raspberry_Pi_OS_Lite(32-bit)/slides_vga': No such file or directory_ see image below.
A quick search on line suggests trying to copy a file with a space in it maybe and I suspect the root of the failure?
[image]
[image]
[image]

In which case the rest may be unnecessary:
My USB drive that I both install OSs from and back up to is Fat32 formatted and already had other downloaded and backed up Pi OSs on it before I backed it up and reformatted it Fat32 again just as a test to eliminate some kind of corruption.

I was backing up virtually vanilla PiOS Lite, as downloaded from the Pi foundation download server, build from 7th May 2021. I download to the os folder on the USB so I can modify the partitions.json before install. The whole OS installed was a tad over 2GB after doing a apt update and apt full-upgrade. The partitions are less than 4GB in total. 

I rolled the SD card back to PINNS 3.5.5, deleted _runinstaller_ but I get the same error. (yes please do something about that otherwise catastrophic _runinstaller_ edit for us linux learners by way of 2 confirmation boxes as that error to edit that file wrote off an afternoons work. I do document what I do along the way but I now have to do it all again). :-S.

procount, 2021-07-04:
Does the folder /settings/os/Raspberry_Pi_OS_Lite_(32-bit)_/slides_vga exist? 
Do you have any files in it? There should be 7 files: (A.png - G.png) 
There shouldn't be any files with spaces because all spaces are replaced with underscores.

Making good progress with the next update.

Echogolf123, 2021-07-04:
I found /os/Raspberry_Pi_OS_Lite_(32-bit)_ on /dev/mmcblk0p5. There was no slides_vga folder under it so I created it, left it empty and ran the back up again under both PINN v3.5.5 and v3.6. Both were then successful. :-)

I then checked [http://downloads.raspberrypi.org/raspios_lite_armhf/](url) where I obtained the original install image from to see if I had missed downloading something but I don't think I did. Certainly the install of the OS ran correctly from the image I downloaded under PINN v3.5.5 and I've done nothing special to it nor deleted anything. Should PINN have created this folder with any contents with the install? I haven't had this issue when backing up either Libreelec configurations or PiOS full install.

Thank you for your help and developing PINN. I use PINN quite a bit for taking OS snapshots when developing small projects which then get replicated to other Pis I have Pick'n'mix style in a home production environment.

I'll look forward to seeing an intelligent runinstaller with the next release.

[image]

lurch, 2021-07-04:
The `slides_vga` directory exists inside http://downloads.raspberrypi.org/raspios_lite_armhf/marketing.tar :wink: 

@procount The error seems to be coming from https://github.com/procount/pinn/blob/master/recovery/mainwindow.cpp#L4832 where you seem to have forgotten to check if the folder exists before you try copying it? :confused:

Echogolf123, 2021-07-04:
> The `slides_vga` directory exists inside http://downloads.raspberrypi.org/raspios_lite_armhf/marketing.tar

Got it. I have marketing.tar within my \os\Raspberry_Pi_OS_Lite_(32-bit) PINN image folder used for the original OS installation.

procount, 2021-07-05:
> where you seem to have forgotten to check if the folder exists before you try copying it?

It appears so. I will add that check. 

It doesn't really explain why the slides_vga folder didn't exist. It suggests that marketing.tar wasn't present when it was installed, although @Echogolf123 says it existed.

procount, 2021-07-05:
There is a new version available at https://sourceforge.net/projects/pinn/files/testing/pinn-361.zip/download
* Add Dialog check with runinstaller. 
* Check for slides_vga when backing up
* Prevent PINN updates overwriting user files

Please check that it fixes your issue.

Echogolf123, 2021-07-05:
Following tests were performed installing Raspberry Pi Lite OS downloaded from here: http://downloads.raspberrypi.org/raspios_lite_armhf/ with:
A) PINN v3.6
B) PINN v3.6.1

under two scenarios:
1) with _marketing.tar_ within the downloaded install folder archive as downloaded.
2) with _marketing.tar_ expanded back to _slides_vga_ folder and marketing.tar deleted from the install folder archive.

**Results:**
**(A)(1) PINN v3.6 and _marketing.tar_**
Operating system installed correctly and booted correctly afterwards.
No marketing slides were observed during the install process.
On the OS SD card under _/dev/mmcblk0p5_ folder _/os/Raspberry_Pi_OS_Lite_(32-bit)_ **no** _slides_vga_ folder was observed.
A subsequent backup attempt failed consistent with the original issue raised.

**(A)(2) PINN v3.6 and _slides_vga_**
Operating system installed correctly and booted correctly afterwards.
Marketing slides were observed during the install process.
On the OS SD card under _/dev/mmcblk0p5_ folder _/os/Raspberry_Pi_OS_Lite_(32-bit)_ _slides_vga_ folder was observed with .png slide images inside.
A subsequent backup attempt completed successfully.
folder _slides_vga_ comprised part of the backed up archive folder.

**(B)(1) PINN v3.6.1 and _marketing.tar_**
Operating system installed correctly and booted correctly afterwards.
No marketing slides were observed during the install process.
On the OS SD card under _/dev/mmcblk0p5_ folder _/os/Raspberry_Pi_OS_Lite_(32-bit)_ **no** _slides_vga_ folder was observed.
A subsequent backup attempt completed successfully. :+1: 
folder _slides_vga_ was not part of the backed up archive folder.

**(B)(2) PINN v3.6.1 and _slides_vga_**
Test wasn't completed.

**Conclusion:** PINN v3.6.1 (and v3.6) is not handling _marketing.tar_, only folder _slides_vga_ for either installs or backups.
**Suggestion:** It maybe a minor issue now that the backup appears to run regardless of the presence of folder _slides_vga_ however so as to be consistent with how the Pi Foundation distribute their NOOBS compatible OS archives, ie with slides_vga folder TARed into marketing.tar, consider having PINN do likewise with both installs and backups.

**(B)(1) PINN v3.6.1 and _marketing.tar_** testing was then continued:
_recovery.cmdline_ with `runinstaller` present at the beginning was restored to the PiOS SD card and the PI rebooted. The following dialog was displayed:
[image]
_No_ was selected. :+1: 
PINN was exited and then booted into PiOS lite. :+1: 
A further reboot into the PINN installer was normal without the dialog and the installed OS clearly still intact. :1st_place_medal: 
An inspection of _recovery.cmdline_ on the SD card. showed `runinstaller` not present within the file. :+1:

closed 2021-07-05 by Echogolf123

procount, 2021-07-06:
Thanks for your test results.
The (A)(1) and (B)(1) tests are not producing the correct results.
PINN _does_ use the marketing.tar file when installing from a download. When it detects marketing.tar, it expands it on the USB drive and then presents the slides from slides_vga.
So either:
1) There is a problem with the LIte marketing.tar file, or
2) The USB drive where marketing.tar was, is not writable and marketing.tar was not expanded. You should be able to tell if the marketing.tar file was expanded on the USB drive after the installation completed.

procount, 2021-07-06:
I'm guessing you downloaded the OS installation files manually from the website, rather than using PINN's built-in facility to do this?
Downloading it through PINN expands marketing.tar into slides_vga when it downloads it. But I confirm having the original marketing.tar file in the downloads folder doesn't seem to work. I would have thought it should so I need to investigate.

lurch, 2021-07-06:
> But I confirm having the original marketing.tar file in the downloads folder doesn't seem to work. I would have thought it should so I need to investigate.

Back in the original NOOBS code, it was assumed that if you were doing a local install, you'd already have marketing.tar expanded into slides_vga. marketing.tar was _only_ used for online installs (because downloading a single tarfile is quicker/easier than downloading separate images). marketing.tar being used for a local install was never a supported (or even thought about) configuration!

But the "usage modes" for NOOBS back in the day were either that you'd have downloaded "full NOOBS" (i.e. as prepared and tested by the NOOBS maintainer(s)) in which case you'd have the OS's marketing.tar already correctly expanded onto the SD card, **OR** you'd be using NOOBS Lite in which case it'd be downloading the marketing.tar from the downloads server and expanding it as necessary.
Creating your own "local install" by manually downloading the files from the downloads server and copying them onto an SD card, was never really considered (which is why a local install doesn't even _look_ for marketing.tar).  (Because originally "full NOOBS" contained **every** OS, it was only later (when the Raspbian images became bigger) that Full NOOBS was trimmed down to contain fewer OSes)

At least, that's how I remember it... :wink: 
(Also back in the day there could be both a `slides_vga` folder (low-res images) and `slides` folder (high-res images), but I guess somewhere along the way the high-res `slides` folder got dropped?)

procount, 2021-07-06:
@lurch - you remember it well, as always.
Yeah, I haven't done anything different to change that original NOOBS behaviour. If you download the OS through PINN, then it works fine as it does the expansion. But if you just download the metafiles manually, then it will fail on Raspios_Lite (sic).

And I've just realised why there is a difference......
All of my converted OS folders on sourceforge already include both marketing.tar and slides_vga/, so downloading them manually is fine. But the OSes hosted by NOOBS only have the marketing.tar file, so it would need to be manually expanded.

(with regards to the `slides` folder, I'm not sure whether support is still there for that. I certainly dropped the creation of those high-res images - it's bad enough having to create one set, let alone 2, and they take up too much unnecessary space or download time. vga looks good enough anyway. I scale all images to be the same size as the largest one)

lurch, 2021-07-06:
...and of course if you _do_ decide to add support to PINN for having a marketing.tar file locally (rather than just downloading it dynamically), you need to decide what to do in the edge-case of there being _both_ a `marketing.tar` and `slides_vga` directory on the SD card :wink:

procount, 2021-07-06:
There is a new PINN version available at https://sourceforge.net/projects/pinn/files/testing/pinn-3611.zip/download

This should now cater for the case where there is only a marketing.tar file. If the slides_vga folder also exists, the tar file will be expanded first, then the slides_vga copied over the top of it. (This maybe slight overkill, but it should guarantee to get the slides. They should be the same anyway. A minor performance inefficiency)

Echogolf123, 2021-07-06:
> I'm guessing you downloaded the OS installation files manually from the website, rather than using PINN's built-in facility to do this?

Yes, that is correct from http://downloads.raspberrypi.org/raspios_lite_armhf/
I do that so I can edit the `partitions.json` file before install. I typically set the partitions size and `"want_maximised": false` 

> Downloading it through PINN expands marketing.tar into slides_vga when it downloads it. But I confirm having the original marketing.tar file in the downloads folder doesn't seem to work. I would have thought it should so I need to investigate.

Understood. Incidentally I downloaded Noobs v3.7 full fat from http://downloads.raspberrypi.org/NOOBS/images/NOOBS-2021-05-28/ and had a look in the `os` folder. You get `slides_vga` not `marketing.tar`.

I also did a test with Noobs_lite v3.7 from http://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2021-05-28/ trying to install the previously downloaded raspios_lite_armhf (see above link) from my USB drive trying both `marketing.tar` and `slides_vga` akin to scenarios (A)(1) and (A)(2), see https://github.com/procount/pinn/issues/536#issuecomment-874371539 . It wasn't a pleasant experience and both times it quickly and completely failed with an _unexpected ")"_ error (whatever that means). After that there was no way out from the blank white screen that came next that I could see and I had to pull the PSU plug. It was so bad I ran an install again using PINN v.3.6.1 with the same archive folder from my USB stick just to be sure it wasn't the OS install archive that was corrupted somehow. That was successful. Noobs seems like it is broken to me at the moment.

My view is that whatever the Pi Foundation are choosing to do sets the standard and that should be followed.
However what they are presently distributing via http://downloads.raspberrypi.org/ is inconsistent because of `marketing.tar` with that which one receives within a full fat Noobs image and that works correctly with Noobs (you hope) and PINNs. I understand why that is now.

In an ideal world I think if `marketing.tar` is what you get at source then any install archive held locally should also have marketing.tar. The actual install should handle marketing.tar end to end and also return it to you with a PINN back up. But presently that is not the standard because Noobs does not do it that way and so the Pi foundation should be politely asked to fix Noobs to make a new and consistent standard.

Noobs doesn't do backups at all and with PINN v3.6.1 backups work without `slides_vga` for both install and back up now. Going forward I think I would be inclined to not take slides_vga or marketing.tar forward to the installation SD card in any event in future. It's just taking up space. For the sake of consistency place an empty `slides_vga` or `marketing.tar`, according to the standard, in the backup archive folder on making a backup.

procount, 2021-07-06:
Well, it's just as @lurch recalled. 
Wherever possible, I have tried to maintain compatibility with NOOBS and only added better functionality on top, so I have to live with some of its foibles. There are many inconsistencies between files installed from a local source and those from the internet which I have had to work around. I think these are borne from the way it developed organically. If I was to redo PINN again, I would probably do it a bit differently next time. But it is what it is.

Having just gone through a number of hoops to get it to work with/without marketing.tar/slides_vga, I will leave it as it is now. The aim was to make the installation of a backup look as close as possible to an original install. You are free to delete the marketing.tar or slides_vga folders if you want to save space, but at least it should work now.

Thanks for your comprehensive test results and reports. If you spot anything else, please let me know.

Echogolf123, 2021-07-06:
No, thank you for PINNs. I use it all the time.
I was going to do another test but install multiple OS at once. How do I manually make an archive folder show up in the Backups tab? I had thought appending #<date>-<time> to the folder name but apparently not.

procount, 2021-07-06:
The important thing is the name of the OS inside os.json (and os_list.json), not what folder it resides in. They are normally the same, except where spaces (and now brackets iirc) are converted to underscores.

Echogolf123, 2021-07-07:
Thank you.
I did some further testing; the four possible combinations of `marketing.tar` and `slides_vga` with four OSs installed and backed up and that worked as expected shown by the backup output. The backup `slides_vga` folder showed that if present the input `slides_vga` images took precedence over the input `marketing.tar` images. Otherwise you get `marketing.tar` images. Where there is neither `slides_vga` nor `marketing.tar` inputs, I didn't get an output `slides_vga` folder.
The only thing that gave me a small anomaly to begin with was all of my test OS had `"want_maximised" : false` in `partitions.json` which was fixed by adding a project space to fill the remaining space. But I think I read somewhere before that at least one of your OS or a project space have to have `"want_maximised" : true`

procount, 2021-07-07:
I don't suppose it is mandatory, but if you don't you will have wasted space on your drive, that's all.
Have you seen http://pinn.mjh.nz ? That will also allow you to adjust the partition sizes.