procount/pinn #538: PINN v3.6 no longer support sdtv modes

Kyuchumimo, 2021-07-07:
When I start the recovery mode with the shift key, then i press the key '4' to switch to NTSC, but it no longer takes effect.
This problem was also present in NOOBS since version v3.3.

Is it possible to downgrade the PINN version please?

procount, 2021-07-07:
It is due to the latest firmware changes for fkms graphic drivers and also the PI4 hardware.
SDTV mode should still be available but you need to add `enable_tvout=1` to config.txt.
See https://www.raspberrypi.org/forums/viewtopic.php?t=251277 which is a bit old but I think still relevent.

Kyuchumimo, 2021-07-07:
My config.txt file remains the same as it was before the update, and of course, I already took into consideration `enable_tvout=1`:

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

# NOOBS Auto-generated Settings:
sdtv_mode=0
hdmi_ignore_hotplug=1
enable_tvout=1

#overscan_left=16
overscan_right=4
overscan_top=-16
overscan_bottom=-16
dtparam=i2c

procount, 2021-07-07:
Is that your config.txt file in the PINN recovery partition, or the one in your Raspbian BOOT partition? It looks like the one in Rapsbian that NOOBS/PINN may have generated. But you need to add enable_tvout=1 to the config.txt file in PINN.

Kyuchumimo, 2021-07-07:
[image]

The config.txt file is located at the root of the micro SD memory card.

lurch, 2021-07-07:
That's not the PINN partition - the PINN partition would contain a `recovery.rfs` file (amongst others). That screenshot looks like a typical "Raspberry Pi OS" boot partition.

IIRC there's something "special" about the PINN partition, which may mean that it doesn't always auto-mount; and you may need to manually mount it using a disk-utility program or something?

procount, 2021-07-07:
@lurch - Yes, because its label is RECOVERY, many distros will not mount it automatically.

@Kyuchumimo 
Each OS (including PINN) has its own set of firmware, boot and kernel files. You need to edit config.txt file in each OS if you want them all to have similar configurations

From the Raspbian(sic) cmdline you could do the following:
```
mkdir -p /tmp/pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
cd /tmp/pinn
nano config.txt
```

Or you can do it in PINN:
```
Go to maintenance menu
highlight pinn
Select Edit Config
add appropriate entries to the config.txt file
```

Kyuchumimo, 2021-07-07:
I solved it, it is enough to add 'enable_tvout=1' in the configuration, what happened then was that PINN when updating, overwritten its own configuration

procount, 2021-07-08:
That last issue of overwriting should be resolved in the next release (after you've installed it - so please check for the overwrite on the next update aswell. Thereafter it should be ok)

lurch, 2021-07-08:
@procount Just out of curiosity, have you thought about using the `include` directive in PINN's `config.txt`? https://www.raspberrypi.org/documentation/configuration/config-txt/misc.md

You could e.g. `include user_config.txt` and then the user could make their own changes in `user_config.txt`, and PINN would still be free to overwrite `config.txt` without having to worry about overwriting any of the user's custom settings? :shrug:   (although there might be implications I haven't considered, which means it isn't as easy as that?)

procount, 2021-07-08:
I did think about it, also for cec_keys.json and joy_keys.json which could suffer similarly.
For now I just prevent expanding over newer files, so as long as the date is correct when they edit files, it should be better. but I admit not ideal.

If I use include files, do I then need to add that inclusion to the EditConfig option?
Ubuntu uses several include this. There is a request to be able to edit all of those include files too.
And should I include the file at the top or the bottom of config.txt (or both?)
Where does it end?

lurch, 2021-07-08:
> For now I just prevent expanding over newer files, so as long as the date is correct when they edit files

I guess that still has the potential to go wrong if they edit files without an internet connection? Is there some "cut-off date" (e.g. 2012) where you could say "if either of the files have dates before this date, then we know that any date-comparison is invalid"? :shrug: 

> .... Where does it end?

LOL, fair enough! :grin:

procount, 2021-07-08:
> I guess that still has the potential to go wrong if they edit files without an internet connection? 

Absolutely. They can set the date through PINN's UI, though.

> Is there some "cut-off date"

Err... No. "unzip -uo"

closed 2021-08-06 by Kyuchumimo