procount/pinn #542: Restore after backup fail

jerome-karabenli, 2021-07-22:
Hi, 

I'm trying to have an more stable server, and backup is the most important part of this. 
I'm using RaspberryPi4 4gb with 2 ssd's attached over usb 3.0 ports. 
I installed a raspbian 32bit lite and an empty slot for future os maybe or backups.
I install OpenMediaVault, and docker for the essential part in my os.

Everything works, backup and restore, until I run a new empty PINN, and restore my backup in this one. 
I can connect via ssh, so `sshd` is working, but everything else is down, `docker, dns resolver, openmediavault` etc. 

Is this normal ?

procount, 2021-07-22:
Are you running PINN on one of the SSDs or on a separate SD card?
So if you backup your OSes with PINN and restore them to the same drive, everything works perfectly - OMV and Docker etc?
But restoring the same backup to a new driver with a fresh copy of PINN doesn't work?

jerome-karabenli, 2021-07-22:
PINN is running for boot and root partions on only one SSD.
Yes, if I backup and restore OS on the same drive it's works perfectly. 
And no, when I restore to a new fresh copy of PINN in the same SSD it doesn't work. To be clear, it's works, the os can boot, I can connect it in ssh, but that's all, everything else is broken.. 
When I try to reinstall omv, it's like my dns resolver doesn't work, BUT when I inspect for more details, I have a message :
`/etc is ready only` even if I try to change this from `raspi-config` panel, it's not working..
I do not have enought experiencing about permissions to resolve that ..

procount, 2021-07-22:
There are some other issue numbers that may be related that you should look at:
#391, and possibly #447

The problem is most likely to do with docker.
As the backup/restore works on the same SSD, I wonder if it has anything to do with the PARTUUIDS of the devices....?
To test this you could use the clone SD card functionality from within PINN to copy the original PINN on the SSD to another one, and then try to restore your backup on there. Cloning the drive should maintain the PARTUUIDs but may still suffer permission issues 🤷‍♂️ 

OMV contains lots of fancy ways to map your drives. That may also be affected. You may have to tell it to reimport your drives. I'm not that experienced with OMV (yet).

Other strategies to try are to take a complete image of your SSD drive and try to restore that to another drive. For docker, couldn't you push your containers up to a docker repository and then pull them down again to your new installation?