procount/pinn #543: OSMC Pi releases

samnazarko, 2021-08-02:
Hi @procount 

Apologies if I'm opening this in the wrong place.

It seems that NOOBS is now deprecated -- at least via the Pi download page, but PINN is alive and well.
We're planning to release an update to OSMC this weekend bringing Kodi v19 and Pi 4 / 400 support.

Currently, files are served for Pi0/1 and Pi2/3/3B+ at http://download.osmc.tv/installers/noobs/. We will be deprecating Pi 0/1 support for performance reasons. 

Is it good enough for me to just push os/partitions-pi4.json and appropriate boot/root partition tarballs, or is anything else needed for you to pick this up?

Cheers

Sam

procount, 2021-08-03:
Hi Sam,

Thanks for your message and for thinking of me.
I'll be happy to support your new version. 

Currently I still pick up OSMC from the NOOBS repository. AFAIK NOOBS is still alive with v3.7 recently released, it's just not well publicised. I'm sure if you send an update to XECDesign, that he'll be happy to add it to the os_list_v3.json file and then all NOOBS and PINN users will get your new version.

Just for completeness, these are the 2 snippets of json that are currently in os_list_v3.json for the 2 OSMC versions:

```
        {
            "os_name": "OSMC_Pi1",
            "description": "A fast and feature filled open source media center",
            "feature_level": 35120124,
            "icon": "http://download.osmc.tv/installers/noobs/NOOBS-Logo-2.png",
            "marketing_info": "http://download.osmc.tv/installers/noobs/marketing.tar",
            "nominal_size": 2128,
            "os_info": "http://download.osmc.tv/installers/noobs/os-pi1.json",
            "partition_setup": "http://download.osmc.tv/installers/noobs/partition_setup.sh",
            "partitions_info": "http://download.osmc.tv/installers/noobs/partitions-pi1.json",
            "release_date": "2020-11-25",
            "supported_hex_revisions": "2,3,4,5,6,7,8,9,d,e,f,10,11,12,14,19,0092,0093",
            "supported_models": [
                "Pi Model",
                "Pi Compute Module Rev",
                "Pi Zero"
            ],
            "tarballs": [
                "http://download.osmc.tv/installers/noobs/boot-rbp1.tar.xz",
                "http://download.osmc.tv/installers/noobs/root-rbp1.tar.xz"
            ]
        },
        {
            "os_name": "OSMC_Pi2",
            "description": "A fast and feature filled open source media center",
            "feature_level": 0,
            "icon": "http://download.osmc.tv/installers/noobs/NOOBS-Logo-2.png",
            "marketing_info": "http://download.osmc.tv/installers/noobs/marketing.tar",
            "nominal_size": 2128,
            "os_info": "http://download.osmc.tv/installers/noobs/os-pi2.json",
            "partition_setup": "http://download.osmc.tv/installers/noobs/partition_setup.sh",
            "partitions_info": "http://download.osmc.tv/installers/noobs/partitions-pi2.json",
            "release_date": "2020-11-25",
            "supported_hex_revisions": "1040,1041,2082",
            "supported_models": [
                "Pi 2",
                "Pi 3",
                "Pi Compute Module 3"
            ],
            "tarballs": [
                "http://download.osmc.tv/installers/noobs/boot-rbp2.tar.xz",
                "http://download.osmc.tv/installers/noobs/root-rbp2.tar.xz"
            ]
        },
```
So you just need to provide the new json snippet for the PI4 version.  You can send a copy to me too just to make sure you're covered.

PINN also supports some additional JSON fields that can provide extra features. See https://github.com/procount/pinn/wiki/JSON-fields if you are interested. They allow for e.g. checksumming of the tarballs, indicating OSMC can be backed up and putting your OS in one of PINN's groups, like "Media". These additions are compatible with (i.e. ignored by) NOOBS.

samnazarko, 2021-08-03:
Hi,

Thanks for the prompt reply and informative response.

I've made some commits for Pi 4 support and will push this with the next release and give you a ping when this is done. 

Will investigate the media / backup and checksum tags, but would prefer to include them after initial release is verified as good.

Cheers

Sam

procount, 2021-08-03:
That's fine. They are just optional fields, anyway.
What will you call the new release: "OSMC_Pi4"? I can override certain fields until you are ready to support them, like the "group" field to make sure it shows up in the same category as the other OSMC versions.

samnazarko, 2021-08-11:
Hi @procount 

We've now released support for Pi 4 and the initial JSON files are here: http://download.osmc.tv/installers/noobs

I'm happy to accept any recommended changes and push them upstream.

I've pinged @XECDesign for inclusion within NOOBS but didn't receive a response yet and imagine that he may be rather busy at the moment.

Cheers

Sam

procount, 2021-08-11:
Have you created a new installation for the Pi4, or just made your OSMC_Pi2 suitable for all PI2/3/4/400 models?
I can only see that you have updated OSMC_pi2, but there is no additional installation.
If it is one installation for all models, then you also need to update os-pi2.json to include the PI4 models. If you can send me (&XECdesign) the updated snippet with required changes for os_list_v3.json, that would help a lot.

samnazarko, 2021-08-11:
Looks like I jumped the gun and the CDN had not synced yet.
The files should be there now and things should be clearer.

procount, 2021-08-11:
Ah yes, I can see them now.

procount, 2021-08-11:
Sam, you forgot to change the OS name in os-pi4.json. It still says OSMC-Pi2.

procount, 2021-08-11:
And you need to remove the comma after "Pi 4"

procount, 2021-08-11:
@samnazarko - Actually, your partitions-pi4.json needs updating to point to the new pi4 tarballs!

procount, 2021-08-11:
(Sorry, doing this at the same time as 5 other things! Will make suggestions for additional fields subsequently)

samnazarko, 2021-08-11:
Sorry -- too much copy-pasta on my side. Have updated and will await suggestions for  additional fields.