procount/pinn #544: The silentinstall option does not auto install OS.

tigerkelly, 2021-08-04:
I added silentinstall and select=allinstalled to the recovery.cmdline file but it stops at the OS selection screen.

My recovery.cmdline contents.
runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json loglevel=2 sdhci.debug_quirks2=4 reserve=+1000 select=allinstalled silentinstall

Thanks

procount, 2021-08-04:
Silentinstall will not work if there are any OSes already installed (to avoid deleting OSes by mistake), and runinstaller will wipe any OSes installed anyway. So with these options, "select=allinstalled" is not going to select any OSes to install as there are none already installed. You'd be better to explicitly list the OSes you want to be installed.

tigerkelly, 2021-08-04:
It was a fresh install of PINN.
I formatted the SD card to fat32 and copied the PINN files to the SD card, added the silentinstall and select options to the recovery.cmdline file and placed it into raspbaerry pi 4.
Only one OS was in the os directory also.
This works using NOOBS but I need the 'reserve' option in PINN so I can control the size of the first partition.
So that should not have been the issue as to why it did not auto install, right?

procount, 2021-08-04:
Which OS were you trying to install? Did it have any flavour.json file?

procount, 2021-08-04:
I just tried a fresh install of PINN and deleted all OSes in /os apart from Raspios FULL. I added silentinstall to recovery.cmdline and booted it. PINN repartitioned the card and installed Raspios as expected.

So another thought: When yours stopped at the OS installation screen, how many OSes were available to be installed? Was PINN connected to the internet?
If you had more than one OS available, it would not have done the silentinstall. You may have to temporarily disable network connections to stop remote repositories from being loaded. I think this is different behaviour than with NOOBS.

tigerkelly, 2021-08-05:
No flavor.json file just the Paspberry PI OS linux 5.10.17.

tigerkelly, 2021-08-05:
I saw no other OS listed on the screen but I had a network cable plugged in.

I will try without the network plugged in today and let you know.

Kelly

tigerkelly, 2021-08-05:
Ok, I was reading the screen wrong, stupid me.
The silentinstall does work.

What was really happening was that is was installing the OS but it was not rebooting after the install, it just went back to the 'Select OS to install' screen with the OS selected and text saying it was installed.

If I reboot the raspberry pi it goes back to the "Select OS to boot' screen of pinn but my OS is not listed in the dialog box.
So why is my OS not listed?

If I add a autoboot.txt file with boot_partition=6 to the pinn partition it will boot into the OS.

So my real problem is that NOOBS will reboot after install and if only one OS is installed it will use the OS.

I am wondering if their is some file I am missing in the OS directory? 
I just copied the OS directory I used in NOOBS to the PINN os directory, do I need any other files?
Are the json files different between NOOBS and PINN?

This install will run on a headlless RPI 4, so it needs to install and boot into the OS without user interaction.
Once the OS boots up I programmatically add the autoboot.txt file to the PINN dos partition so that the next time it boots a little faster into OS.

lurch, 2021-08-05:
IIRC with an `autoboot.txt` file, PINN gets bypassed entirely (until the file is deleted). So I'm just curious why you're using PINN to do this "silent OS install", rather than installing directly from a `.img` file?

tigerkelly, 2021-08-05:
I need to be able to update the OS in the field.
The SD card is not something the user would have psychical access to.
With an .img file I can not update the SD card easily.

But with NOOBS or PINN I have scripts and programs that allow me to download an OS update to the SD card and then mount the NOOBS or PINN DOS partition and copy in the OS update and remove the autoboot.txt file which causes the NOOBS or PINN to reinstall the OS.

The product is a Drone viewing system.
It would be to much to explain here but what it does is allow up to 4 realtime drone videos to be viewed in one location, with recording and video backhaul to a remote location for the commander to see.

lurch, 2021-08-05:
Ahhh, fair enough. That sounds like a good reason :laughing: 

If you haven't already seen it, maybe you'd be interested in something like https://www.balena.io/os/ ? (a company that I used to work for)

(Feel free to delete this comment @procount if you feel it's inappropriate!)

procount, 2021-08-05:
'''
What was really happening was that is was installing the OS but it was not rebooting after the install, it just went back to the 'Select OS to install' screen with the OS selected and text saying it was installed.
'''
When I did my test above, it booted straight into Raspios. I'll try to see what conditions would prevent that. Having more than one OS would cause it, but I don't think that is your case.

procount, 2021-08-05:
> If I reboot the raspberry pi it goes back to the "Select OS to boot' screen of pinn but my OS is not listed in the dialog box.
So why is my OS not listed?

Can you post a copy of /settings/installed_os.json?

procount, 2021-08-05:
And if you repeat it and it doesn't boot Raspios, could you post a copy of /tmp/debug.log?

I guess your OS is based on Raspios but not exactly the same. Are you using Checksums? Are they correct? 
Maybe your OS was installed, but set non-bootable due to an installation error 🤷‍♂️

tigerkelly, 2021-08-05:
I do not have any any file call installed_os.json on the PINN install disk or even a directory call settings.
I have not done anything with checksum.  I did nothing with checksum when using NOOBS, so I did not even think about it.

I also can not find a /tmp/debug.log file.

procount, 2021-08-05:
Sorry, forgetting you don't know all the details...
Do you have access to PINN via keyboard and monitor? Or just headless?
If headless, add `ssh` to recovery.cmdline to enable you to login over ssh with Putty or similar.

See https://github.com/procount/pinn/wiki/Troubleshooting for how to access the command shell.
The debug log is only stored in temporary RAM, so it is lost if you reboot. So you need to copy it off somewhere (see step 5).
The PINN root directory (/dev/mmcblk0p1) is mounted read-only at /mnt, hence the shenanigans to make it read/write in that step.
PINN's setting partition (/dev/mmcblk0p5) is mounted on /settings. You can also access that from Raspios if you mount the partition.

procount, 2021-08-05:
The debug log is also available from the Maintenance Menu. Select any OS (e.g. PINN), click Fix up and then Show Debug log.
If you have added `vncshare` to recovery.cmdline, you can access PINN through VNC and then copy & paste the log. (`forcetrigger` is also useful if you are using `vncshare`)

tigerkelly, 2021-08-06:
Sorry this took so long.

I first tried ssh but could not find the ip address. I found 1 address on my network at 192.168.0.4 that would give me a login prompt but root/raspberry would not log me in.

So I attached a monitor and keyboard to get the 3 files.
The first files are after it said it installed the OS.
The second files are after I rebooted the card and is came to the Select boot OS screen.
It did not show my OS named DroneView.
[pinn_info.zip](https://github.com/procount/pinn/files/6945038/pinn_info.zip)

procount, 2021-08-06:
Your checksums are wrong, so although your OS was installed, it was marked as not bootable, so it doesn't show up in the menus.
Because you installed silently, no error messages were shown.
The easiest way to continue is to delete the checksums in partitions.json. Those are the lines beginning sha512sum. There may be one in os.json or os_list.json too.
If you edit /settings/installed_os.json and change bootable: false to bootable:true, it might allow your current install to boot.

procount, 2021-08-06:
Checksums are a new feature in PINN compared to NOOBS.
Each tar file and the partition_setup.sh script are checksummed and those values are stored in the json files.
PINN will check that the checksums are correct, but NOOBS ignores them, so that's why it works on NOOBS and not in PINN.
If you want to use checksums, the each time you update the tar files, you should run sha512sum on each file and update the json files with the new checksum values. If you don't want to use them, delete them from the json files.

tigerkelly, 2021-08-06:
Yes that worked.

I choose to ignore check sums for the time being,

Thank you very much.

closed 2021-08-06 by procount