procount/pinn #547: Windows

actraiser30, 2021-08-11:
Hello there
can you add more windows in pinn?

Windows 3.1/95/2000/XP/7/10

thanks

lurch, 2021-08-11:
Sorry for the thumbs-down reaction (which I've now removed), I assumed that this would be a non-starter for copyright reasons. But I didn't actually realise that PINN already includes [Windows98](https://sourceforge.net/projects/pinn/files/os/win98-pi/)!

actraiser30, 2021-08-11:
No problem
About the copyright thing you can just add the windows installers and
people can just add their own legit cd key asked in installation

actraiser30, 2021-08-11:
I forgot win98 runs in qemu and its very slow and buggy,it needs to run
from dosbox
Thanks

procount, 2021-08-11:
I don't have time to convert images from qemu to dosbox I'm afraid. Unless it's a simple conversion from an image file that already exists (and I would need a URL for that) to PINN/NOOBs format, I doubt it will get done.

actraiser30, 2021-08-23:
is it possible to make the img file/partition of windows 2GB or 4GB its only 256mb and i cant install some games to try
or tell me how can i do it my self i understand you dont have time for this things
thanks

lurch, 2021-08-23:
Does https://pinn.mjh.nz/ help?

procount, 2021-08-23:
Hmm.
I don't think Matt's webpage will help, because it only expands the "flexible" partitions.
IIRC, Windows on the Pi uses about 3-4 partitions. I imagine the 1st boot partition is only 256MB, but this should be sufficient for its purpose. I don't think this is where you should be installing games - Are you just trying to copy the game to the SD card whilst it is inserted in a PC?

actraiser30, 2021-08-23:
Yes,but the 256mb img partition of windows its not enough,windows are
installed in an img file inside rapbian os that loads with qemu,and that
img file is 256mb with less that 100mb left in the windows.img file not in
the partition, thats the file i want to expand.not the raspbian partition.

lurch, 2021-08-23:
It sounds like that's something you ought to be asking to wherever/whoever you downloaded your "Windows image" from (and/or check the qemu and/or dosbox documentation - maybe you can add extra "virtual drives" to the emulation?), rather than the PINN developer?

actraiser30, 2021-08-23:
> Maybe I was thinking of a different implementation. Sorry, my knowledge is
> not enough in this area.

No,problem,i will figure it out.

Even windows 11 work in Rpi

Thanks a lot

actraiser30, 2021-08-23:
I did not download the image anywere it was included in pinns,otherwise i
wont be asking anything.read the first post before you "make up things"

lurch, 2021-08-23:
Sorry, I wasn't "making up things", but I've never run (any version of) Windows on a Pi, so I didn't know how it worked.
You previously talked about both qemu and dosbox, and in a recent comment you said "windows are installed in an img file inside rapbian os that loads with qemu" so I assumed that this was an img file that you had downloaded yourself - I had no idea that the Win98 version of PINN actually installed Raspbian.

actraiser30, 2021-08-23:
No problem,pinns rocks anyway

JeromTWinL, 2021-10-11:
@actraiser30 at your windows list, only windows 10 is supported as full system to run on Raspberry Pi 3/4/400

See my project:
https://jeromtwinl.github.io/windows-arm-pinn-installer