procount/pinn #548: can't get tar.xz or gz or.img oses to show up in pinn

harrma, 2021-08-19:
I have tried putting it in os folder on sd card and flash drive and just in flash drive I have tried formatting drives and raspberry pi img installer and pinn image but they just won't show up, please help?

lurch, 2021-08-19:
PINN requires OSes to be packaged in a special PINN-specific format, it can't just flash generic `.tar.xz` or `.img` format OSes.
See https://github.com/procount/pinn/wiki/OSes-supported-by-PINN

DoctorOctagonapus, 2021-09-04:
How can I download these OS's, specifically older versions?

procount, 2021-09-04:
The OSes listed in https://github.com/procount/pinn/wiki/OSes-supported-by-PINN are the total available across all models. When PINN is connected to the internet it will download the currently available list of OSes and filter it according to which ones are suitable for your RPi model and whether they support installation to SSD or not. So you will rarely see all of these OSes available. You may select any OSes that are listed and they are downloaded automatically as they are installed.
PINN will generally only present the latest version of an OS that is available, although some OSes (e.g. Lineage) have multiple versions available. Is there are any specific OS you were thinking about?

DoctorOctagonapus, 2021-09-04:
I was fortunate, I needed LibreELEC 9.2.x but it looks like the repo hasn't been updated since they released 10.