procount/pinn #549: "Replace" option does not change /boot partition size

sltvtr, 2021-08-24:
Hello.

There are two options in PINN allowing to change OS and they are "Reinstall" and "Replace".
And from what I have read:
- "Reinstall" installs new image of the same OS and keeps partition alignment untouched.
- "Replace" takes new OS and places it into old OS space (or unused Project Space). 

The second option should (IMHO) create partitions accordingly to partitions.json of the new OS. Am I right or You're gonna stop me right here? :)

Why I'm asking:
I've recently (maybe week ago) installed new OSMC_Pi4 image on Raspberry Pi 4, plus Batocera_RPi4 plus one Project Space. After that my SD card partition table looks like this:

```
Device          Boot    Start      End  Sectors  Size Id Type
/dev/mmcblk0p1           8192   137215   129024   63M  e W95 FAT16 (LBA)
/dev/mmcblk0p2         137216 62333951 62196736 29.7G  5 Extended
/dev/mmcblk0p5         139264   204797    65534   32M 83 Linux

#1. Batocera
/dev/mmcblk0p6         204800  6811647  6606848  3.2G  c W95 FAT32 (LBA)
/dev/mmcblk0p7        6815744 23650301 16834558    8G 83 Linux

#2. OSMC
/dev/mmcblk0p8       23650304 23814141   163838   80M  c W95 FAT32 (LBA)
/dev/mmcblk0p9       23814144 44843005 21028862   10G 83 Linux

#3. Project space
/dev/mmcblk0p10      44843008 45150207   307200  150M  c W95 FAT32 (LBA)
/dev/mmcblk0p11      45154304 62333951 17179648  8.2G 83 Linux
```

This partition table is correct at this moment.

Last week I had found an issue in OSMC which wouldn't allow to make an OS update because the /boot partition was too small. I have reported that issue and it was very quickly fixed - updated partition.json with greater size of /boot:
https://github.com/osmc/osmc/commit/3d8a4ecb3041748b765b2609de2e02b3388789ad

So I did reinstall this OS, using PINN of course, and partition table didn't change, but it should not, so thats OK. (Note - there was no info/warning/etc - the install finished fine).
Then, I did "replace" OSMC for OSMC and also nothing changed, OS installed fine but partition layout didn't change - still 80MB /boot partition. And this is what I thought would change.
So I've did another thing - I have replaced Project Space for OSMC. Again - installed fine, but no change in partition table, so this time I have 150MB /boot on OSMC.
And last thing I did was to replace OSMC (that first install) for Project Space. All fine, but again, no changes in paritions, so at the end it looks exactly the same (only OSes switched places):

```
Device          Boot    Start      End  Sectors  Size Id Type
/dev/mmcblk0p1           8192   137215   129024   63M  e W95 FAT16 (LBA)
/dev/mmcblk0p2         137216 62333951 62196736 29.7G  5 Extended
/dev/mmcblk0p5         139264   204797    65534   32M 83 Linux

#1. Batocera
/dev/mmcblk0p6         204800  6811647  6606848  3.2G  c W95 FAT32 (LBA)
/dev/mmcblk0p7        6815744 23650301 16834558    8G 83 Linux

#2. Project Space
/dev/mmcblk0p8       23650304 23814141   163838   80M  c W95 FAT32 (LBA)
/dev/mmcblk0p9       23814144 44843005 21028862   10G 83 Linux

#3. OSMC
/dev/mmcblk0p10      44843008 45150207   307200  150M  c W95 FAT32 (LBA)
/dev/mmcblk0p11      45154304 62333951 17179648  8.2G 83 Linux
```
Is this how it all should work? Or am I missing something?
Right now I can't tell what is the difference between "reinstall" or "replace" - both acted the same...

Thanks,
Best regards.

lurch, 2021-08-24:
> Is this how it all should work?

I'm not the PINN developer, but that all sounds entirely reasonable to me.
I think if you want to resize partitions, that's a job for a tool outside of PINN. Try e.g. gparted?

> Right now I can't tell what is the difference between "reinstall" or "replace"

Sounds like "reinstall" re-installs an OS onto the same partitions it was on before (i.e. all other OSes remain untouched, and you have the same OSes available afterwards as you did before), but "replace" allows you to change what OS is on the partitions (so you end up with a different set of OSes afterwards)? Which again, sounds entirely logical to me? :shrug:

procount, 2021-08-24:
The only difference between Reinstall and Replace is the source of the installation. Reinstall does not need you to select the OS, it knows what it should be from the target. Whereas Replace needs you to specify the (compatible) source.

Just use Grparted to resize partitions. I'm not going to reimplement all of that functionality within PINN.

sltvtr, 2021-08-24:
Hello,

thanks for the answers. Now it's all clear. 
I just went to far with assumed functionality ;)
I thought that bootfs + rootfs is common space for one OS _and_ can be accordingly resized within this common space (increase boot and decrease root and vice versa), but it was my mistake.

Can it be safely changed with gparted though? Changing sizes of Space Project partitions without touching any others?

Best regards.

lurch, 2021-08-24:
> Can it be safely changed with gparted though? Changing sizes of Space Project partitions without touching any others?

I've not tried it myself, but I don't see why not? You'd obviously need to reduce the size of the "big" partition first, before increasing the size of the "small" partition.

sltvtr, 2021-08-24:
Ok, I'll try this.
Thanks again for the clarification, now I know how this works.
I think it's safe to mark as closed/resolved.