procount/pinn #550: DietPi won't boot from PINN version installed on SSD (old firmware version?)

waltwright, 2021-08-28:
Note that this is hand-typed and may be mistyped.
Diagnostic message:

Trying partition: 10
lba: 73392128 oem: 'mkfs.fat' volume: '  v        ^  '
rsc 1 fat-sectors 120 c-count 30651 c-size 4 r-dir 1 r-sec 32
Read config.txt bytes        3160 hnd 0x00000040
HUB [01:00] 2.16 000000:01 init port 4 speed 1
Read start4cd.elf bytes    762880 hnd 0x000029f6
Read fixup4cd.dat bytes      3030 hnd 0x000000bf
Firmware: d2b1b7fb01475cb3914b2086299e32d724e832f1  Jul    9 2019 14:38:27
0x00c03114 0x00000000 0x00000000
start4cd.elf: is not compatible
This board requires newer software

procount, 2021-08-28:
ping @matthuisman - Are you able to update DietPi to the latest version?

matthuisman, 2021-08-28:
done. should now be latest 7.5.2
https://github.com/matthuisman/pinn-os/commit/f9c91c1994d95fa030241a90d4af941f3f4d9ebb

waltwright, 2021-08-29:
Repeatable error at 49MB download point.

I think I'm doing this correctly, but I'm getting a repeatable error when I select only DietPi from the PINN Install (Main) Menu and then Install from the toolbar (PINN v3.6 - Built: Jun 30 2021)
    Checksum error
    An incorrect file checksum has been detected in http://downloads.sourceforge.net/project/pinn-matthulsman/os/DietPi/boot.tar.xz

Thanks so much for the effort!

matthuisman, 2021-08-29:
i've just downloaded the boot.tar.xz and confirmed its sha512 is 
386e13e7919b3353f9bde4eea44557d5e577c9c647abb6a1395e1c7eab2d1664440d533dd0d9baae5d45aa212e1ca89912f1d391f28ad56068f95016b33119d5

which is listed here:
https://github.com/matthuisman/pinn-os/blob/master/DietPi/partitions.json

your not using my pinn resize web app by any chance?
That wont scrape the new files until later tonight

waltwright, 2021-08-29:
Thanks for the updated information. Although I had previously rebooted with the same error, this time it made it all the way through. Yes, I originally used the PINN resize web app, but only originally. This is running directly from PINN, to reinstall on top of the previous version of my DietPi partition. Thanks for thinking beyond the issue, though.

The results are in... And it's a good image. Thanks again!!

matthuisman, 2021-08-29:
Oh. Yeh. It'll still be using the resize url.
I just updated that 30mins ago, so that's why it suddenly started working :)

waltwright, 2021-08-29:
It turns out the the new DietPi install didn't reinstall at sda10. Somehow, all partitions above 7 disappeared (all the way up to sda39+40 where Lineage was installed, but not yet configured). DietPi installed at sda6?+sda7 (instead of reusing sda10+11)  and it took up the whole rest of the disk ~300-400 GB.

Currently, blkid shows only /dev/sda1 "RECOVERY", sda5 "SETTINGS", sda6 "boot", sda7 "root", but sda8 - sda40 are gone.

Previously, Raspberry Pi OS "Full" was at sda 6+7, arch4 (not yet configured) was at 8+9, DietPi was at 10+11 (but wasn't able to configure older version), Raspberry Pi OS Lite was at 12+13, etc.
My /data partition (shared with symlinks to common data) was at sda20.

I'm following up in case there is a problem with the image, but maybe it had something to do with the PINN resize web app which initially set up my partition sizes. User error is less likely, but always possible. I had successfully reloaded OS installs  previously.

Is there any way of easily recovering the lost partitions? Or do you have any advice or lessons learned for me related to not falling into that hole again? Thanks.

matthuisman, 2021-08-29:
@procount should be able to advise.
Ive never done the reinstall stuff. I would have just updated the os inside the os itself rather than via pinn. 

All I can say is the latest version still uses just two partitions . Boot partition did get larger though.

I'm not sure how pinn tries to do the reinstalling stuff. Hopefully procount can advise more on what may have happened.

procount, 2021-08-29:
Unfortunately, the symptoms sound like you did as Matt said and did a fresh install of just DietPi, which would have wiped out all of your OSes. This is independent of whether you used Matt's resizing URL or not. The RE-install option is under the Maintenance Menu.
I am trying my best to prevent users making this sort of error (and accidentally running 'runinstaller') by adding more dialog warnings and changing the install icon to a scary skull icon. Perhaps I need to add even more safety features, like having to enter a phrase in a dialog like "DELETE ALL MY OSES" before installing new OSes....?
I also have some ideas to set OSes read-only (within PINN) so that additional actions are required to unlock them before they can be overwritten.
It's difficult to make this foolproof, since the repetition of clicking through dialogs regularly means it's easy just to click through them automatically without really reading them. Any ideas how to improve this are always welcome.
This doesn't really help you now, though, and you should assume you have lost all your partitions.

However, PINN's repartitioning and re-formatting of the card may not actually erase the previous data. So since Dietpi is quite a small OS, and assuming you haven't written any additional data to your drive, there is a small chance you can get this data back. I believe there are programs that can scan your drive looking for previous partition structures and restore them. This can be quite difficult and time consuming, but useful if the data you have lost is important. If you want to go down this route, I suggest you no longer use your drive and take a backup image of the WHOLE drive using dd to an image file, then apply the recovery tools to your image. This way you can have multiple attempts at recovery in case the recovery process itself doesn't work and corrupts any data that is there. This may be more difficult if your SSD is very large.

matthuisman, 2021-08-29:
@procount 
with the latest version of dietpi, the 1st boot partition actually now wants larger size.
I assume with the re-install code, it will just try keep the current size right?
as resizing non-last partitions involves shifting all the data.
in this case i think the old size would still be large enough for the new data

procount, 2021-08-29:
Yes, this is a potential problem, because reinstall, replace and using projectspaces all keep the existing partition layout and sizes.
In these cases, any existing data is being overwritten and lost, so there is potential to adjust the partition sizes within the OS allocation without having to shift data around. In this way I can pinch a bit of the rootfs space and allocate it to the boot partition. It is a bit risky as it may mess up the surrounding partitions, so it will need some development and testing. Meanwhile, the user could manually adjust the partition sizes using Gparted prior to re-installing or replacing an OS.

procount, 2021-08-29:
@waltwright - https://www.simplified.guide/linux/disk-recover-partition-table might help you.

waltwright, 2021-08-30:
I appreciate all of the information that you shared. Yes, it is possible that I installed from the MAIN menu instead of from the MAINTENANCE menu; i.e. user error. Yes, I agree that a more descriptive warning would be helpful, if that was the case, something like "WARNING: This action will overwrite ALL previously installed OS partitions. Use the MAINTENANCE Menu to reinstall a single OS partition that would NOT touch any other previously installed OS partitions." I recommend this simple enhancement, even if eventually you might implement your other idea to auto-lock previously installed partitions too.

It is also an interesting idea to auto-lock each previously installed partition that could through a dialog that asks the user to put a check mark next to each installed partition to show that the user understands that the previously installed OS partitions will be overwritten by invocation of the Main Menu Install function. Of course that method would require more programming logic than simply throwing the previously discussed warning dialog.

Luckily, I did not lose any important data, as this is a means for me to extend my knowledge and segregate my areas of hobby interests, e.g. GPIO experimentation, RetroPie, minimal installs, and other specialized installs. One thing I am working on is a way of sharing some things between systems (like Documents, Pictures, Videos, RetroPie roms, .bashrc, etc., but NOT sharing other things that could cause problems (like some home dot-files). This also might provide a way of backing up that data more specifically and more frequently.

Thanks again.

closed 2021-08-30 by waltwright