procount/pinn #552: Missing OS selection

gearmesh, 2021-09-03:
I downloaded the latest PINN ver 3.5.6 but I don't see nearly the number of OS's that I see on YouTubers with their older version. How do I get to see all the versions?
Thanks. I tried looking everywhere for this answer. I did see someone mention an "...you can set showall..." but I didn't see that option anywhere or a config file to change.
Any tip would be appreciated.

procount, 2021-09-03:
PINN only shows you OSes that are suitable for your model of PI. Showall will allow you to see and install others but that doesn't mean they will work with your Pi.

lurch, 2021-09-03:
I believe it's up to each OS to specify (in its metadata `*.json` files) which models of Raspberry Pi it's compatible with, so if you're seeing fewer OSes than expected, maybe it simply means that not many OSes have declared themselves as compatible with the CM4? :shrug:  (as there are obviously fewer owners of CM4 boards than there are of the "regular" Raspberry Pis)

procount, 2021-09-03:
If you use the showall option and find some pi4 OSes work on your cm4, please let me know and I can add the  necessary data to make them work.