procount/pinn #553: Kali is not installing after being selected through PINN Resizer web app

waltwright, 2021-09-04:
A few weeks ago, I noticed that Kali did not get installed after using Matt Huisman's PINN Resizer web application; however, I thought that maybe I had accidently unchecked it. However, I had the opportunity to blow away and reinstall my systems today and I have the same issue. My system is RPi 4 with SSD only selected. The results of the recovery.cmdline file as created by the PINN Resizer web app are here and kali_linux_full_32_bit =12529 is clearly inside the file.

SSD = 480GB, so I used 446,400MB as the size (7% less than the 1000 bytes = kbyte "new math" calculation for SSD size).

runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline no_group no_default_source select="allnetwork" disablesdimages alt_image_source=http://pinn.mjh.nz/os_list.json?dietpi=5017&raspberrypioslite32-bit=5087&retropie4=5015&arch4=4430&batocera_rpi4=4771&datapartition=322017&projectspace-1=6393&projectspace-2=8182&projectspace-3=10077&kali_linux_full_32-bit=12529&raspberrypiosfull32-bit=12378&twister=14507&ubuntu2010=12144&lineage18-rpi4-atv=13667&projectspace-4=10086 loglevel=2 sdhci.debug_quirks2=4

Kali is listed; however, it is the only OS or configuration that does not show up on the initiation of the PINN Install Main Menu. It just isn't there.

waltwright, 2021-09-04:
After more investigation, I now see that Matt's Resizer selects a Kali version for the RPi4, but that this project does not recognize any Kali distros as ready for the RPi4, so it is likely filtering out Kali.

procount, 2021-09-04:
The problem is that the Kali 20.04 version that PINN currently has, although suitable for the Pi4, is not suitable for installation to SSD. It may need a later version of Kali that supports that.
Matt's website does not know whether you will install to SD card or SSD, so it allows it to be selected, but then PINN rejects it as not being suitable.

closed 2021-09-05 by waltwright