procount/pinn #554: PINN 3.6 totally ignores all attempts to enter Recovery Mode

DoctorOctagonapus, 2021-09-04:
Running on Raspberry Pi 4, upgraded to 3.6, since then the Recovery Mode has been totally inaccessible by any means.

Message on boot reads "To enter recovery mode, hold SHIFT, click mouse, or press joystick", but the keyboard and mouse are apparently totally locked out at this point. All combinations of holding various shift keys and spamming all mouse buttons do absolutely nothing and the system proceeds regardless to the boot menu as if the keyboard and mouse have been frozen out by the system. Is there any way of forcing the recovery menu or am I flattening my card and doing a total reinstall?

procount, 2021-09-04:
That's very odd. It suggests the USB is not working. You can add the 'forcetrigger` option to recovery.cmdline to force recovery mode to be entered. This should allow you to test whether any keyboard/mouse interaction works.

DoctorOctagonapus, 2021-09-04:
I thought it was odd. The keyboard and mouse work perfectly once it's at the OS selection menu, but not before. The forcetrigger option works and again no issues with the keyboard and mouse.

Is it worth adding the recovery menu as an option in the boot menu rather than requiring a key combination, or is that not even possible?

procount, 2021-09-04:
Not really. Due to the architecture, you have to reboot to get from the boot menu to the recovery menu, so it doesn't really help.

But this only started happening in v3.6? What version did you have previously? Which model of PI4 do you have? You might want to look at the troubleshooting section of the wiki and post the results of the debug log and dmesg to see if there is anything that can be learnt from there. You're not using wireless a keyboard/mouse are you?  This version has an updated kernel and firmware I think, so there might be a difference in there I suppose.

lurch, 2021-09-09:
> The keyboard and mouse work perfectly once it's at the OS selection menu, but not before.

I wonder if your keyboard & mouse might be taking a long time to enumerate, and so aren't seen by PINN early-enough in the boot process?

procount, 2021-09-09:
The output of the debug log and dmesg would help identify that.
Unfortunately, the time allowed to press the shift key etc. is fixed at 4 secs. Perhaps I need to make that settable too.....?

procount, 2021-09-09:
Meanwhile you can add remotetimeout=XX and forcetrigger to recovery.cmdline to workaround the issue (where XX is the time to wait in seconds before going to the boot screen). This might provide sufficient time to allow the keyboard to be detected.

procount, 2021-12-28:
closing as no feedback.

closed 2021-12-28 by procount