procount/pinn #555: Zephus bartop issues

procount, 2021-09-06:
In answer to https://www.raspberrypi.org/forums/viewtopic.php?p=1910378#p19103787

Zephus,
 
Thanks for your logs, but I am a little confused by the devices that are reported.
-------------------------------
[    3.155482] usb 1-1.1: new low-speed USB device number 3 using xhci_hcd
[    3.297224] usb 1-1.1: New USB device found, idVendor=17ef, idProduct=608c, bcdDevice= 1.07
[    3.297229] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.297233] usb 1-1.1: Product: Lenovo Calliope USB Keyboard
[    3.297236] usb 1-1.1: Manufacturer: Lenovo
[    3.304201] input: Lenovo Lenovo Calliope USB Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:17EF:608C.0001/input/input0
[    3.365673] hid-generic 0003:17EF:608C.0001: input,hidraw0: USB HID v1.11 Keyboard [Lenovo Lenovo Calliope USB Keyboard] on usb-0000:01:00.0-1.1/input0
[    3.371823] input: Lenovo Lenovo Calliope USB Keyboard System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:17EF:608C.0002/input/input1
[    3.435573] input: Lenovo Lenovo Calliope USB Keyboard Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:17EF:608C.0002/input/input2
[    3.435691] hid-generic 0003:17EF:608C.0002: input,hidraw1: USB HID v1.11 Device [Lenovo Lenovo Calliope USB Keyboard] on usb-0000:01:00.0-1.1/input1
-------------------------------

The above indicates you have a Lenovo Calliope KEYBOARD that uses 3 input devices on port 1. It is being reported as a gamepad with 0 joysticks and 5 buttons. 

-------------------------------
Looking for  "/mnt/Lenovo_Lenovo_Calliope_USB_Keyboard_Consumer_Control_0A_5B.json" 
Loading Mappings mappings from  "/mnt/joy_keys.json" 
-------------------------------

This is very odd, but google indicates other systems have problems with this too.

Then you have a Shanwan PS3 compatible gamepad on port 3:

-------------------------------
[    3.535486] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[    3.670002] usb 1-1.3: New USB device found, idVendor=2563, idProduct=0575, bcdDevice= 2.00
[    3.670008] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.670012] usb 1-1.3: Product: PS3/PC Gamepad
[    3.670016] usb 1-1.3: Manufacturer: SHANWAN
[    3.675626] input: SHANWAN PS3/PC Gamepad as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:2563:0575.0003/input/input3
[    3.675775] hid-generic 0003:2563:0575.0003: input,hidraw2: USB HID v1.10 Gamepad [SHANWAN PS3/PC Gamepad] on usb-0000:01:00.0-1.3/input0
-------------------------------

This is detected by PINN. It cannot find the correct mapping file so it uses the default joy_keys.json, but there is an error reading this joystick:
-------------------------------
Looking for  "/mnt/SHANWAN_PS3/PC_Gamepad_6A_13B.json" 
Loading Mappings mappings from  "/mnt/joy_keys.json" 
Error reading joystick 
-------------------------------

After a while this joystick disconnects itself and then presents itself as an X-BOX-360 pad

-------------------------------
[    4.336934] usb 1-1.3: USB disconnect, device number 4
[    5.121340] usb 1-1.3: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.10
[    5.121346] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    5.121350] usb 1-1.3: Product: Controller
[    5.121354] usb 1-1.3: Manufacturer: SHANWAN
[    5.123956] input: Microsoft X-Box 360 pad as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/input/input6
-------------------------------

PINN did not detect this version, but the json file would have been called something like "/mnt/Microsoft_X-Box_360_pad_0A_0B.json" (you would need to replace the 0's with the appropriate numbers).
This also seems to disconnect after a while and then the joystick presents itself as an Android Gamepad

-------------------------------
[    5.361101] usb 1-1.3: USB disconnect, device number 6
[    5.925488] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[    6.070131] usb 1-1.3: New USB device found, idVendor=2563, idProduct=0526, bcdDevice= 1.00
[    6.070136] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.070140] usb 1-1.3: Product: Android Gamepad
[    6.070144] usb 1-1.3: Manufacturer: SHANWAN
[    6.075053] input: SHANWAN Android Gamepad as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:2563:0526.0005/input/input7
[    6.075213] hid-generic 0003:2563:0526.0005: input,hidraw2: USB HID v1.10 Gamepad [SHANWAN Android Gamepad] on usb-0000:01:00.0-1.3/input0
[    6.078534] input: SHANWAN Android Gamepad System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:2563:0526.0006/input/input8
[    6.145608] input: SHANWAN Android Gamepad Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:2563:0526.0006/input/input9
[    6.145698] hid-generic 0003:2563:0526.0006: input,hidraw4: USB HID v1.01 Device [SHANWAN Android Gamepad] on usb-0000:01:00.0-1.3/input1
-------------------------------

There doesn't seem to be a problem with this last version.

-------------------------------
Looking for  "/mnt/SHANWAN_Android_Gamepad_8A_15B.json" 
Loading Mappings mappings from  "/mnt/joy_keys.json" 
-------------------------------

There also seems to be a second joystick of this type on port 4.
-------------------------------
[    3.909915] usb 1-1.4: New USB device found, idVendor=2563, idProduct=0575, bcdDevice= 2.00
[    3.909921] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.909925] usb 1-1.4: Product: PS3/PC Gamepad
[    3.909929] usb 1-1.4: Manufacturer: SHANWAN
[    3.917593] input: SHANWAN PS3/PC Gamepad as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:2563:0575.0004/input/input4
[    3.917925] hid-generic 0003:2563:0575.0004: input,hidraw3: USB HID v1.10 Gamepad [SHANWAN PS3/PC Gamepad] on usb-0000:01:00.0-1.4/input0
[    4.219438] input: uinput-sample as /devices/virtual/input/input5
-------------------------------

PINN can only handle 2 joysticks. These must be the first 2 joysticks that the linux system finds as /dev/js0 and /dev/js1.
Your keyboard is being identified first as a gamepad, which is a bit of a problem. This may use up one of the joystick devices. You might be able to solve this by swapping the order of the USB ports that you plug your controllers into.
Unless you kept swapping controllers when you booted PINN, the shanwan controller seems to behave as 3 different types of controllers. It could be that PINN is not communicating properly with it, so it keeps swapping until a device driver claims it?

As it seems pretty random which type of joystick your shanwan devices will behave as, you probably need to create separate json files for each version, so no matter which type is presented, PINN can use the correct mapping files. The PS3 version has a '/' in the name, which means you will need to put it in a folder of its own.

So my advice:
1. Just plug one joystick in at a time for testing.
2. Swap the USB port that your keyboard is plugged into with one of the USB controllers, preferably so it is on port 4 (see the logs).
3. take the original joy_keys.json file and copy it to the following files on the SD card:

SHANWAN_Android_Gamepad_8A_15B.json
SHANWAN_PS3/PC_Gamepad_6A_13B.json
Microsoft_X-Box_360_pad_0A_0B.json (You will have to change the zeros to the correct numbers if this is ever detected by PINN.

4. Check the debug logs to make sure the correct file is being loaded instead of joy_keys.json (make a note of the filename)
5. View the debug log (see https://github.com/procount/pinn/blob/master/README_PINN.md#show-debug-log)
6. Press buttons and move the joysticks on the gamepad, refreshing the debug log as you go, and note the debug messages about which buttons and joysticks with their values PINN is detecting.
7. Capture the debug log as you did before.
8. Use the debug log to create a new json file for your joystick.

Zephus77, 2021-09-06:
If I understand I must create 3 files 👍 
SHANWAN_Android_Gamepad_8A_15B.json
SHANWAN_PS3/PC_Gamepad_6A_13B.json
Microsoft_X-Box_360_pad_0A_0B.json

because I can't create "SHANWAN_PS3/PC_Gamepad_6A_13B.json" . I have an error "the name of file can't contain "/" character" 
Where can I find the number for create "Microsoft_X-Box_360_pad_0A_0B.json" ?

Zephus77, 2021-09-07:
This is the new files...
with 2 buttons I can down and up the selection but no button for validate and the joystick don't move the cursor.

`Keyboard found: "/dev/input/event1" 
3.6 
Board revision is  12595474 
Found recovery.rfs at "mmcblk0p1" 
PINN drive: "/dev/mmcblk0" 
mount: /dev/mmcblk0 is already mounted or /mnt busy
Simulate OK 
CecListener thread started 
Looking for  "/mnt/cec_keys.json" 
Loading Mappings mappings from  "/mnt/cec_keys.json" 
CecListener done initializing 
Looking for  "/mnt/SHANWAN_Android_Gamepad_8A_15B.json" 
Loading Mappings mappings from  "/mnt/SHANWAN_Android_Gamepad_8A_15B.json" 
Joy1 key detected 
New =  mainwindow ,  Main Menu 
setWindow  "mainwindow" 
setMenu  "Main Menu" 
Starting dbus 
Starting dhcpcd 
Default language is  "en" 
Default keyboard layout is  "gb" 
"Joystick axis #7 Value: 0" 
"Joystick axis #7 Value: 1" 
"Joystick axis #7 Value: -1" 
Default display mode is  "0" 
Number of bootables =  2 
New drive selected: "mmcblk0" 
umount: /dev/sda1: not mounted
Network up in 7.889 seconds 
Requesting current time 
Downloading "http://downloads.sourceforge.net/project/pinn/BUILD-DATA" to "BUILD|/tmp/BUILD-DATA" 
Downloading "http://downloads.sourceforge.net/project/pinn/README.md" to "README|/tmp/README.md" 
Downloading "http://raw.githubusercontent.com/procount/pinn-os/master/os/overrides.json" to "GROUP|/tmp/overrides.json" 
downloadRepoList:  "http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json" 
Downloading "https://raw.githubusercontent.com/procount/pinn-os/master/os/overrides.json" to "GROUP|/tmp/overrides.json" 
Time set to:  QDateTime("Tue Sep 7 05:13:20 2021") 
Redirection - Re-trying download from "https://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json" 
downloadRepoList:  "https://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json" 
Downloading "https://freefr.dl.sourceforge.net/project/pinn/BUILD-DATA" to "BUILD|/tmp/BUILD-DATA" 
Downloading "https://freefr.dl.sourceforge.net/project/pinn/README.md" to "README|/tmp/README.md" 
"GROUP" 
Downloaded  "GROUP" : "/tmp/overrides.json" 
"BUILD" 
Downloaded  "BUILD" : "/tmp/BUILD-DATA" 
"README" 
Downloaded  "README" : "/tmp/README.md" 
BUILD_IGNORE... 
BUILD_CURRENT... 
Version:  "PINN Version: v3.6" , "v3.6" 
BUILD_NEW... 
Version:  "PINN Version: v3.6" , "v3.6" 
Number of bootables =  2 
Number of bootables =  2 
checking size of file:  "http://download.osmc.tv/installers/noobs/boot-rbp4.tar.xz" 
checking size of file:  "http://download.osmc.tv/installers/noobs/root-rbp4.tar.xz" 
Number of bootables =  2 
Number of bootables =  2 
checking size of file:  "http://mirrors.dotsrc.org/osmc/osmc/download/installers/noobs/boot-rbp4.tar.xz" 
checking size of file:  "http://mirrorservice.org/sites/edge.samnazarko.co.uk/osmc/download/installers/noobs/root-rbp4.tar.xz" 
libpng warning: Unknown iTXt compression type or method
libpng warning: Unknown iTXt compression type or method
libpng warning: Unknown iTXt compression type or method
Number of bootables =  2 
Length: 10547912 files left:  1   "OSMC_Pi4" 
Length: 265587372 files left:  0   "OSMC_Pi4" 
"Joystick axis #7 Value: 32767" 
"Joystick axis #7 Value: 0" 
[... 549 lines cut ...]
[    8.406781] bcmgenet: Skipping UMAC reset
[    8.407905] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[    8.408061] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.358472] [EXFAT] trying to mount...
[    9.537805] [EXFAT] mounted successfully
[    9.543927] [EXFAT] trying to unmount...
[    9.543978] [EXFAT] unmounted successfully
[   13.691933] urandom_read: 3 callbacks suppressed
[   13.691936] random: wpa_supplicant: uninitialized urandom read (32 bytes read)
[   13.704007] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   13.845568] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   37.041593] random: crng init done
[   37.041598] random: 1 urandom warning(s) missed due to ratelimiting
[  150.325542] usb 1-1.3: new low-speed USB device number 6 using xhci_hcd
[  150.467021] usb 1-1.3: New USB device found, idVendor=17ef, idProduct=608c, bcdDevice= 1.07
[  150.467026] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  150.467030] usb 1-1.3: Product: Lenovo Calliope USB Keyboard
[  150.467033] usb 1-1.3: Manufacturer: Lenovo
[  150.473992] input: Lenovo Lenovo Calliope USB Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:17EF:608C.0004/input/input6
[  150.535735] hid-generic 0003:17EF:608C.0004: input,hidraw2: USB HID v1.11 Keyboard [Lenovo Lenovo Calliope USB Keyboard] on usb-0000:01:00.0-1.3/input0
[  150.541863] input: Lenovo Lenovo Calliope USB Keyboard System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:17EF:608C.0005/input/input7
[  150.605635] input: Lenovo Lenovo Calliope USB Keyboard Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:17EF:608C.0005/input/input8
[  150.605756] hid-generic 0003:17EF:608C.0005: input,hidraw3: USB HID v1.11 Device [Lenovo Lenovo Calliope USB Keyboard] on usb-0000:01:00.0-1.3/input1
[  368.596702] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
`

procount, 2021-09-07:
> because I can't create "SHANWAN_PS3/PC_Gamepad_6A_13B.json" . I have an error "the name of file can't contain "/" character"

You need to create a folder/directory called "SHANWAN_PS3" and create the file "PC_Gamepad_6A_13B.json" within that directory. (I think - I've never come across this before).

> Where can I find the number for create "Microsoft_X-Box_360_pad_0A_0B.json" 

You have to wait until PINN detects that configuration and tries to load the file. It will provide the correct numbers then in the name.

You obviously managed to swap your controllers around as your keyboard is now in port 3 and is detected after the gamepad - good!

Your controller is detected as "SHANWAN_Android_Gamepad_8A_15B.json", so this is the file you need to modify. See https://github.com/procount/pinn/blob/master/joy_keys.md#new-joystick-file-format for details.

This controller has 8 analog joystick axes and 15 digital buttons. The json file needs to map the button and axis numbers to button names in the inputs section.

For each button you want to map, first choose a name for it. This name must end with "_btn". So if you have a button called 'A' you may want to call it "A_btn". Press this button and look at the debug log. When you press and release it, you will see 2 messages in the debug log as follows:

    "Joystick btn #2 Value: 1" 
    "Joystick btn #2 Value: 0"

This means this button is identified as button 2. So in the inputs section of your json file you need a line to map this like:

    "A_btn":                [2,1],

Repeat this for all buttons.
Please choose names from https://github.com/procount/pinn/blob/master/joy_keys.md#joy-codes where possible.

Analog joysticks are normally centred in the middle. They have 2 directions: up/down and/or left/right. Each direction needs to be mapped to a name. When you move a joystick down, you may get a message like:

    "Joystick axis #7 Value: 32767" 

and

    "Joystick axis #7 Value: -32767"

when you move it to up.
If you want to name this joystick "LAnalog", then you should map it as follows:

    "LAnalogUp_axis":       [ 7, -1],
    "LAnalogDown_axis":     [ 7, 1],

Joystick names must end in "_axis".
You can add a deadzone value if the joystick does not centre properly, but that doesn't look like the case in your logs.

Please make sure your json file follows the correct syntax otherwise it won't load. For example, every line in a list must have a comma at the end, apart from the last line. You can use https://jsonlint.com/ to validate your file.

Once you have mapped your inputs, you can map their names to various key codes for each menu. If you have used the preferred names, these menu sections should not need much alteration.

If your joystick suddenly stops working, it has probably decided to configure itself as another type, so you need to repeat the above mapping for this new name of joystick and modify the appropriate json filename.

Some analog axis may present as a button and an analog joystick at the same time. In this case, choose whichever one you want to use, but not both. If both are mapped in a menu, you may get double-presses.

Zephus77, 2021-09-08:
I must create an "INPUT:" section or can I notice the mapping in "ANY"
section ?

procount, 2021-09-08:
Yes you need an Inputs section. See some of the other joystick json files on the PINN partition as examples.
It is a 2 step process.
First you may the button or axis ID to a button name, e.g. A_btn, then you use that name in the menu sections further down. See https://github.com/procount/pinn/blob/master/joy_keys.md for details.

Zephus77, 2021-09-10:
Perfect !!!! It s work .Thanks a lot !!!! It s very nice. Thanks for your
help.
Just, when pinn reboot, everytime installation menu appear and I must click
on escape to go on os choice menu. How can I do to show os choice at reboot
?

procount, 2021-09-10:
If you have "forcetrigger" in your recovery.cmdline file, remove it.

Zephus77, 2021-09-11:
There is no  "forcetrigger" inrecovery.cmdline or recovery.cmdline.new
file...

procount, 2021-09-11:
Please post a debug log now you have your joystick working. It should indicate if it detected a trigger from somewhere.
(Try to post it preserving the line endings so I can see the separate lines, as it makes it easier to read).