procount/pinn #556: add NOOBS to PINN install

efa, 2021-09-08:
now that NOOBS is deprecated, and Imager miss the multiboot feature, many users that has NOOBS with more than an OS installed may want to switch to PINN, but do not want to reainstall everything.

Please add a feature to upgrade NOOBS to PINN, keeping all the OS as so.

procount, 2021-09-09:
There's no need to add any new features to do this. It is already compatible. Perhaps what you are asking for is just some instructions to do this.

efa, 2021-09-09:
I looked at the documentations but I can't find other method to install PINN other than formatting the SD card

efa, 2021-09-09:
can you please tell me how to upgrade NOOBS to PINN on an existing SD?

procount, 2021-09-09:
All of the instructions to install PINN assume you are starting from scratch, so the formatting option is only necessary if your SD card is new. Your SD card is already formatted so no need to do that. The documentation should also talk about flashing the img version, but that is not relevant here.
I shall write some more comprehensive instructions, but here are some basic instructions in brief:

**_To upgrade from NOOBS:_**
1. Please make a complete image backup of your whole SD card. Use win32disk imager, Balena Etcher, SD card copier, dd, whatever you like, but make a backup in case you do anything wrong, or in case I forget a step. I will not be responsible if you trash all the OSes that you want to preserve! In fact, I suggest you make a copy of your SD card and then try to upgrade your copy from NOOBS to PINN. If anything goes wrong, just make another copy of your original, thus keeping your original NOOBS SD card safe.
2. Download the latest copy of [pinn-lite.zip](https://sourceforge.net/projects/pinn/files/pinn-lite.zip/download)
3. unzip pinn-lite.zip over the top of your NOOBS installation (i.e. in the first FAT partition of your SD card). If you are doing this on Windows, it will probably be the only partition Windows will report. This is the partition that includes a recovery.rfs file.
4. **VERY IMPORTANT!** - Edit recovery.cmdline and delete the "runinstaller" option from the text line in that file, and save the file back to the SD card. If you do not do this, PINN will delete all your OSes as it re-partitions your SD card for initial PINN use! You did create that initial backup didn't you? :wink:
5. BOOT your upgraded SD card which should now use PINN instead of NOOBS.

You should now be able to benefit from most of the new features available in PINN. Since your OSes were originally installed from NOOBS, it is possible that some PINN features may not work with them, or they will not appear in the correct tabbed categories, but these should be minor issues. If you have any, please let me know.

lurch, 2021-09-09:
Will the NOOBS partition (which of course may have been resized by NOOBS-lite) always be big enough for the PINN-lite files?

procount, 2021-09-09:
Good point. Hopefully! If not it's time to break out gparted, which is why I skipped it in these brief instructions. That's another reason why creating a backup is **SOooo** important.

procount, 2021-09-09:
I should probably go through the process myself and analyse the partition sizes etc. Full NOOBS should not be a problem because you can always delete the OS installation files if you run out of space. NOOBS-Lite will be more tricky. Maybe there are some PINN files that can be omitted for a minimal installation (joystick json files, unwanted DTBs etc). But IIRC, NOOBS & PINN both specify the same minimum partition size of 63MB, so I think it should be ok.

procount, 2021-09-17:
I just tried this for myself and there is still 13.9MB left over after copying PINN-lite over the top of NOOBS.

narspt, 2021-10-03:
Hi, @procount: I did replace NOOBS by PINN on an old RPi system I still have for long time with dual boot Raspbian Jessie + Libreelec 7, right after replace I noticed some issues that may interest you:
1. Raspbian boots properly but Libreelec doesn't, after select it on boot menu the screen just gets black and then tv turns off due to no video signal, it seems to freeze like that forever - no boot to Libreelec, can't understand why... I did look at partitions and all seems fine, and I didn't really change anything other than replacing NOOBS to PINN on the 1st partition.
2. CEC doesn't work at all on PINN, I'm not sure if this is related to the NOOBS to PINN migration (something missing on settings partition or so...) or if PINN CEC implementation just doesn't like my LG TV, but I can tell that CEC always worked on Libreelec/Kodi, anyway I didn't test it with a proper/fresh PINN install yet to confirm.
3. When looking at the PINN 'Replace' option to replace an existing OS, the list of available OSs to "replace to" only shows one single entry for Libreelec, nothing more is listed.

narspt, 2021-10-04:
Just to add: I did test to delete all PINN files on 1st partition and restored NOOBS files there, now Libreelec can boot properly again... think this confirms that there was nothing else wrong with this Libreelec install itself.

procount, 2021-10-04:
@narspt - that is really odd. I really didn't think anything in PINN could cause this.
I would appreciate it if you could do some tests to help investigate the cause of this.

Do you know the boot partition number of your LibreElec?
If so, with NOOBS on the first partition, please add a file called autoboot.txt with the following line in it:
```
boot_partition=8
```
(Please replace 8 with the boot partition of LibreElec).

This should bypass NOOBS and boot straight into LibreElec.
Now replace the NOOBS files with the PINN files again, but keep the autoboot.txt file.
Does it still boot directly into LibreElec?
Now delete autoboot.txt and boot. It should boot into PINN.
Can you select LibreElec and boot into it, or does it result in the blank screen again?

lurch, 2021-10-04:
@procount I have a vague memory about some kind of change happening with the 'reboot' mechanism some time ago - could that have anything to do with this, where the firmware in PINN (`start.elf`) and the firmware in LibreElec are working in slightly different ways? :man_shrugging:

procount, 2021-10-04:
In that case, it would suggest that replacing the OLD NOOBS with the latest NOOBS should also cause the same issue 🤷‍♂️ 
@narspt - Could you also try this (without the autoboot.txt file)?

narspt, 2021-10-05:
Thanks @procount and @lurch, I did a few tests and found that my issue with Libreelec indeed also happens with more recent versions of NOOBS, actually it seems to boot fine until NOOBS v2.4 and since v2.7 I can reproduce the problem with no boot and no video signal exactly as with PINN. Also I can add that the issue happens specifically with Libreelec 7.0.2, but then I tested to update it to 7.0.3 (it's a minor update that basically just updates firmware to add support to Pi2+) and found that with 7.0.3 I can boot it properly with recent NOOBS/PINN then I think @lurch my be correct, probably the problem only happens with more recent versions of NOOBS/PINN being unable to boot OS's including old firmwares!
The test with autoboot.txt was similar to manually selecting OS and produces exactly same problem with recent NOOBS.

Btw, @procount concerning the PINN CEC issue I also pointed above, do you think it may be related to my NOOBS to PINN upgrade (i.e. eventually files/settings missing on SETTINGS partition?) or just incompatibility with my TV really? Later with more time I plan to try a clean install of PINN anyway.

procount, 2021-10-05:
I don't think your CEC issue is related to the NOOBS/PINN upgrade. All the necessary files are on the PINN partition.
CEC is quite a generic term and several manufacturers implement it slightly differently under different brand names.
I use the Broadcom library implementation as used in Berryboot. It may not be suitable for all TV manufacturers. 
I provide a cec_keys.json file to map the remote keycodes to normal keypresses to cater for different keycodes and preferences, but even this may not be sufficient to get all TV brands working. Problems have already been reported with LG TVs in #164 that are still not resolved AFAIK.

narspt, 2021-10-05:
I see, it's not a key mapping issue... my TV just doesn't detect that a CEC device is connected at all when PINN running, probably something to do with initialization really... if you eventually get some free time to look at it and you need help testing you can count on me.